ZoneMinder/zoneminder #3059: Monitor live stream, event review, mp4 download, all broken over one night

cyrildtm, 2020-10-17:
**Describe Your Environment**
- Version of ZoneMinder [You are running the most recent version of ZoneMinder, v1.34.21.No update is necessary.]
- How you installed ZoneMinder [https://wiki.zoneminder.com/Debian_10_Buster_with_Zoneminder_1.34.x_from_ZM_Repo]
- Full name and version of OS
Host OS: Windows 10 Pro 2004 19041.572. Hyper-V Guest: Debian 10 Buster apt up to date
- Browser name and version (if this is an issue with the web interface)

**If the issue concerns a camera**
- Make and Model Raspberry Pi OS v4l2rtspserver
- frame rate 1080p30
- resolution
- ZoneMinder Source Type: ffmpeg

**Describe the bug**
The system was working with zero problems 24 hours ago. Left it running overnight. It broke. 

In FireFox: 
Cannot view Monitor live stream. The Monitor video frame is a spinning circle. 
Cannot view Events without problem. Video plays for 1-2 seconds then gets stuck. Playback time keeps counting towards infinity. 

In Chrome:
Cannot view Monitor live stream. There is no picture or video in the webpage. 
Cannot view Events without problem. Video plays for 1-2 seconds at 1x speed then gets very slow playing with frames apparently skipped. 

Downloaded MP4 videos seem to have the same issue in VLC. 

**To Reproduce**
Steps to reproduce the behavior:
1. Set up a new Debian VM and Zoneminder server. 
2. Leave it running for 3 days. 
3. Observe the errors. 

**Expected behavior**
I don't know what to expect, I just want it to work as yesterday. 

**Debug Logs**

Restarted Host OS, Guest VM, service apache2, service zoneminder, numerous times. 
Checked paths ZM_PATH_ZMS, cgi-bin, /var/run/zm, /tmp/zm, /var/log/zm, all exist and have right permissions with (x) bit and owner www-root:www-root
Time zone changed to somewhere else and back. 
In fact I did not change anything after it gets working with no issues. 

Cleared error messages. Tried open Monitor and Events. Here are the new error messages. 

```

Date/Time | Component | Server | PID | Level | Message | File | Line
-- | -- | -- | -- | -- | -- | -- | --
2020-10-17 18:26:42 | zmc_m1 |   | 1026 | WAR | Buffer overrun at index 31, image 32116, slow down capture, speed up analysis or increase ring buffer size | zm_monitor.cpp | 2497
2020-10-17 18:26:42 | zmc_m1 |   | 1026 | WAR | Buffer overrun at index 35, image 32120, slow down capture, speed up analysis or increase ring buffer size | zm_monitor.cpp | 2497
2020-10-17 18:25:02 | zmc_m1 |   | 1026 | WAR | Buffer overrun at index 14, image 29129, slow down capture, speed up analysis or increase ring buffer size | zm_monitor.cpp | 2497
2020-10-17 18:25:02 | zmc_m1 |   | 1026 | WAR | Buffer overrun at index 11, image 29126, slow down capture, speed up analysis or increase ring buffer size | zm_monitor.cpp | 2497
2020-10-17 18:24:53 | zmc_m1 |   | 1026 | WAR | Buffer overrun at index 8, image 28853, slow down capture, speed up analysis or increase ring buffer size | zm_monitor.cpp | 2497
2020-10-17 18:24:48 | zmc_m1 |   | 1026 | WAR | Buffer overrun at index 8, image 28718, slow down capture, speed up analysis or increase ring buffer size | zm_monitor.cpp | 2497
2020-10-17 18:13:20 | web_js |   | 661 | ERR | getStreamCmdResponse stream error: Socket /var/run/zm/zms-318862s.sock does not exist. This file is created by zms, and since it does not exist, either zms did not run, or zms exited early. Please check your zms logs and ensure that CGI is enabled in apache and check that the PATH_ZMS is set correctly. Make sure that ZM is actually recording. If you are trying to view a live stream and the capture process (zmc) is not running then zms will exit. Please go to http://zoneminder.readthedocs.io/en/latest/faq.html#why-can-t-i-see-streamed-images-when-i-can-see-stills-in-the-zone-window-etc for more information. - checkStreamForErrors() | ?view=watch |  
2020-10-17 18:13:20 | web_php |   | 661 | ERR | Socket /var/run/zm/zms-318862s.sock does not exist. This file is created by zms, and since it does not exist, either zms did not run, or zms exited early. Please check your zms logs and ensure that CGI is enabled in apache and check that the PATH_ZMS is set correctly. Make sure that ZM is actually recording. If you are trying to view a live stream and the capture process (zmc) is not running then zms will exit. Please go to http://zoneminder.readthedocs.io/en/latest/faq.html#why-can-t-i-see-streamed-images-when-i-can-see-stills-in-the-zone-window-etc for more information. | /usr/share/zoneminder/www/includes/functions.php | 2188
2020-10-17 18:12:52 | zms_m1 |   | 1385 | ERR | Backtrace 5: /lib/x86_64-linux-gnu/libc.so.6(clone+0x3f) [0x7f13b2d7d4cf] | zm_signal.cpp | 104
2020-10-17 18:12:52 | zms_m1 |   | 1385 | ERR | Backtrace 4: /lib/x86_64-linux-gnu/libpthread.so.0(+0x7fa3) [0x7f13b5043fa3] | zm_signal.cpp | 104
2020-10-17 18:12:52 | zms_m1 |   | 1385 | ERR | Backtrace 3: /usr/lib/zoneminder/cgi-bin/nph-zms(_ZN11VideoStream23StreamingThreadCallbackEPv+0x150) [0x55d23273eec0] | zm_signal.cpp | 104
2020-10-17 18:12:52 | zms_m1 |   | 1385 | ERR | Backtrace 2: /usr/lib/zoneminder/cgi-bin/nph-zms(_ZN11VideoStream19ActuallyEncodeFrameEPKhibj+0x40) [0x55d23273e750] | zm_signal.cpp | 104
2020-10-17 18:12:52 | zms_m1 |   | 1385 | ERR | Backtrace 1: /lib/x86_64-linux-gnu/libpthread.so.0(+0x12730) [0x7f13b504e730] | zm_signal.cpp | 104
2020-10-17 18:12:52 | zms_m1 |   | 1385 | ERR | Backtrace 0: /usr/lib/zoneminder/cgi-bin/nph-zms(_Z14zm_die_handleriP9siginfo_tPv+0x7a) [0x55d2327770ba] | zm_signal.cpp | 104
2020-10-17 18:12:52 | zms_m1 |   | 1385 | ERR | Signal address is (nil), from 0x55d23273e750 | zm_signal.cpp | 83
2020-10-17 18:12:52 | zms_m1 |   | 1385 | ERR | Got signal 11 (Segmentation fault), crashing | zm_signal.cpp | 52
2020-10-17 18:12:52 | zms_m1 |   | 1384 | ERR | Could not open codec. Error code -542398533 "Generic error in an external library" | zm_mpeg.cpp | 311
2020-10-17 18:12:51 | zms_m1 |   | 1383 | ERR | Backtrace 5: /lib/x86_64-linux-gnu/libc.so.6(clone+0x3f) [0x7f3d046c34cf] | zm_signal.cpp | 104
2020-10-17 18:12:51 | zms_m1 |   | 1383 | ERR | Backtrace 4: /lib/x86_64-linux-gnu/libpthread.so.0(+0x7fa3) [0x7f3d06989fa3] | zm_signal.cpp | 104
2020-10-17 18:12:51 | zms_m1 |   | 1383 | ERR | Backtrace 3: /usr/lib/zoneminder/cgi-bin/nph-zms(_ZN11VideoStream23StreamingThreadCallbackEPv+0x150) [0x558801c70ec0] | zm_signal.cpp | 104
2020-10-17 18:12:51 | zms_m1 |   | 1383 | ERR | Backtrace 2: /usr/lib/zoneminder/cgi-bin/nph-zms(_ZN11VideoStream19ActuallyEncodeFrameEPKhibj+0x40) [0x558801c70750] | zm_signal.cpp | 104
2020-10-17 18:12:51 | zms_m1 |   | 1383 | ERR | Backtrace 1: /lib/x86_64-linux-gnu/libpthread.so.0(+0x12730) [0x7f3d06994730] | zm_signal.cpp | 104
[... 80 lines cut ...]
2020-10-17 18:11:20 | zms_m1 |   | 1249 | ERR | Backtrace 0: /usr/lib/zoneminder/cgi-bin/nph-zms(_Z14zm_die_handleriP9siginfo_tPv+0x7a) [0x56423ff5f0ba] | zm_signal.cpp | 104
2020-10-17 18:11:20 | zms_m1 |   | 1249 | ERR | Backtrace 1: /lib/x86_64-linux-gnu/libpthread.so.0(+0x12730) [0x7ff65a50d730] | zm_signal.cpp | 104
2020-10-17 18:11:20 | zms_m1 |   | 1249 | ERR | Backtrace 2: /usr/lib/zoneminder/cgi-bin/nph-zms(_ZN11VideoStream19ActuallyEncodeFrameEPKhibj+0x40) [0x56423ff26750] | zm_signal.cpp | 104
2020-10-17 18:11:20 | zms_m1 |   | 1249 | ERR | Backtrace 3: /usr/lib/zoneminder/cgi-bin/nph-zms(_ZN11VideoStream23StreamingThreadCallbackEPv+0x150) [0x56423ff26ec0] | zm_signal.cpp | 104
2020-10-17 18:11:20 | zms_m1 |   | 1249 | ERR | Backtrace 4: /lib/x86_64-linux-gnu/libpthread.so.0(+0x7fa3) [0x7ff65a502fa3] | zm_signal.cpp | 104
2020-10-17 18:11:20 | zms_m1 |   | 1249 | ERR | Backtrace 5: /lib/x86_64-linux-gnu/libc.so.6(clone+0x3f) [0x7ff65823c4cf] | zm_signal.cpp | 104
2020-10-17 18:11:20 | zms_m1 |   | 1250 | ERR | Could not open codec. Error code -542398533 "Generic error in an external library" | zm_mpeg.cpp | 311
2020-10-17 18:11:20 | zms_m1 |   | 1251 | ERR | Backtrace 5: /lib/x86_64-linux-gnu/libc.so.6(clone+0x3f) [0x7fa3813484cf] | zm_signal.cpp | 104
2020-10-17 18:11:20 | zms_m1 |   | 1251 | ERR | Backtrace 4: /lib/x86_64-linux-gnu/libpthread.so.0(+0x7fa3) [0x7fa38360efa3] | zm_signal.cpp | 104
2020-10-17 18:11:20 | zms_m1 |   | 1251 | ERR | Backtrace 3: /usr/lib/zoneminder/cgi-bin/nph-zms(_ZN11VideoStream23StreamingThreadCallbackEPv+0x150) [0x5602aad48ec0] | zm_signal.cpp | 104
[... cut ...]
2020-10-17 18:11:19 | zms_m1 |   | 1247 | ERR | Backtrace 2: /usr/lib/zoneminder/cgi-bin/nph-zms(_ZN11VideoStream19ActuallyEncodeFrameEPKhibj+0x40) [0x564ed9978750] | zm_signal.cpp | 104
2020-10-17 18:11:19 | zms_m1 |   | 1247 | ERR | Backtrace 3: /usr/lib/zoneminder/cgi-bin/nph-zms(_ZN11VideoStream23StreamingThreadCallbackEPv+0x150) [0x564ed9978ec0] | zm_signal.cpp | 104
2020-10-17 18:11:19 | zms_m1 |   | 1247 | ERR | Backtrace 4: /lib/x86_64-linux-gnu/libpthread.so.0(+0x7fa3) [0x7fca99724fa3] | zm_signal.cpp

```

connortechnology, 2020-10-19:
We have just released 1.34.22 which contains some fixes that may be relevant.  Please try it and report back.

cyrildtm, 2020-10-21:
Hi Connor, 
Thank you for the reply. 
I left it running for a few days, and it seemed to be working fine without any problems. It is weird to me that the Event replay fixes itself. 
For the Monitor, I have to remove the "mp4" type from the Options -> Images -> Live config, so it's now streaming JPEG. It's weird too since mp4 worked before. 
I will check the new version when I have time.

cyrildtm, 2020-10-26:
Here are some updates. So I tried the new version 1.34.22. After upgrade and OS reboot, broken replay videos started to intermittently show up. I checked some of the broken Events, exported the whole event, and all JPG files looked fine, but the MP4 file was broken. The frame freezes shortly after the playback begun, both in VLC (PC) and Firefox. 
It looked like ffmpeg somehow crashed and the encoding was interrupted, so the MP4 file could not be played back. But I did not get any error message.

cyrildtm, 2020-10-27:
I found [this post](https://forums.zoneminder.com/viewtopic.php?t=23861) from many years ago. 
The symptoms are similar to what I currently have. I followed the suggested fix and appended < /dev/null to FFMPEG_OUTPUT_OPTIONS. But I did not get any useful logs that were related to output encoding. Turning on debug log levels and restarting zm service resulted a lot of info and debug logs generated in the web log window, but nothing related to the ffmpeg text output like those in that post. I have no idea how to confirm whether or not ffmpeg was interrupted.

connortechnology, 2020-10-27:
We will need to see the zmc logs with debug turned on.  

What settings do you have in Monitor -> storage?

connortechnology, 2021-04-20:
Is this still an issue?

cyrildtm, 2021-08-21:
I'm now using 1.36.5 with Quadro nvenc. Event viewing is good. Still can't watch live stream with mp4 format, but that will be a separate issue.

closed 2021-08-21 by cyrildtm