ZoneMinder/zoneminder #3062: ES / ML not starting / object not detected

repomanz, 2020-10-21:
**Describe Your Environment**
- Version of ZoneMinder - latest release version
- How you installed ZoneMinder - unraid container (dlandon)
- Full name and version of OS: unraid
- Browser name and version - mozilla / n/a

**Describe the bug**
Hook processing doesn't seem to be working after upgrading ZM/ES/ML to latest version.  Was working flawlessly before upgrading to latest.  I have looked through configurations, even pulled the latest objectconfig.ini from your github link and the ES 6.0 changes.  Validated my zmevent ini file also has hooks enabled, start/end events, etc

I can attach the zmeventnotification.ini and objectconfig.ini if necessary.

snippet from the eventserver ui log:
|----> FORK:garage-low-res (1), eid:472 Not sending event end alarm, as we did not send a start alarm for this, or start hook processing failed

**To Reproduce**
1) update from previous release to latest ZM/ES/ML releases
2) run the update script on the objectconfig.ini file
3) start zoneminder

**Expected behavior**
Events from Zoneminder / Eventserver are sent to ML for object detection.  Detected objects are returned with yolo_vX

**Debug Logs**
"|----> FORK:garage-low-res (1), eid:472 Not sending event end alarm, as we did not send a start alarm for this, or start hook processing failed"

Will provider further logs upon request.

```

repomanz, 2020-10-22:
enabled some logging.  I do not currently see a zm detect log yet.

[](https://pastebin.com/MBhTRcvp)

knight-of-ni, 2020-10-23:
Hi @repomanz Our github repo is for ZoneMinder bug reports only. While we love the fact that others write code to extend the capabilities of zoneminder, that in no way obligates us to troubleshoot them here. 

We can certainly take a look if the author of zmeventserver finds something in specific in zoneminder that has gone wrong, but this is the best place to start: https://github.com/pliablepixels/zmeventnotification.git

closed 2020-10-23 by knight-of-ni