ZoneMinder/zoneminder #3064: Cause column needs to wrap the Event[Notes] field

vajonam, 2020-10-26:
**Describe Your Environment**
- Version of ZoneMinder - master snapshot
- How you installed ZoneMinder - PPA
- Full name and version of OS - Ubuntu 18.04 LTS (Docker Container) 
- Browser name and version (if this is an issue with the web interface) (Chrome 86+) 

**Describe the bug**
The notes field inside the cause column needs to wrap otherwise it doesn't layout correctly pushing the event id field out of view and not clicable. I store notes about the person/detection in the notes field for each event for historical purposes as I fine tune my system. 

[image]

In the notes field I use `\n` and `\t` to create bit of formatting to its easy to see

closed 2020-10-29 by connortechnology

vajonam, 2020-10-30:
@connortechnology I just added this locally to by build It didn't affect the events in the list view which was what the ticket was for, does it need to be compiled in? if so I can wait for snapshot build to include this

connortechnology, 2020-10-30:
Hmm I just worked on the \n's and didn't notice about \t.  I suppose we could replace those with some &nbsp's.

vajonam, 2020-10-30:
Can we not use the code that formats the code in the popup in the list
view? That is what I was thinking of doing.

connortechnology, 2020-10-30:
There isn't any code in the popup, it's just a textarea.  Doing what textareas do.

perhaps the next best thing would be to stick it in a blockquote or something.

vajonam, 2020-11-05:
Looks like there is a space in the `<br />` , look in the popup. 

[image]

commit 41e88fad (2020-10-29): translate \n to <br/>. Fixes #3064