ZoneMinder/zoneminder #3071: Customizable favicon/app icon

KyleMaas, 2020-11-05:
**Describe Your Environment**
- Version 1.34.22
- Installed via RPMFusion
- OS is CentOS 8
- Applies to all web browsers

**Describe the bug**
I would love to be able to customize the favicon and app/homescreen icon (<link rel="apple-touch-icon" /> and <link rel="icon" />) of ZoneMinder via the web UI.  The reason is twofold: (1) that if used in a corporate environment and with employees who bookmark the interface, it would help them easier identify it without having to dive into the internals of ZoneMinder to change it, and (2) that with ZoneMinder being a web app, it can be pinned to a homescreen on an Apple or Android device, and this would allow for a more useful icon to show up when pinned.

**To Reproduce**
Feature request.  Not a reproducible bug.

**Expected behavior**
1. Navigate to Options->Web.
2. Attach a favicon in either .ico or .png format.
3. Attach a separate homescreen icon in .png format, possibly with options for different sizes.

**Debug Logs**
No debug logs, as this is a feature request and not a bug.

mnoorenberghe, 2020-11-05:
Did you try adding a redirect/alias in your Apache configuration? That would be the simplest

KyleMaas, 2020-11-05:
That is a cleaner way to do it than what I was previously doing (changing out the favicon file in the ZoneMinder directory).  But it would still be much nicer if you could configure that through the web UI.

connortechnology, 2020-11-05:
I agree.  We have code in there that looks for skins/classic/css/whatever/favicon.ico, however in reality, there could be many favicons for various sizes/devices etc.  So this should really be a nice big textarea config setting.

SteveGilvarry, 2021-06-11:
Ok I will take this one on at some point in the next few weeks.

KyleMaas, 2022-04-16:
Any progress on this?

SteveGilvarry, 2024-10-04:
@connortechnology textarea? I was thinking two options.
1. Upload file with a size selection so 16,32 etc and rename the file to replace the existing icons.
2. Upload file with size selection and update DB with file name, then change page to pull file name and location from DB.

connortechnology, 2024-10-05:
Well, my thought is, if you use a favicon generator, you typically get a zip with all the icons and some example html to cutnpaste.  So If we upload a zip file (or a single .ico) and provide a textarea they can cutnpaste the html into to use, that would be enough.