ZoneMinder/zoneminder #3078: Error on launching log console

ncpe20012003, 2020-11-14:
**Describe Your Environment**
- Version of ZoneMinder - 1.34.22
- How you installed ZoneMinder - PPA
- Full name and version of OS - Ubuntu 20.04
- Browser name and version - Chrome Version 86.0.4240.193 (Official Build) (x86_64)

**Describe the bug**
New install on Ubuntu 20.04. One monitor is configured and working as expected. On the home console view, the "Log" clickable link is turning different colors indicating the different messages being logged. I can look at journalctl from terminal and see log messages from the zoneminder service. When I click the "Log" link from the home console view, it spawns a small error window with "TypeError: Array.convert is not a function" and I have to click okay to continue. The "Log" popup then appears, but is empty.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to home console view.
2. Click on "Log".
3. See error - "TypeError: Array.convert is not a function"
4. Click OK to continue.
5. Empty "Log" console appears.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Debug Logs**
```

<insert debug logs here, please make sure they are within the ``` quotes so they are formatted properly>

```

connortechnology, 2020-12-03:
So far I've been able to reproduce this or figure out where it is coming from.

connortechnology, 2020-12-03:
Perhaps look in javascript console for more information: Right click, Inspect Element, Console tab