ZoneMinder/zoneminder #3079: API Error

Vader699, 2020-11-16:
**Describe Your Environment**
- Version of ZoneMinder [release version, development version, or commit]
v1.35.14
- How you installed ZoneMinder [e.g. PPA, RPMFusion, from-source, etc]
deb http://ppa.launchpad.net/iconnor/zoneminder-master/ubuntu focal main
- Full name and version of OS
Ubuntu 20.04 server
- Browser name and version (if this is an issue with the web interface)

**Describe the bug**
When going to /zm/api/monitors.json I am getting 
```
{"success":false,"data":{"name":"An Internal Error Has Occurred.","message":"An Internal Error Has Occurred.","url":"\/zm\/api\/monitors.json?p=%2Fmonitors.json","exception":{"class":"PDOException","code":"42S22","message":"SQLSTATE[42S22]: Column not found: 1054 Unknown column 'Monitor.HourEvents' in 'field list'"}}} 
```
as a response.

Vader699, 2020-11-17:
I guess a complete system reboot fixed my issue.  More so than just restarting services.  My install is working fine now.  Sorry to bother you.

closed 2020-11-17 by Vader699

Cramac, 2020-12-02:
after updating from 1.35.13 to 1.35.14 the same problem appeared