ZoneMinder/zoneminder #3080: "Cannot write to event folder", but events are written just fine.

mcowger, 2020-11-17:
**Describe Your Environment**
- Version of ZoneMinder  v1.34.22.
- How you installed ZoneMinder: Docker Container
- Full name and version of OS: Ubuntu 18.04
- Browser name and version NA

**If the issue concerns a camera**
NA

**Describe the bug**
Repeated errors in logs:
```
Nov 17 03:44:18 324118554c30 web_php[4877]: WAR [Cannot write to event folder /var/cache/zoneminder/events. Check that it exists and is owned by the web account user.]
Nov 17 03:44:19 324118554c30 web_php[4877]: WAR [Cannot write to event folder /var/cache/zoneminder/events. Check that it exists and is owned by the web account user.]
Nov 17 03:44:20 324118554c30 web_php[4877]: WAR [Cannot write to event folder /var/cache/zoneminder/events. Check that it exists and is owned by the web account user.]
Nov 17 03:44:20 324118554c30 web_php[4877]: WAR [Cannot write to event folder /var/cache/zoneminder/events. Check that it exists and is owned by the web account user.]
Nov 17 03:44:21 324118554c30 web_php[4877]: WAR [Cannot write to event folder /var/cache/zoneminder/events. Check that it exists and is owned by the web account user.]
Nov 17 03:44:21 324118554c30 web_php[4878]: WAR [Cannot write to event folder /var/cache/zoneminder/events. Check that it exists and is owned by the web account user.]
```
However, events *are* being written to correctly, I see them in the events history, and can manage/delete them normally.

**To Reproduce**
Steps to reproduce the behavior:
Configure storage, add camera.

**Expected behavior**
Expected: No error

**Debug Logs**
(debug logs aren't notably different)
```

Nov 17 03:49:56 324118554c30 zmc_m1[5499]: INF [zmc_m1] [Garage: images:300 - Capturing at 16.67 fps, capturing bandwidth 674131bytes/sec]
Nov 17 03:49:57 324118554c30 web_php[4776]: WAR [Cannot write to event folder /var/cache/zoneminder/events. Check that it exists and is owned by the web account user.]
Nov 17 03:49:57 324118554c30 zmc_m2[5505]: INF [zmc_m2] [FrontDoor: images:300 - Capturing at 14.29 fps, capturing bandwidth 445139bytes/sec]
Nov 17 03:49:58 324118554c30 web_php[4776]: WAR [Cannot write to event folder /var/cache/zoneminder/events. Check that it exists and is owned by the web account user.]
Nov 17 03:49:59 324118554c30 web_php[4776]: WAR [Cannot write to event folder /var/cache/zoneminder/events. Check that it exists and is owned by the web account user.]
Nov 17 03:50:00 324118554c30 web_php[4776]: WAR [Cannot write to event folder /var/cache/zoneminder/events. Check that it exists and is owned by the web account user.]
Nov 17 03:50:01 324118554c30 zma_m1[5548]: INF [zma_m1] [Garage: 300 - Analysing at 14.29 fps]
Nov 17 03:50:01 324118554c30 web_php[4776]: WAR [Cannot write to event folder /var/cache/zoneminder/events. Check that it exists and is owned by the web account user.]
Nov 17 03:50:01 324118554c30 zma_m2[5549]: INF [zma_m2] [FrontDoor: 300 - Analysing at 14.29 fps]
Nov 17 03:50:02 324118554c30 web_php[4776]: WAR [Cannot write to event folder /var/cache/zoneminder/events. Check that it exists and is owned by the web account user.]
Nov 17 03:50:03 324118554c30 web_php[4776]: WAR [Cannot write to event folder /var/cache/zoneminder/events. Check that it exists and is owned by the web account user.]
Nov 17 03:50:03 324118554c30 zmc_m1[5499]: INF [zmc_m1] [Garage: images:400 - Capturing at 14.29 fps, capturing bandwidth 617848bytes/sec]
Nov 17 03:50:03 324118554c30 zmc_m2[5505]: INF [zmc_m2] [FrontDoor: images:400 - Capturing at 16.67 fps, capturing bandwidth 530473bytes/sec]
Nov 17 03:50:04 324118554c30 web_php[4776]: WAR [Cannot write to event folder /var/cache/zoneminder/events. Check that it exists and is owned by the web account user.]
Nov 17 03:50:04 324118554c30 web_php[4776]: WAR [Cannot write to event folder /var/cache/zoneminder/events. Check that it exists and is owned by the web account user.]
Nov 17 03:50:04 324118554c30 web_php[4776]: WAR [Cannot write to event folder /var/cache/zoneminder/events. Check that it exists and is owned by the web account user.]
Nov 17 03:50:05 324118554c30 web_php[4776]: WAR [Cannot write to event folder /var/cache/zoneminder/events. Check that it exists and is owned by the web account user.]
Nov 17 03:50:06 324118554c30 web_php[4776]: WAR [Cannot write to event folder /var/cache/zoneminder/events. Check that it exists and is owned by the web account user.]
Nov 17 03:50:07 324118554c30 web_php[4776]: WAR [Cannot write to event folder /var/cache/zoneminder/events. Check that it exists and is owned by the web account user.]
Nov 17 03:50:07 324118554c30 zma_m1[5548]: INF [zma_m1] [Garage: 400 - Analysing at 16.67 fps]
Nov 17 03:50:08 324118554c30 zma_m2[5549]: INF [zma_m2] [FrontDoor: 400 - Analysing at 14.29 fps]
Nov 17 03:50:08 324118554c30 web_php[4776]: WAR [Cannot write to event folder /var/cache/zoneminder/events. Check that it exists and is owned by the web account user.]
Nov 17 03:50:09 324118554c30 web_php[4776]: WAR [Cannot write to event folder /var/cache/zoneminder/events. Check that it exists and is owned by the web account user.]
Nov 17 03:50:09 324118554c30 web_php[4776]: WAR [Cannot write to event folder /var/cache/zoneminder/events. Check that it exists and is owned by the web account user.]

```

connortechnology, 2020-11-26:
I am wondering if there is an issue with the web user not being the same user as the zm recording processes.  You said Docker but there are MANY docker images out there.  Which one?  Let's have the output of:
ls -l /var/cache/zoneminder/events
ps aux | grep zm

SteveGilvarry, 2020-11-26:
I have also seen this when you delete Default storage area or don’t use it
so events are writing fine in a new area but the default one is not setup
correctly.

BlueMax, 2020-11-28:
I've seen similar here https://github.com/ZoneMinder/zoneminder/issues/3081 regarding `default` storage path.
The above case should be handled by `zmlinkcontent.sh` though if permissions are wrong indeed.
The other thing is `/var/cache/zoneminder/events`. I'm not sure if the `cache` path (ZM_CACHEDIR) is the proper place to drop events. I wouldn't do it.

connortechnology, 2020-11-28:
It isn't.  I think for 1.36 we will move to /var/lib/zoneminder to match redhat.  

I think we need to look in zma logs for the answer.  I'm suspecting a permissions problem. 
sudo chown www-data -R /var/cache/zoneminder

mcowger, 2020-11-29:
Unfortunately, I've decided that ZM isn't going to meet my needs and abandoned its use, so I wont be able to provide logs.  However, I *can* confirm the www-data user was able to write to that directory without issue.

closed 2020-12-03 by connortechnology

ijmunro69, 2024-11-06:
Had this exact issue after creating a new storage area and applying that to all my cams and also deleting the Default one. Solution is to just recreate the default storage area as it was and check that the folder exists and has correct permissions. You don't need to use it with any cams, just make it and leave it be.