ZoneMinder/zoneminder #3082: Monitor settings tab do not show content

Tony763, 2020-11-28:
**Describe Your Environment**
- v1.35.14
- from-source
- Ubuntu 18.04 with updates from 20.10
- Chromium Verze 87.0.4280.66 (Official) snap (64bit)

**Describe the bug**
Only General tab is shown correctly in monitor settings, other ar not visible, but can by accessed/changed.
Firefox works good.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to monitor settings
2. Click on another tab than general
3. Nothing is visible

[image]
[image]

connortechnology, 2020-11-28:
Works fine for me.  Check Javscript console (Right click, Inspect Element, console tab).  
Clicking  on a tab runs some very simple javascript that hides one tab and shows another.

Tony763, 2020-11-28:
[image]
[image]

Table is selected to be active, save & cancel buttons are visible, but content not.

connortechnology, 2021-05-10:
Ping.  Is this still an issue with current master?

Tony763, 2021-05-10:
I am sorry, I cant test it right now, I will close this and if it will be a issue, I will reopen it.