ZoneMinder/zoneminder #3083: Video is not stored

Tony763, 2020-11-28:
**Describe Your Environment**
v1.35.14 - ffmpeg version N-100082-g81503ac58a-5 with cuda
from-source
Ubuntu 18.04 with updates from 20.10

**If the issue concerns a camera**
- Annker & UNV
- 12 & 25
- 3840x2160 & 2688x1520
- ZoneMinder Source Type: ffmpeg

**Describe the bug**
Video is not stored / folders are not crated

**To Reproduce**

**Expected behavior**

**Debug Logs**
```

Date/Time,Component,Server,PID,Level,Message,File,Line
"2020-11-28 21:56:09",zmfilter_2,,520,WAR,"DiskSpace: Event does not exist at /var/cache/zoneminder/events/2/2020-11-28/2739:ZoneMinder::Event: AlarmFrames => 4 AvgScore => 43 Cause => Motion DefaultVideo => 2739-video.mp4 EndDateTime => 2020-11-28 21:44:02 Frames => 33 Height => 2160 Id => 2739 Length => 7.19 MaxScore => 98 MonitorId => 2 Name => Event-2739 Notes => Motion: New Orientation => ROTATE_0 SaveJPEGs => 3 Scheme => Medium StartDateTime => 2020-11-28 21:43:55 StateId => 1 TotScore => 172 Width => 3840",zmfilter.pl,'-
"2020-11-28 21:56:09",zmfilter_2,,520,WAR,"DiskSpace: Event does not exist at /var/cache/zoneminder/events/2/2020-11-28/2740:ZoneMinder::Event: AlarmFrames => 1 AvgScore => 12 Cause => Motion EndDateTime => 2020-11-28 21:46:10 Frames => 53 Height => 2160 Id => 2740 Length => 9.92 MaxScore => 12 MonitorId => 2 Name => Event-2740 Notes => Motion: New Orientation => ROTATE_0 SaveJPEGs => 3 Scheme => Medium StartDateTime => 2020-11-28 21:46:01 StateId => 1 TotScore => 12 Width => 3840",zmfilter.pl,'-
"2020-11-28 21:56:09",zmfilter_2,,520,WAR,"DiskSpace: Event does not exist at /var/cache/zoneminder/events/2/2020-11-28/2741:ZoneMinder::Event: AlarmFrames => 1 AvgScore => 88 Cause => Motion EndDateTime => 2020-11-28 21:50:11 Frames => 35 Height => 2160 Id => 2741 Length => 9.67 MaxScore => 88 MonitorId => 2 Name => Event-2741 Notes => Motion: New Orientation => ROTATE_0 SaveJPEGs => 3 Scheme => Medium StartDateTime => 2020-11-28 21:50:02 StateId => 1 TotScore => 88 Width => 3840",zmfilter.pl,'-

"2020-11-28 21:50:13",zmeventnotification,,991,INF,"|----> FORK:ANKER (2), eid:2741 Event 2741 for Monitor 2 has finished",zmeventnotification.pl,'-
"2020-11-28 21:50:13",zma_m2,,947,INF,"ANKER: 400 - Analysing at 5.00 fps",zm_monitor.cpp,1338
"2020-11-28 21:50:12",zma_m2,,947,INF,"ANKER: 392 - Closing event 2741, alarm end",zm_monitor.cpp,1709
"2020-11-28 21:50:12",zma_m2,,947,INF,"ANKER: 392 - Left alarm state (2741) - 35(1) images",zm_monitor.cpp,1704
"2020-11-28 21:50:11",zmc_m1,,978,INF,"uvn: images:1000 - Capturing at 25.00 fps, capturing bandwidth 1018679bytes/sec",zm_monitor.cpp,2610
"2020-11-28 21:50:09",zmfilter_2,,520,WAR,"DiskSpace: Event does not exist at /var/cache/zoneminder/events/2/2020-11-28/2739:ZoneMinder::Event: AlarmFrames => 4 AvgScore => 43 Cause => Motion DefaultVideo => 2739-video.mp4 EndDateTime => 2020-11-28 21:44:02 Frames => 33 Height => 2160 Id => 2739 Length => 7.19 MaxScore => 98 MonitorId => 2 Name => Event-2739 Notes => Motion: New Orientation => ROTATE_0 SaveJPEGs => 3 Scheme => Medium StartDateTime => 2020-11-28 21:43:55 StateId => 1 TotScore => 172 Width => 3840",zmfilter.pl,'-
"2020-11-28 21:50:09",zmfilter_2,,520,WAR,"DiskSpace: Event does not exist at /var/cache/zoneminder/events/2/2020-11-28/2740:ZoneMinder::Event: AlarmFrames => 1 AvgScore => 12 Cause => Motion EndDateTime => 2020-11-28 21:46:10 Frames => 53 Height => 2160 Id => 2740 Length => 9.92 MaxScore => 12 MonitorId => 2 Name => Event-2740 Notes => Motion: New Orientation => ROTATE_0 SaveJPEGs => 3 Scheme => Medium StartDateTime => 2020-11-28 21:46:01 StateId => 1 TotScore => 12 Width => 3840",zmfilter.pl,'-
"2020-11-28 21:50:09",zmfilter_2,,520,WAR,"DiskSpace: Event does not exist at /var/cache/zoneminder/events/2/2020-11-28/2741:ZoneMinder::Event: AlarmFrames => 1 AvgScore => 88 Cause => Motion Frames => 20 Height => 2160 Id => 2741 Length => 5.60 MaxScore => 88 MonitorId => 2 Name => New Event Notes => Motion: New Orientation => ROTATE_0 SaveJPEGs => 3 Scheme => Medium StartDateTime => 2020-11-28 21:50:02 StateId => 1 TotScore => 88 Width => 3840",zmfilter.pl,'-

"2020-11-28 21:50:07",zmc_m1,,978,INF,"uvn: images:900 - Capturing at 25.00 fps, capturing bandwidth 1017184bytes/sec",zm_monitor.cpp,2610
"2020-11-28 21:50:07",zmeventnotification,,991,INF,"|----> FORK:ANKER (2), eid:2741 Sending event_start notification for EID:2741 over MQTT",zmeventnotification.pl,'-
"2020-11-28 21:50:06",zma_m1,,981,INF,"uvn: 200 - Analysing at 5.88 fps",zm_monitor.cpp,1338
"2020-11-28 21:50:05",zmc_m2,,944,INF,"ANKER: images:800 - Capturing at 12.50 fps, capturing bandwidth 2229761bytes/sec",zm_monitor.cpp,2610
"2020-11-28 21:50:05",zmesdetect_m2,,995,FAT,"error: Traceback:Traceback (most recent call last):  File ""/usr/local/lib/python3.8/dist-packages/zmes_hook_helpers/utils.py"", line 178, in download_files    input_file = opener.open(url)  File ""/usr/lib/python3.8/urllib/request.py"", line 531, in open    response = meth(req, response)  File ""/usr/lib/python3.8/urllib/request.py"", line 640, in http_response    response = self.parent.error(  File ""/usr/lib/python3.8/urllib/request.py"", line 569, in error    return self._call_chain(*args)  File ""/usr/lib/python3.8/urllib/request.py"", line 502, in _call_chain    result = func(*args)  File ""/usr/lib/python3.8/urllib/request.py"", line 649, in http_error_default    raise HTTPError(req.full_url, code, msg, hdrs, fp)urllib.error.HTTPError: HTTP Error 404: Not FoundDuring handling of the above exception, another exception occurred:Traceback (most recent call last):  File ""/var/lib/zmeventnotification/bin/zm_detect.py"", line 236, in main_handler    filename1, filename2, filename1_bbox, filename2_bbox = utils.download_files(  File ""/usr/local/lib/python3.8/dist-packages/zmes_hook_helpers/utils.py"", line 180, in download_files    g.logger.Error(e)  File ""/usr/local/lib/python3.8/dist-packages/pyzm/ZMLog.py"", line 373, in Error    _log('ERR',message,caller)  File ""/usr/local/lib/python3.8/dist-packages/pyzm/ZMLog.py"", line 306, in _log    conn.execute(cmd)  File ""/usr/local/lib/python3.8/dist-packages/sqlalchemy/engine/base.py"", line 1011, in execute    return meth(self, multiparams, params)  File ""/usr/local/lib/python3.8/dist-packages/sqlalchemy/sql/elements.py"", line 298, in _execute_on_connection    return connection._execute_clauseelement(self, multiparams, params)  File ""/usr/local/lib/python3.8/dist-packages/sqlalchemy/engine/base.py"", line 1124, in _execute_clauseelement    ret = self._execute_context(  File ""/usr/local/lib/python3.8/dist-packages/sqlalchemy/engine/base.py"", line 1316, in _execute_context    self._handle_dbapi_exception(  File ""/usr/local/lib/python3.8/dist-packages/sqlalchemy/engine/base.py"", line 1514, in _handle_dbapi_exception    util.raise_(exc_info[1], with_traceback=exc_info[2])  File ""/usr/local/lib/python3.8/dist-packages/sqlalchemy/util/compat.py"", line 182, in raise_    raise exception  File ""/usr/local/lib/python3.8/dist-packages/sqlalchemy/engine/base.py"", line 1276, in _execute_context    self.dialect.do_execute(  File ""/usr/local/lib/python3.8/dist-packages/sqlalchemy/engine/default.py"", line 593, in do_execute    cursor.execute(statement, parameters)  File ""/usr/local/lib/python3.8/dist-packages/mysql/connector/cursor_cext.py"", line 248, in execute    prepared = self._cnx.prepare_for_mysql(params)  File ""/usr/local/lib/python3.8/dist-packages/mysql/connector/connection_cext.py"", line 647, in prepare_for_mysql    result[key] = self._cmysql.convert_to_mysql(value)[0]_mysql_connector.MySQLInterfaceError: Python type HTTPError cannot be converted",zm_detect.py,240

"2020-11-28 21:50:05",zmesdetect_m2,,995,ERR,"Error downloading files: Python type HTTPError cannot be converted",zm_detect.py,239
"2020-11-28 21:50:05",web_php,,518,FAT,"Can't create frame capture images from video because there is no video file for this event at /var/cache/zoneminder/events/2/2020-11-28/2741/",/usr/share/zoneminder/www/views/image.php,267

"2020-11-28 21:50:04",zmesdetect_m2,,995,INF,"Importing local classes for Object/Face",zm_detect.py,225
"2020-11-28 21:50:04",web_php,,175,INF,"Login successful for user ""admin""",/usr/share/zoneminder/www/api/app/Controller/AppController.php,86
"2020-11-28 21:50:04",zmesdetect_m2,,995,INF,"'---------| pyzm version: 0.2.4, ES version: 6.0.5 , OpenCV version: 4.5.0|------------",zm_detect.py,196
[... cut ...]
"2020-11-28 21:37:03",zmesdetect_m1,,2123,ERR,"Error downloading files: Python type HTTPError cannot be converted",zm_detect.py,239
"2020-11-28 21:36:48",zmc_m1,,494,ERR,"Error writing trailer Error number 143 occurred",zm_videostore.cpp,545
"2020-11-28 21:36:41",zmesdetect_m1,,2098,ERR,"Error downloading files: Python type HTTPError cannot be converted",zm_detect.py,239
"2020-11-28 21:36:38",zmc_m1,,494,ERR,"Error writing trailer Error number 143 occurred",zm_videostore.cpp,545
"2020-11-28 21:36:34",zmc_m2,,1696,ERR,"Error writing trailer Error number 105 occurred",zm_videostore.cpp,545

```

Tony763, 2020-11-28:
FFMPEG should contain all necessary:

```
ffmpeg version N-100082-g81503ac58a-5 Copyright (c) 2000-2020 the FFmpeg developers
  built with gcc 10 (Ubuntu 10.2.0-13ubuntu1)
  configuration: --prefix=/usr --extra-version=5 --enable-cuda-nvcc --enable-cuvid --enable-nvenc --enable-nonfree --enable-libnpp --extra-cflags='-I/usr/local/cuda-11.1/include -I/home/dev/build/include' --extra-ldflags='-L/usr/local/cuda-11.1/lib64 -L/home/dev/build/lib64' --toolchain=hardened --libdir=/usr/lib/x86_64-linux-gnu --incdir=/usr/include/x86_64-linux-gnu --arch=amd64 --enable-gpl --disable-stripping --disable-filter=resample --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librabbitmq --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opencl --enable-opengl --enable-sdl2 --enable-libzimg --enable-libdrm --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared
  libavutil      56. 61.100 / 56. 61.100
  libavcodec     58.114.100 / 58.114.100
  libavformat    58. 64.100 / 58. 64.100
  libavdevice    58. 11.103 / 58. 11.103
  libavfilter     7. 91.100 /  7. 91.100
  libswscale      5.  8.100 /  5.  8.100
  libswresample   3.  8.100 /  3.  8.100
  libpostproc    55.  8.100 / 55.  8.100

```

connortechnology, 2020-11-28:
I suspect tha tyou have nothing selected in the Storage tab of your monitor.  You need to choose how to store the video

Tony763, 2020-11-28:
[image]
Default have 86%

johncolby, 2020-11-28:
I presume you had a working setup previously? Or is this a new install? 

If everything was working fine, is there any chance you are now getting directory naming errors like this?

https://forums.zoneminder.com/viewtopic.php?f=42&t=30207

Tony763, 2020-11-28:
New docker image, but config files are from previous installation. 1.34 worked fine.

connortechnology, 2020-11-28:
Ook how about debug logs from zma? zma is the process that writes the event data to disk so the answer will be there.

BlueMax, 2020-11-29:
Did you ever change the default path in `options/storage`?

Tony763, 2020-11-29:
Docker redirect default location to my disk. `-v "/home/tony/Zoneminder/data":"/var/cache/zoneminder":rw \`
What I see now, that when a new event occur:
[image]
New directory is created, but number is -1.
[image]

After second camera trigered, nubers of evets seems to rolled back and events  started from few numbers back.
[image]

But directory make next numbers:
[image]

Tony763, 2020-11-29:
Just compiled from different commit:
a67fb0bb16cb9718c5bee74a64f5c6c8fc5f015a **eslint** 27.11. NOK
416d68a9718764dc09d8e4fa5f546632bf94df9b r**ough in support for frames view search function** 14.11. OK

vajonam, 2020-11-30:
@Tony763 that was fixed in 159edbfce9d983eac7f4f5c87a117a50bc14dbcd

Tony763, 2020-11-30:
New error occourred, 
Log is full of `Memory map file '/dev/shm/zm.mmap.2' does not exist in zmMemAttach. zmc might not be running.`

[image]
[image]

After recompilation, I restarted container. I also tried tried a suggested `` with no luck.
[image]

What is odd is a time of log and real time, I differ by a lot and difference is still growing. Each new row appear in cca 15 sec and differ from previous by 0-1 sec. But Load is under 1 so its not by cpu overload.

vajonam, 2020-11-30:
AKAIK that is just zmtrigger recovering from a restart or a previous state, once zmc start capturing video you are fine.

Tony763, 2020-11-30:
Yop, you are right, message disappeared, but log still differ by time. Difference started with  7 mins, now its 9 mins.

vajonam, 2020-11-30:
can you close this issue. possibly open a different issue for the logging (which I dont really understand from your explanation) .

Tony763, 2020-11-30:
Issue solved.

closed 2020-11-30 by Tony763