ZoneMinder/zoneminder #3086: Slow log, new lines have a time in past

Tony763, 2020-11-30:
**Describe Your Environment**
v1.35.14 -latest master - ffmpeg version N-100082-g81503ac58a-5 with cuda
from-source
Ubuntu 18.04 with updates from 20.10

**If the issue concerns a camera**
- Annker
- 12
- 3840x2160 
- ZoneMinder Source Type: ffmpeg

**Describe the bug**
Log is out of sync with time, new lines appear with time in past., Difference between times increases in time. 

**To Reproduce**

**Expected behavior**
When new line appear, it should have actual time.

**Debug Logs**

Check screen video: [log.zip](https://github.com/ZoneMinder/zoneminder/files/5618382/log.zip)

[image]
[image]

vajonam, 2020-11-30:
Can you confirm that the time inside your container matched? also logs aren't written all the time in INFO/WARN levels, so a delay of a few minutes is normal. have look in the logs folder /var/log/zm (for me) for all the logs.

solarpenguin1, 2021-04-19:
Note how the same error message repeats over and over?  I am having the same issue with version 1.34.23 and 1.34.25.

[image]
[image]

connortechnology, 2021-04-20:
Fixed in aaa130937764730147670cdbe29b4aec17e7c6dc released in 1.34.26

closed 2021-04-20 by connortechnology