ZoneMinder/zoneminder #3089: Lost cameras due to "Invalid font file"

willbshaw, 2020-12-03:
**Describe Your Environment**
- Version of ZoneMinder [release version, development version, or commit] 1.35.14
- How you installed ZoneMinder [e.g. PPA, RPMFusion, from-source, etc] RPM
- Full name and version of OS: Mint
- Browser name and version (if this is an issue with the web interface)

**If the issue concerns a camera**
- Make and Model
- frame rate
- resolution
- ZoneMinder Source Type:

**Describe the bug**
A clear and concise description of what the bug is.

Updated to 1.35.14 and lost cameras.  Log file shows "Invalid font file".  Found a config item about font file that showed "/usr/share/zoneminder/fonts/default.zmfnt", but there is no such folder.  I do not know if this is what is causing the loss of cameras (because it cannot overlay the timestamp?).

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Debug Logs**
```

<insert debug logs here, please make sure they are within the ``` quotes so they are formatted properly>

2020-12-03 14:41:00 | zmdc |   | 16304 | INF | 'zma -m 3' crashed, signal 134 | zmdc.pl | -
-- | -- | -- | -- | -- | -- | -- | --
2020-12-03 14:41:00 | zma_m3 |   | 17514 | PNC | Invalid font file. | zm_image.cpp | 498
2020-12-03 14:41:00 | zmdc |   | 17514 | INF | 'zma -m 3' started at 20/12/03 14:41:00 | zmdc.pl | -
2020-12-03 14:41:00 | zmdc |   | 16304 | INF | 'zma -m 3' starting at 20/12/03 14:41:00, pid = 17514 | zmdc.pl | -
2020-12-03 14:40:59 | zmdc |   | 16304 | INF | Starting pending process, zma -m 3 | zmdc.pl | -
2020-12-03 14:40:58 | zmdc |   | 16304 | INF | 'zmc -m 3' crashed, signal 134 | zmdc.pl | -
2020-12-03 14:40:58 | zmc_m3 |   | 17503 | PNC | Invalid font file. | zm_image.cpp | 498
2020-12-03 14:40:58 | zmdc |   | 16304 | INF | 'zma -m 2' crashed, signal 134 | zmdc.pl | -
2020-12-03 14:40:58 | zmdc |   | 17503 | INF | 'zmc -m 3' started at 20/12/03 14:40:58 | zmdc.pl | -
2020-12-03 14:40:58 | zmdc |   | 16304 | INF | 'zmc -m 3' starting at 20/12/03 14:40:58, pid = 17503 | zmdc.pl | -
2020-12-03 14:40:58 | zmdc |   | 16304 | INF | Starting pending process, zmc -m 3 | zmdc.pl | -
2020-12-03 14:40:57 | zmdc |   | 16304 | INF | 'zma -m 1' crashed, signal 134 | zmdc.pl | -
2020-12-03 14:40:57 | zma_m2 |   | 17494 | PNC | Invalid font file. | zm_image.cpp | 498
2020-12-03 14:40:57 | zmdc |   | 16304 | INF | 'zmc -m 2' crashed, signal 134

```

connortechnology, 2020-12-03:
grab the font file from https://github.com/ZoneMinder/zoneminder/blob/master/fonts/default.zmfnt and put it in /usr/share/zoneminder/fonts

willbshaw, 2020-12-03:
That got me back up and running.  Thank you so much!!!

Still don't know why there was not a folder or font, or why it only stopped working with this version.

Thanks again.

closed 2020-12-03 by willbshaw