ZoneMinder/zoneminder #3091: Linked monitors broken in latest releases

johncolby, 2020-12-05:
**Describe Your Environment**
- Version of ZoneMinder: `1.35.15~20201204114937-focal`
- How you installed ZoneMinder: PPA https://launchpad.net/~iconnor/+archive/ubuntu/zoneminder-master
- Full name and version of OS: Ubuntu 20.04
    <details>

    ```bash
    ❯ inxi -Faz
    System:
      Kernel: 5.4.0-54-generic x86_64 bits: 64 compiler: gcc v: 9.3.0
      parameters: BOOT_IMAGE=/boot/vmlinuz-5.4.0-54-generic
      root=UUID=d62ea949-c916-41b8-92ac-c64cfa9512b0 ro quiet splash
      vt.handoff=7
      Console: tty 1 dm: GDM3 3.36.3 Distro: Ubuntu 20.04.1 LTS (Focal Fossa)
    Machine:
      Type: Desktop Mobo: Gigabyte model: GA-78LMT-USB3 6.0 v: SEx
      serial: <filter> BIOS: Award v: F2 date: 11/25/2014
    CPU:
      Topology: 8-Core model: AMD FX-8350 bits: 64 type: MCP arch: Bulldozer
      family: 15 (21) model-id: 2 stepping: N/A microcode: 6000852
      L2 cache: 2048 KiB
      flags: avx lm nx pae sse sse2 sse3 sse4_1 sse4_2 sse4a ssse3 svm
      bogomips: 63996
      Speed: 1919 MHz min/max: 1400/4000 MHz boost: enabled Core speeds (MHz):
      1: 1389 2: 1399 3: 1399 4: 1399 5: 1399 6: 1398 7: 1399 8: 1399
      Vulnerabilities: Type: itlb_multihit status: Not affected
      Type: l1tf status: Not affected
      Type: mds status: Not affected
      Type: meltdown status: Not affected
      Type: spec_store_bypass
      mitigation: Speculative Store Bypass disabled via prctl and seccomp
      Type: spectre_v1
      mitigation: usercopy/swapgs barriers and __user pointer sanitization
      Type: spectre_v2 mitigation: Full AMD retpoline, IBPB: conditional, STIBP:
      disabled, RSB filling
      Type: srbds status: Not affected
      Type: tsx_async_abort status: Not affected
    Graphics:
      Device-1: NVIDIA GP107 [GeForce GTX 1050 Ti] vendor: Gigabyte
      driver: nvidia v: 390.138 bus ID: 01:00.0 chip ID: 10de:1c82
      Display: server: X.org 1.20.8 driver: nvidia
      unloaded: fbdev,modesetting,nouveau,vesa compositor: gnome-shell
      tty: 80x15
      Message: Advanced graphics data unavailable in console. Try -G --display
    Audio:
      Device-1: AMD SBx00 Azalia vendor: Gigabyte driver: snd_hda_intel
      v: kernel bus ID: 00:14.2 chip ID: 1002:4383
      Device-2: NVIDIA GP107GL High Definition Audio vendor: Gigabyte
      driver: snd_hda_intel v: kernel bus ID: 01:00.1 chip ID: 10de:0fb9
      Sound Server: ALSA v: k5.4.0-54-generic
    Network:
      Device-1: Realtek RTL8111/8168/8411 PCI Express Gigabit Ethernet
      vendor: Gigabyte driver: r8169 v: kernel port: ce00 bus ID: 03:00.0
      chip ID: 10ec:8168
      IF: eth1 state: up speed: 1000 Mbps duplex: full mac: <filter>
      IF-ID-1: br-3bb03841d807 state: up speed: N/A duplex: N/A mac: <filter>
      IF-ID-2: br0 state: up speed: N/A duplex: N/A mac: <filter>
      IF-ID-3: docker0 state: down mac: <filter>
      IF-ID-4: docker_gwbridge state: up speed: N/A duplex: N/A mac: <filter>
      IF-ID-5: tap0 state: unknown speed: 10 Mbps duplex: full mac: <filter>
      IF-ID-6: tun0 state: unknown speed: 10 Mbps duplex: full mac: N/A
      IF-ID-7: veth0a1183d state: up speed: 10000 Mbps duplex: full
      mac: <filter>
      IF-ID-8: veth197c09f state: up speed: 10000 Mbps duplex: full
      mac: <filter>
    Drives:
      Local Storage: total: 1.14 TiB used: 282.63 GiB (24.3%)
      SMART Message: Unable to run smartctl. Root privileges required.
      ID-1: /dev/sda vendor: Samsung model: SSD 850 EVO 250GB size: 232.89 GiB
      block size: physical: 512 B logical: 512 B speed: 3.0 Gb/s
[... 369 lines cut ...]
11/30/20 00:06:45.168207 zma_m2[1129966].DB1-zm_monitor.cpp/1716 [Not leaving ALERT beacuse image_count(1759)-last_alarm_count(1736) > post_event_count(25) and timestamp.tv_sec(1606723605) - recording.tv_src(1606723593) >= min_section_length(10)]
11/30/20 00:06:45.168284 zma_m2[1129966].DB1-zm_event.cpp/637 [Writing capture frame 57 to /mnt/Data/zoneminder/events/2/2020-11-30/1100711/00057-capture.jpg]
11/30/20 00:06:45.169110 zma_m2[1129966].DB1-zm_event.cpp/680 [Frame delta is 1606723593.872324 - 1606723605.158957 = 11.29]
11/30/20 00:06:45.169137 zma_m2[1129966].DB1-zm_event.cpp/724 [Not Adding 1 frames to DB because write_to_db:0 or frames > analysis fps 5.128205 or BULK]
11/30/20 00:06:45.304275 zma_m2[1129966].DB1-zm_monitor.cpp/1716 [Not leaving ALERT beacuse image_count(1760)-last_alarm_count(1736) > post_event_count(25) and timestamp.tv_sec(1606723605) - recording.tv_src(1606723593) >= min_section_length(10)]
11/30/20 00:06:45.304361 zma_m2[1129966].DB1-zm_event.cpp/637 [Writing capture frame 58 to /mnt/Data/zoneminder/events/2/2020-11-30/1100711/00058-capture.jpg]
11/30/20 00:06:45.305824 zma_m2[1129966].DB1-zm_event.cpp/680 [Frame delta is 1606723593.872324 - 1606723605.287351 = 11.42]
11/30/20 00:06:45.305849 zma_m2[1129966].DB1-zm_event.cpp/724 [Not Adding 2 frames to DB because write_to_db:0 or frames > analysis fps 5.128205 or BULK]
11/30/20 00:06:45.507835 zma_m2[1129966].DB1-zm_monitor.cpp/1716 [Not leaving ALERT beacuse image_count(1761)-last_alarm_count(1736) > post_event_count(25) and timestamp.tv_sec(1606723605) - recording.tv_src(1606723593) >= min_section_length(10)]
11/30/20 00:06:45.507944 zma_m2[1129966].DB1-zm_event.cpp/637 [Writing capture frame 59 to /mnt/Data/zoneminder/events/2/2020-11-30/1100711/00059-capture.jpg]
11/30/20 00:06:45.509429 zma_m2[1129966].DB1-zm_event.cpp/680 [Frame delta is 1606723593.872324 - 1606723605.477776 = 11.61]
11/30/20 00:06:45.509455 zma_m2[1129966].DB1-zm_event.cpp/724 [Not Adding 3 frames to DB because write_to_db:0 or frames > analysis fps 5.128205 or BULK]
11/30/20 00:06:45.711500 zma_m2[1129966].INF-zm_monitor.cpp/1704 [Driveway lowres: 1762 - Left alarm state (1100711) - 59(9) images]
11/30/20 00:06:45.711571 zma_m2[1129966].INF-zm_monitor.cpp/1709 [Driveway lowres: 1762 - Closing event 1100711, alarm end]
11/30/20 00:06:45.711595 zma_m2[1129966].DB1-zm_event.cpp/573 [Inserting 3 frames]
11/30/20 00:06:45.718414 zma_m2[1129966].DB1-zm_event.cpp/598 [INSERT FRAMES: sql was INSERT INTO `Frames` (`EventId`, `FrameId`, `Type`, `TimeStamp`, `Delta`, `Score`) VALUES
( 1100711, 57, 'Normal', from_unixtime( 1606723605 ), 11.29, 0 ),
( 1100711, 58, 'Normal', from_unixtime( 1606723605 ), 11.42, 0 ),
( 1100711, 59, 'Normal', from_unixtime( 1606723605 ), 11.61, 0 )]
11/30/20 00:06:45.718466 zma_m2[1129966].DB1-zm_event.cpp/271 [Updating frames delta by 0 sec 0 usec]
11/30/20 00:07:33.310726 zma_m2[1129966].INF-zm_monitor.cpp/1338 [Driveway lowres: 2000 - Analysing at 5.00 fps]
```
</details>

Any others needed?

xefil, 2020-12-05:
Same here on zoneminder 1.35.14~20201129162543-focal on Ubuntu 20.04.1 LTS (Focal Fossa).
Following.

shanepcs, 2020-12-05:
same issue with 1.35.15~20201204114937-focal.  The issue started for me with 1.35.14~20201129162543-focal.

cotman, 2020-12-05:
Ditto on linked monitors being broken - I had to reinstall due to a failed hard drive, and so started again from a recent `master' PPA (I'm currently on `1.35.15~20201204114937-focal` and, as above, that's not working with high resolution/fps Nodect monitors linked to low resolution/fps Modect monitors).

As an aside from the issue itself, it would be great if the documentation could be updated to state what the minimal requirements are for linked monitors to work correctly e.g. if the Modect monitor needs/doesn't need to have a video writer enabled, whether the Nodect monitor needs/doesn't need to have any motion detection zones etc. etc.

closed 2020-12-06 by connortechnology

johncolby, 2020-12-06:
Awesome, thanks Isaac!!!

dangelsaurus, 2021-07-24:
has this issue come up again in 1.36.5?

ferengi82, 2021-08-05:
same Problem here on 1.36.5

synsi23b, 2021-08-18:
I have the same Problem, I only noticed it, because there is a problem with my new camera. The camera needs to be restarted every once in a while, which causes the zmc_mX process to be restarted or lose its association in some way.

If the high-res linked monitor starts new first, ( I guess ) it is linking to the old low-res process and doesn't get the events from the now restarted low-res MoCord process.

The temporary fix is, to switch the NoDect linked monitor to the None function and than enable it again. Or restarting zoneminder.service entirely.

Ubuntu 20.04 lxc container

maddios, 2021-12-01:
I have the same issue apparently in the 1.36.10 also, I've not done much debugging yet, but i regularly have no recordings in my linked nodect monitors while the modect ones record tons

virnik0, 2023-11-17:
Same issue on 1.37.48

Linked monitors misses too many triggers, not even recording. ZME reacts, but as the monitor isn't recording, it doesn't analyze. Solved for now by forcing lower res on full hd monitor, and doing analysis on it. This way, records are still full hd, but viewing the monitor ses downscaled picture and analysis happens on low-res as well. Win-Win scenario. Worked fine on 1.36

commit 36b5702c (2020-12-06): Increase LinkedMonitor debugging.  Fix LinkedMonitors.  Fixes #3091