ZoneMinder/zoneminder #3093: ZoneMinder fails to write fixed length videos from multiple 1440p H264/H.265 cameras in H.264 Video Passthrough mode

lcchelpdesk, 2020-12-10:
Hi, we were using using customized compiled 1.30 version with changes from this commit:
https://github.com/ZoneMinder/ZoneMinder/pull/1367
All of our 50 cameras are H264/H.265 with Hikvision "+" optimizations. They generate storage effective streams which we need to store with ZoneMinder and view when needed. No live view or picture analysis for our ZM setup is necessary. Recording was done in H264 Video Passthrough mode. This setup served us perfectly through 3-4 years, but unfortunately two of tree really old HDD drives in the system array have failed and I decided to replace drives and reinstall new version, were H264 recording feature was advertised to be implemented in the main branch.

I can record video with the same setting with current 1.34 version, but load on CPU from 50 cameras is so large, that almost no videos get recorder. CPU usage was always 40.0 (linux). Our setup runs on dual Zeon server with 2x8 CPU's with 40GB ram. Cameras film in 10 FPS mode. I tried differing Max FPS setting to 1 instead of 11 we used earlier. This lowered CPU load to around 1-4 but it still fails to store mp4 records, perhaps it constantly bottlenecks when videos get finalized every 30min.

If 1367 commit was closed maybe this feature can be implemented or somehow activated in current version? Is there an easy way to disable decoder or at least make recording of fixed length records stable? As I mention live view is absolutely unnecessary, just record and play record.

We need to either solve this issue really fast or try to restore old version since in current quarantine time with everyone working from home recordings are absolutely necessary to have.

connortechnology, 2020-12-10:
Do not set maxfps when recording from rtsp.  There is a red warning when you edit that field for a reason.  If you want to reduce fps do it in the camera.

lcchelpdesk, 2020-12-10:
Good point, but it still not solve my unnecessary stream decoding CPU overhead problem. Is there a way to maybe somehow indirectly limit it?
Can buffer settings somehow limit decoder workload? Either way it won't show live video, since in h.265+ mode I-frame can take up to 300 frames to appear. That would eat all my 40GB of RAM within 5 cameras. It it impossible to use why not provide an option to totally disable it. I have set like this now:
[image]

I started playing with cameras and it is able to handle half of them, but htop already shows near 100% load on all cores. Running process that is totally useless...

connortechnology, 2020-12-10:
Yeah h265 is a problem.  Don't use it.  Use h264.

lcchelpdesk, 2020-12-10:
I can playback records just fine. Are you telling the if I switch all cameras back to H.264+ it will lower CPU load? Changing config for 50 cameras takes time and already makes me a bit crazy. :)

It just frustrates me the we did perfectly with modified H.264 branch of 1.30 version recording H.265+ with CPU load near zero with decoder disabled and now moving 4 years further this problem still not solved. As if making decoder module compulsory active is religion that can't be doubt.

connortechnology, 2020-12-10:
Oh I see.  You were using some experimental patch.  Look, this is coming but few people are funding it.  
You might be interested in this:
https://forums.zoneminder.com/viewtopic.php?f=9&t=27537

lcchelpdesk, 2020-12-10:
Thanks! I will take a look tomorrow, my work time has long passed for today.
I just check switching one of the cameras that was creating high load from H.265+ to H.264+ and it had absolutely no impact on CPU load. It was both ~40% on one core before and after changing.

connortechnology, 2020-12-10:
Yes 4 years of no one funding that feature.  Implementing it in a useful way that doesn't break functionality for everyone else is HARD.  Most people like live view.  Just because you don't doesn't mean that any of us should care about your niche case.  

That being said I've been working of the restructuring needed for four years and looking at it today I can probably give you this feature in a few days.

connortechnology, 2020-12-10:
h264 just lets you specify a shorted keyframe interval so we don't have to buffer as much.  There will be a slight cpu benefit but it's more about the ram.  Also the ability for browsers to play the video.

connortechnology, 2020-12-10:
I just released 1.35.16 with this feature.  works great.

lcchelpdesk, 2020-12-11:
Hi, I can't even express how happy I am about new options in latest release - 1.35.16. Absolutely solved our problem, CPU load is almost at zero all the time and records are going through rock solid.
We are non profit organization are very tight on the budget, that's why we almost always go with open source projects. And ZoneMinder was almost perfect for our needs, only lack of options to disable decoding was missing.
We are very thankful for your hard work and effort to keep this project alive!

Recording optimized versions of H.264/H.265 with variable I-frame depending of image conditions (like H.265+ or Super H.265) is very efficient in terms of saving storage space without much of quality loss, but are very cost worthy on RAM and a bit less on CPU to decode. Just a suggestion, most of the cameras have option to stream couple of streams, perhaps it would be great idea to use full high res stream for H.264 Camera Passthrough recording, and for live view decoding and analysis add additional low res source from the same camera.

closed 2020-12-16 by connortechnology