ZoneMinder/zoneminder #3094: %MN% not working in email filter

rhaamo, 2020-12-14:
**Environment**
- zoneminder_1.35.16~20201214170417-focal_amd64.deb
- do_debian_package.sh
- Ubuntu 20.04.1 LTS

**bug**
The filter template element `%MN%` is empty.

I have recently upgraded my ZM and I've found out that the monitor name is now missing from filter 'Alert Email' emails I get.

Email subject is `ZoneMinder: Alarm - -725888 (116 - 59 30)`, in the web ui: `ZoneMinder: Alarm - %MN%-%EI% (%ESM% - %ESA% %EFA%)`

Or in body:
```
 Monitor  : 
 Event Id : 725888
```
```
  Monitor  : %MN%
  Event Id : %EI%
```

In https://github.com/ZoneMinder/zoneminder/blob/master/docs/userguide/filterevents.rst the `%MN%` is sill mentioned, has it changed name or it is a bug ?

connortechnology, 2020-12-14:
It would be a bug. 

One thing: I moved the email body text into the Filter itself so that different filters can send different email content.  Please check that the body in the filter is what you want it to be.

rhaamo, 2020-12-14:
my filter looks like that

[image]

vs email

[image]

connortechnology, 2020-12-15:
I believe I have fixed the problem. Took two commits to get it right.  bdf9a029adee3cbdb8ed13b8e6b51951d6126c65 and 9fec11c4a99dc6bc9e2109d41ea4dbd35adb5f09

rhaamo, 2020-12-15:
I was just rebuilding zm after seeing them.
I will let you know when I get some alerts.

rhaamo, 2020-12-15:
thank you for the fixes !

that seems good now.

closed 2020-12-15 by rhaamo