ZoneMinder/zoneminder #3095: ZM page refresh appears to cause socket errors and broken pipe errors

cnighswonger, 2020-12-16:
**Describe Your Environment**
- Version of ZoneMinder:

1.32.3

- How you installed ZoneMinder:

https://zmrepo.zoneminder.com/debian/release

- Full name and version of OS:

Distributor ID:	Debian
Description:	Debian GNU/Linux 9.13 (stretch)
Release:	9.13
Codename:	stretch

- Browser name and version:

Firefox 78.5.0esr (64-bit)

**If the issue concerns a camera**

Camera make does not matter for this issue.

**Describe the bug**

When doing a refresh of the montage view the log fills with the following:

```
Dec 16 09:35:47 dvr zms_m11[13228]: WAR [zms_m11] [Unable to send stream frame: Broken pipe]
Dec 16 09:35:47 dvr web_php[12742]: ERR [Socket /var/run/zm/zms-208035s.sock does not exist.  This file is created by zms, and since it does not exist, either zms did not run, or zms exited early.  Please check your zms logs and ensure that CGI is enabled in apache and check that the PATH_ZMS is set correctly.  Make sure that ZM is actually recording.  If you are trying to view a live stream and the capture process (zmc) is not running then zms will exit. Please go to http://zoneminder.readthedocs.io/en/latest/faq.html#why-can-t-i-see-streamed-images-when-i-can-see-stills-in-the-zone-window-etc for more information.]
Dec 16 09:35:47 dvr web_php[12742]: ERR [Socket /var/run/zm/zms-065537s.sock does not exist.  This file is created by zms, and since it does not exist, either zms did not run, or zms exited early.  Please check your zms logs and ensure that CGI is enabled in apache and check that the PATH_ZMS is set correctly.  Make sure that ZM is actually recording.  If you are trying to view a live stream and the capture process (zmc) is not running then zms will exit. Please go to http://zoneminder.readthedocs.io/en/latest/faq.html#why-can-t-i-see-streamed-images-when-i-can-see-stills-in-the-zone-window-etc for more information.]
Dec 16 09:35:48 dvr zms_m11[13238]: WAR [zms_m11] [Unable to send stream frame: Broken pipe]
Dec 16 09:35:57 dvr web_php[13096]: ERR [Socket /var/run/zm/zms-960730s.sock does not exist.  This file is created by zms, and since it does not exist, either zms did not run, or zms exited early.  Please check your zms logs and ensure that CGI is enabled in apache and check that the PATH_ZMS is set correctly.  Make sure that ZM is actually recording.  If you are trying to view a live stream and the capture process (zmc) is not running then zms will exit. Please go to http://zoneminder.readthedocs.io/en/latest/faq.html#why-can-t-i-see-streamed-images-when-i-can-see-stills-in-the-zone-window-etc for more information.]
```

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the montage view.
2. Refresh the page.
3. Check the log.
4. See the flood of such entries.

**Expected behavior**

Page refreshes should not flood the log. It seems that ZM should handle such incidents more gracefully.

**Debug Logs**

Let me know specifically, and I will post them.

connortechnology, 2020-12-16:
1.32.3 is old and no longer supported.  Update to 1.34 series and if you still have the problem you can reopen this.

closed 2020-12-16 by connortechnology

cnighswonger, 2020-12-16:
> 1.32.3 is old and no longer supported. Update to 1.34 series and if you still have the problem you can reopen this.

root@dvr:~# apt upgrade zoneminder
Reading package lists... Done
Building dependency tree       
Reading state information... Done
zoneminder is already the newest version (1.32.3-stretch).
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
root@dvr:~# 

https://zmrepo.zoneminder.com/debian/release does not appear to contain the latest release.

I would prefer to use the package manager to maintain my production box.

cnighswonger, 2020-12-16:
> > 1.32.3 is old and no longer supported. Update to 1.34 series and if you still have the problem you can reopen this.
> 
> https://zmrepo.zoneminder.com/debian/release does not appear to contain the latest release.
> 

Got it.

https://zmrepo.zoneminder.com/debian/release-1.34

Putting it here for posterity.

Thanks.

cnighswonger, 2020-12-16:
Okay. The flooding is not as bad with 1.34, but we still get this sort of thing:

```
Dec 16 11:34:13 dvr web_php[31439]: ERR [Socket /var/run/zm/zms-210802s.sock does not exist.  This file is created by zms, and since it does not exist, either zms did not run, or zms exited early.  Please check your zms logs and ensure that CGI is enabled in apache and check that the PATH_ZMS is set correctly.  Make sure that ZM is actually recording.  If you are trying to view a live stream and the capture process (zmc) is not running then zms will exit. Please go to http://zoneminder.readthedocs.io/en/latest/faq.html#why-can-t-i-see-streamed-images-when-i-can-see-stills-in-the-zone-window-etc for more information.]
```
I'm assuming that the back end has no idea what caused the socket to vanish and throws a warning anyway.

Maybe the solution is just to lower the log level.

connortechnology, 2020-12-16:
The error means that the frontend js is trying to connect to the backend php which is trying to connect to zms and it just isn't there.  

Why isn't it there?  A healthy system should not generate these errors very often.