ZoneMinder/zoneminder #3097: No Live streaming  debian 10

ptroms, 2020-12-19:
System details:
Debian 10 4.19.0-13-amd64
mysql  Ver 15.1 Distrib 10.3.27-MariaDB
PHP Version 7.3.19-1
Zoneminder v1.34.22
Broswer(s):Opera,Version:72.0.3815.400 | Firefox 84.0
---
I have fresh zoneminder installed on my support server. Over the last few day i had spend sometime play around with it. It is capturing the cameras stream, save it, played back, all functionality in therms of streaming and the connections form cameras into the server has no problem at all. 
I have to say that, there is no configuration error.
When zoneminder start/restarts i have "the classic" "can't exec no such file or directory".
--
logs
--
zmdc.log

> 19/12/20 14:32:26.712699 zmdc[31615].INF [ZMServer:411] ['zmeventnotification.pl' starting at 20/12/19 14:32:26, pid = 31670]
> 19/12/20 14:32:26.713441 zmdc[31670].INF [ZMServer:411] ['zmeventnotification.pl' started at 20/12/19 14:32:26]
> Can't exec "/usr/bin/zmeventnotification.pl": No such file or directory at /usr/bin/zmdc.pl line 503, <CLIENT> line 1.
> 19/12/20 14:32:26.723905 zmdc[31670].DBG [ZoneMinder::Logger:294] [LogOpts: level=DBG/DBG, screen=OFF, database=INF, logfile=DBG->/var/log/zm/zmdc.log, syslog=INF]
> 19/12/20 14:32:26.724312 zmdc[31670].FAT [ZMServer:503] [Can't exec: No such file or directory]
> 19/12/20 14:32:26.734527 zmdc[31615].ERR [ZMServer:706] ['zmeventnotification.pl' exited abnormally, exit status 255]
> Update agent starting at 20/12/19 14:32:26
> 19/12/20 14:32:26.909699 zmdc[31615].INF [ZMServer:411] ['zmstats.pl' starting at 20/12/19 14:32:26, pid = 31674]
> 19/12/20 14:32:26.909790 zmdc[31674].INF [ZMServer:411] ['zmstats.pl' started at 20/12/19 14:32:26]
> 19/12/20 14:32:26.915008 zmdc[31615].INF [ZMServer:411] [Starting pending process, zmeventnotification.pl]
> 19/12/20 14:32:26.922874 zmdc[31615].INF [ZMServer:411] ['zmeventnotification.pl' starting at 20/12/19 14:32:26, pid = 31675]
> 19/12/20 14:32:26.922879 zmdc[31675].INF [ZMServer:411] ['zmeventnotification.pl' started at 20/12/19 14:32:26]
> Can't exec "/usr/bin/zmeventnotification.pl": No such file or directory at /usr/bin/zmdc.pl line 503.
> 19/12/20 14:32:26.933027 zmdc[31675].DBG [ZoneMinder::Logger:294] [LogOpts: level=DBG/DBG, screen=OFF, database=INF, logfile=DBG->/var/log/zm/zmdc.log, syslog=INF]
> 19/12/20 14:32:26.933300 zmdc[31675].FAT [ZMServer:503] [Can't exec: No such file or directory]
> 19/12/20 14:32:26.942268 zmdc[31615].ERR [ZMServer:706] ['zmeventnotification.pl' exited abnormally, exit status 255]
> 19/12/20 14:32:31.950801 zmdc[31615].INF [ZMServer:411] [Starting pending process, zmeventnotification.pl]
> 19/12/20 14:32:31.958649 zmdc[31615].INF [ZMServer:411] ['zmeventnotification.pl' starting at 20/12/19 14:32:31, pid = 31677]
> 19/12/20 14:32:31.958657 zmdc[31677].INF [ZMServer:411] ['zmeventnotification.pl' started at 20/12/19 14:32:31]
> Can't exec "/usr/bin/zmeventnotification.pl": No such file or directory at /usr/bin/zmdc.pl line 503.
--
web_php.log

> 12/19/20 14:32:37.261267 web_php[25548].DBG [192.168.1.2] [View: request Request: status Action:  User: admin] at /usr/share/zoneminder/www/index.php line 213
> 12/19/20 14:32:38.235714 web_php[24858].ERR [192.168.1.2] [Socket /run/zm/zms-456467s.sock does not exist.  This file is created by zms, and since it does not exist, either zms did not run, or zms exited early.  Please check your zms logs and ensu$
> 12/19/20 14:32:38.247648 web_php[24858].DBG [192.168.1.2] [Setting cookie parameters to lifetime(3600) path(/) domain () secure() httpOnly(1) name:ZMSESSID] at /usr/share/zoneminder/www/includes/session.php line 20
> 12/19/20 14:32:38.251525 web_php[25548].DBG [192.168.1.2] [Setting cookie parameters to lifetime(3600) path(/) domain () secure() httpOnly(1) name:ZMSESSID] at /usr/share/zoneminder/www/includes/session.php line 20
> 12/19/20 14:32:38.255554 web_php[24858].DBG [192.168.1.2] [View: request Request: log Action:  User: admin] at /usr/share/zoneminder/www/index.php line 213
> 12/19/20 14:32:38.258239 web_php[25548].DBG [192.168.1.2] [View: watch Request:  Action:  User: admin] at /usr/share/zoneminder/www/index.php line 213
> 12/19/20 14:32:39.542953 web_php[25548].DBG [192.168.1.2] [Setting cookie parameters to lifetime(3600) path(/) domain () secure() httpOnly(1) name:ZMSESSID] at /usr/share/zoneminder/www/includes/session.php line 20
> 12/19/20 14:32:39.548077 web_php[25548].DBG [192.168.1.2] [View: request Request: stream Action:  User: admin] at /usr/share/zoneminder/www/index.php line 213
> 12/19/20 14:32:39.551495 web_php[25548].DBG [192.168.1.2] [Sending command 99] at /usr/share/zoneminder/www/ajax/stream.php line 56
> 12/19/20 14:32:39.601440 web_php[24858].DBG [192.168.1.2] [Setting cookie parameters to lifetime(3600) path(/) domain () secure() httpOnly(1) name:ZMSESSID] at /usr/share/zoneminder/www/includes/session.php line 20
> 12/19/20 14:32:39.605828 web_php[24858].DBG [192.168.1.2] [View: request Request: status Action:  User: admin] at /usr/share/zoneminder/www/index.php line 213
> 12/19/20 14:32:40.665570 web_php[25548].ERR [192.168.1.2] [Socket /run/zm/zms-259389s.sock does not exist.  This file is created by zms, and since it does not exist, either zms did not run, or zms exited early.  Please check your zms logs and ensu$
> 12/19/20 14:32:40.677828 web_php[25548].DBG [192.168.1.2] [Setting cookie parameters to lifetime(3600) path(/) domain () secure() httpOnly(1) name:ZMSESSID] at /usr/share/zoneminder/www/includes/session.php line 20
> 12/19/20 14:32:40.681906 web_php[24858].DBG [192.168.1.2] [Setting cookie parameters to lifetime(3600) path(/) domain () secure() httpOnly(1) name:ZMSESSID] at /usr/share/zoneminder/www/includes/session.php line 20
> 12/19/20 14:32:40.685175 web_php[25548].DBG [192.168.1.2] [View: request Request: log Action:  User: admin] at /usr/share/zoneminder/www/index.php line 213
> 12/19/20 14:32:40.686611 web_php[24858].DBG [192.168.1.2] [View: watch Request:  Action:  User: admin] at /usr/share/zoneminder/www/index.php line 213
> 12/19/20 14:32:41.922501 web_php[24858].DBG [192.168.1.2] [Setting cookie parameters to lifetime(3600) path(/) domain () secure() httpOnly(1) name:ZMSESSID] at /usr/share/zoneminder/www/includes/session.php line 20
> 12/19/20 14:32:41.928529 web_php[24858].DBG [192.168.1.2] [View: request Request: status Action:  User: admin] at /usr/share/zoneminder/www/index.php line 213
> 12/19/20 14:32:41.986289 web_php[24858].DBG [192.168.1.2] [Setting cookie parameters to lifetime(3600) path(/) domain () secure() httpOnly(1) name:ZMSESSID] at /usr/share/zoneminder/www/includes/session.php line 20
> 12/19/20 14:32:41.990527 web_php[24858].DBG [192.168.1.2] [View: request Request: stream Action:  User: admin] at /usr/share/zoneminder/www/index.php line 213
> 12/19/20 14:32:41.994210 web_php[24858].DBG [192.168.1.2] [Sending command 99] at /usr/share/zoneminder/www/ajax/stream.php line 56
> 12/19/20 14:32:43.110683 web_php[24858].ERR [192.168.1.2] [Socket /run/zm/zms-744502s.sock does not exist.  This file is created by zms, and since it does not exist, either zms did not run, or zms exited early.  Please check your zms logs and ensu$
> 12/19/20 14:32:43.122371 web_php[24858].DBG [192.168.1.2] [Setting cookie parameters to lifetime(3600) path(/) domain () secure() httpOnly(1) name:ZMSESSID] at /usr/share/zoneminder/www/includes/session.php line 20
--
zmfilter_1.log

> 19/12/20 14:32:26.306981 zmfilter_1[31650].DBG [ZoneMinder::Logger:294] [LogOpts: level=DBG/DBG, screen=OFF, database=DBG, logfile=DBG->/var/log/zm/zmfilter_1.log, syslog=DBG]
> 19/12/20 14:32:26.316136 zmfilter_1[31650].INF [main:180] [Scanning for events using filter id '1']
> 19/12/20 14:32:26.319180 zmfilter_1[31650].DBG [main:199] [Reloading filters]
> 19/12/20 14:32:26.323785 zmfilter_1[31650].DBG [main:255] [Found filter 'PurgeWhenFull']
> 19/12/20 14:32:26.351738 zmfilter_1[31650].DBG [main:269] [Got 1 filters]
> 19/12/20 14:32:26.374623 zmfilter_1[31650].DBG [ZoneMinder::Filter:113] [Filter::Execute SQL (SELECT E.*,
[... 7 lines cut ...]
>           WHERE (E.EndTime IS NOT NULL) AND ( E.Archived = 0 and 17 >= 95 ) ORDER BY E.Id ASC LIMIT 0,100)]
> 19/12/20 14:32:26.377863 zmfilter_1[31650].DBG [ZoneMinder::Filter:127] [Loaded 0 events for filter PurgeWhenFull using query (SELECT E.*,
>        unix_timestamp(E.StartTime) as Time,
>        M.Name as MonitorName,
>        M.DefaultRate,
>        M.DefaultScale
>          FROM Events as E
>          INNER JOIN Monitors as M on M.Id = E.MonitorId
[... cut ...]
> 2020-12-19 14:33:27 4831 [Warning] Access denied for user 'www-data'@'localhost' (using password: NO)
> 2020-12-19 14:33:40 4746 [Warning] Aborted connection 4746 to db: 'zm' user: 'friend456' host: 'localhost' (Got an error reading communication packets)
> 2020-12-19 14:33:40 4741 [Warning] Aborted connection 4741 to db: 'zm' user: 'friend456' host: 'localhost' (Got an error reading communication packets)
> 2020-12-19 14:33:40 4756 [Warning] Aborted connection 4756 to db: 'zm' user: 'friend456' host: 'localhost' (Got an error reading communication packets)
--

ptroms, 2020-12-19:
One more important log.
zmwatch.log

> 19/12/20 14:32:26.455568 zmwatch[31660].DBG [ZoneMinder::Logger:294] [LogOpts: level=DBG/DBG, screen=OFF, database=DBG, logfile=DBG->/var/log/zm/zmwatch.log, syslog=DBG]
> 19/12/20 14:32:26.463195 zmwatch[31660].INF [main:72] [Watchdog starting, pausing for 30 seconds]
> 19/12/20 14:32:56.471708 zmwatch[31660].DBG [ZoneMinder::Memory:300] [Shared data appears valid for monitor 1: 1]
> 19/12/20 14:32:56.475757 zmwatch[31660].DBG [main:103] [Monitor 1 LastWriteTime is 1608388376.]
> 19/12/20 14:32:56.479240 zmwatch[31660].DBG [main:127] [Monitor 1 last captured 0 seconds ago, max is 45]
> 19/12/20 14:33:06.486273 zmwatch[31660].DBG [ZoneMinder::Memory:300] [Shared data appears valid for monitor 1: 1]
> 19/12/20 14:33:06.490163 zmwatch[31660].DBG [main:103] [Monitor 1 LastWriteTime is 0.]
> 19/12/20 14:33:16.493554 zmwatch[31660].DBG [ZoneMinder::Memory:300] [Shared data appears valid for monitor 1: 1]
> 19/12/20 14:33:16.497453 zmwatch[31660].DBG [main:103] [Monitor 1 LastWriteTime is 1608388396.]
> 19/12/20 14:33:16.500503 zmwatch[31660].DBG [main:127] [Monitor 1 last captured 0 seconds ago, max is 45]
> 19/12/20 14:33:16.503691 zmwatch[31660].DBG [main:174] [Monitor 1 last analysed 0 seconds ago, max is 45]
> 19/12/20 14:33:26.508060 zmwatch[31660].ERR [ZoneMinder::Memory::Mapped:90] [Memory map file '/dev/shm/zm.mmap.1' does not exist in zmMemAttach.  zmc might not be running.]
> 19/12/20 14:33:26.516342 zmwatch[31660].INF [main:136] [Restarting capture daemon for Front, shared data not valid]
> 19/12/20 14:33:26.521722 zmwatch[31660].DBG [ZoneMinder::General:67] [Testing valid shell syntax]
> 19/12/20 14:33:26.528025 zmwatch[31660].DBG [ZoneMinder::General:71] [Running as 'www-data', su commands not needed]
> 19/12/20 14:33:26.531169 zmwatch[31660].DBG [ZoneMinder::General:144] [Command: /usr/bin/zmdc.pl restart zmc -m 1]
> 19/12/20 14:33:26.739537 zmwatch[31660].DBG [ZoneMinder::General:152] [Output: Restarting zmc -m 1
> 'zmc -m 1' sending stop to pid 31745 at 20/12/19 14:33:26]
> 19/12/20 14:33:36.744373 zmwatch[31660].DBG [ZoneMinder::Memory:300] [Shared data appears valid for monitor 1: 1]
> 19/12/20 14:33:36.747659 zmwatch[31660].DBG [main:103] [Monitor 1 LastWriteTime is 1608388416.]
> 19/12/20 14:33:36.750369 zmwatch[31660].DBG [main:127] [Monitor 1 last captured 0 seconds ago, max is 45]
For any more specific details, or questions post it down please.
reg
p

connortechnology, 2020-12-19:
You havn't installed the event server.  https://zmeventnotification.readthedocs.io/en/latest/guides/install.html

You told it you did so it is trying to start it. However you did not, so you get an error.

connortechnology, 2020-12-19:
Of course that has nothing to do with a failure to stream.  The failure to stream is likely caused by an apache misconfig.  The steps to fix that are found in the very error message that you  have posted here and have clearly not followed.

ptroms, 2020-12-20:
Thank you for quick reply.
You have mention that the "ERR getStreamCmdResponse stream error socket" is related with some apache2 library. I have check the recent list of debian 10 buster-proposed-updates but there is no apache2 release yet. 
I am not quite sure how the the development/solving issue works. Could you elaborate me how, or here can I get the solution(s) from please?
Currently I have a system upgrade mostly for high-end 4K resolution IP-CAM. I have tested most of then and they are running very well with ZM. So as a part of my contribution I would write some installation guidance for your wiki website.
By the way the "event server (ES)" works great. Jason supports zmninja app as well. 
Good job guys.

connortechnology, 2020-12-21:
not apache library problem.  apache CONFIG problem.  You havn't correctly configured apache, which is the web server.  

Please see the following for the sections relating to zms.
https://zoneminder.readthedocs.io/en/latest/installationguide/debian.html?highlight=zms#easy-way-debian-jessie

ptroms, 2020-12-21:
This is pretty much what I have done in the first place. 
It is worth to mention that I have had a a service running on port tcp 9k, during the installation process. But now is running on different port and port 9000 is allocated to the ZM stuff. 
So what would be the best way of dealing with the problem. Upgrade apache2, or wait to next version of ZM, or some other solution may exist. I m happy to test it.
regards
p

connortechnology, 2021-04-20:
Either read the docs again, good the problem for everyone else discussing and fixing the same problem or use paid support to have us fix your system for you.
Updating will not fix your problem, it is a simple misconfig.

closed 2021-04-20 by connortechnology