ZoneMinder/zoneminder #3098: Update Password function for user

pkolmann, 2020-12-22:
- Version of ZoneMinder: v1.34.22
- How you installed ZoneMinder: via deb from https://zmrepo.zoneminder.com/debian/release-1.34 buster/
- Full name and version of OS: Debian Buster amd64

In Zoneminder you can set a password per user as admin. There is no UI for the user to change his own password. It would be nice for a user to able to change his/her password when needed without the admin.

**To Reproduce**
Steps to reproduce the behavior:
1. Use ZM Authenitcation
2. Login as User
3. search for a possibility to change the password.

**Expected behavior**
have the possibility to change my own user password via Web Page

knight-of-ni, 2020-12-22:
Tthanks for reporting. This functionality already exists with the ZM_USER_SELF_EDIT under Options -> Config.

pkolmann, 2020-12-28:
Thanks @knight-of-ni I haven't seen this option. My bad...