ZoneMinder/zoneminder #3104: Local camera : Device or ressource busy

TP234-CPE, 2021-01-02:
**Environment**
- ZoneMinder Version: 1.34.22-buster1 installed from-source
- Debian GNU/Linux 10

**Describe the bug**
I use analog cameras with a 4ch capture card using the tw68 driver (2 devices video0&1 both having 2 channels 0&1) . When I set up the monitor with source type Local I get a fatal error Device or ressource busy, whatever palett color, video format PAL/NTCS, video size I put. I also tried with different brand of cameras nothing seems to work beside using LibVLC as a source type to use a VLC stream as a workaround but it can only work with one channel per device. I also found others pretty similar closed issue  #2406, #733, and #2405 but with no solutions. I also read that some people didn't had this issue with the kernel 3 and before as they had to manually install the driver. Starting with kernel 4, the tw68 driver was rewritten to be directly into inside. 

**To Reproduce**
Using a tw68 analog card with any analog camera and a linux kernel in version 4 minimum.

**Expected behavior**
Zoneminder to access the video stream of the camera.

**Debug Logs**
On the web interface :
```
2021-01-02 19:18:10	zmc_dvideo0		9248	ERR	Zone 11/All for monitor Monitor-11 extends outside of image dimensions, (0,0), (751,581), ignoring	zm_zone.cpp	926
2021-01-02 19:18:10	zmc_dvideo0		9248	FAT	Failed to set video standard 255: 16 Device or resource busy	zm_local_camera.cpp	957
2021-01-02 19:18:10	zmdc		2030	ERR	'zmc -d /dev/video0' exited abnormally, exit status 255	zmdc.pl	
2021-01-02 19:17:30	zmc_dvideo0		9246	ERR	Zone 11/All for monitor Monitor-11 extends outside of image dimensions, (0,0), (751,581), ignoring	zm_zone.cpp	926
2021-01-02 19:17:30	zmc_dvideo0		9246	FAT	Failed to set video standard 255: 16 Device or resource busy	zm_local_camera.cpp	957
2021-01-02 19:17:30	zmdc		2030	ERR	'zmc -d /dev/video0' exited abnormally, exit status 255	zmdc.pl	
2021-01-02 19:17:10	zmc_dvideo0		9245	ERR	Zone 11/All for monitor Monitor-11 extends outside of image dimensions, (0,0), (751,581), ignoring	zm_zone.cpp	926
2021-01-02 19:17:10	zmc_dvideo0		9245	FAT	Failed to set video standard 255: 16 Device or resource busy	zm_local_camera.cpp	957
2021-01-02 19:17:10	zmdc		2030	ERR	'zmc -d /dev/video0' exited abnormally, exit status 255	zmdc.pl	
2021-01-02 19:17:00	zmc_dvideo0		9240	ERR	Zone 11/All for monitor Monitor-11 extends outside of image dimensions, (0,0), (751,581), ignoring	zm_zone.cpp	926
2021-01-02 19:17:00	zmc_dvideo0		9240	FAT	Failed to set video standard 255: 16 Device or resource busy	zm_local_camera.cpp	957
2021-01-02 19:17:00	zmdc		2030	ERR	'zmc -d /dev/video0' exited abnormally, exit status 255	zmdc.pl	
2021-01-02 19:16:54	zmwatch		2123	WAR	Restarting capture daemon for Monitor-11, no image since startup. Startup time was 1609611037 - now 1609611414 > 45	zmwatch.pl	
2021-01-02 19:16:45	zmc_dvideo0		9233	DB1	LogOpts: level=DB9 effective=DB9, screen=OFF, database=DB9, logfile=DB9->/tmp/zm/loglevel9, syslog=DB9	zm_logger.cpp	255
2021-01-02 19:16:45	zmc_dvideo0		9233	DB1	Detected a x86\x86-64 processor with SSE4.2	zm_utils.cpp	275
2021-01-02 19:16:44	zmwatch		2123	WAR	Restarting capture daemon for Monitor-11, no image since startup. Startup time was 1609611037 - now 1609611404 > 45
```
This is all I know, hope someone will have an idea of what i could do. I join a file with a part of my debug log level 9, tell me what I could add to help you understanding what is happening.

TP234-CPE, 2021-01-03:
Couldn't upload the txt so there is a link to it so there is a link to it: https://gofile.io/d/wUz6E8

TP234-CPE, 2021-02-16:
Since the  link died I'm reposting a part of the log level 9 file  : 
`
02/16/21 17:01:06.444988 zmc_dvideo0[3486].DB1-zm_logger.cpp/255 [LogOpts: level=DB9 effective=DB9, screen=OFF, database=WAR, logfile=DB9->/tmp/zm/loglevel9, syslog=DB9]
02/16/21 17:01:06.445358 zmc_dvideo0[3486].DB1-zm_utils.cpp/275 [Detected a x86\x86-64 processor with SSE4.2]
02/16/21 17:01:06.445397 zmc_dvideo0[3486].DB1-zm_monitor.cpp/1996 [Loading Monitors with SELECT `Id`, `Name`, `ServerId`, `StorageId`, `Type`, `Function`+0, `Enabled`, `LinkedMonitors`, `AnalysisFPSLimit`, `AnalysisUpdateDelay`, `MaxFPS`, `AlarmMaxFPS`,`Device`, `Channel`, `Format`, `V4LMultiBuffer`, `V4LCapturesPerFrame`, `Protocol`, `Method`, `Options`, `User`, `Pass`, `Host`, `Port`, `Path`, `Width`, `Height`, `Colours`, `Palette`, `Orientation`+0, `Deinterlacing`, `DecoderHWAccelName`, `DecoderHWAccelDevice`, `RTSPDescribe`, `SaveJPEGs`, `VideoWriter`, `EncoderParameters`, `RecordAudio`, `Brightness`, `Contrast`, `Hue`, `Colour`, `EventPrefix`, `LabelFormat`, `LabelX`, `LabelY`, `LabelSize`,`ImageBufferCount`, `WarmupCount`, `PreEventCount`, `PostEventCount`, `StreamReplayBuffer`, `AlarmFrameCount`, `SectionLength`, `MinSectionLength`, `FrameSkip`, `MotionFrameSkip`, `FPSReportInterval`, `RefBlendPerc`, `AlarmRefBlendPerc`, `TrackMotion`, `Exif`, `SignalCheckPoints`, `SignalCheckColour` FROM `Monitors` WHERE `Function` != 'None' AND `Type` = 'Local' AND `Device`='/dev/video0']
02/16/21 17:01:06.445511 zmc_dvideo0[3486].DB4-zm_db.cpp/112 [Success running query: SELECT `Id`, `Name`, `ServerId`, `StorageId`, `Type`, `Function`+0, `Enabled`, `LinkedMonitors`, `AnalysisFPSLimit`, `AnalysisUpdateDelay`, `MaxFPS`, `AlarmMaxFPS`,`Device`, `Channel`, `Format`, `V4LMultiBuffer`, `V4LCapturesPerFrame`, `Protocol`, `Method`, `Options`, `User`, `Pass`, `Host`, `Port`, `Path`, `Width`, `Height`, `Colours`, `Palette`, `Orientation`+0, `Deinterlacing`, `DecoderHWAccelName`, `DecoderHWAccelDevice`, `RTSPDescribe`, `SaveJPEGs`, `VideoWriter`, `EncoderParameters`, `RecordAudio`, `Brightness`, `Contrast`, `Hue`, `Colour`, `EventPrefix`, `LabelFormat`, `LabelX`, `LabelY`, `LabelSize`,`ImageBufferCount`, `WarmupCount`, `PreEventCount`, `PostEventCount`, `StreamReplayBuffer`, `AlarmFrameCount`, `SectionLength`, `MinSectionLength`, `FrameSkip`, `MotionFrameSkip`, `FPSReportInterval`, `RefBlendPerc`, `AlarmRefBlendPerc`, `TrackMotion`, `Exif`, `SignalCheckPoints`, `SignalCheckColour` FROM `Monitors` WHERE `Function` != 'None' AND `Type` = 'Local' AND `Device`='/dev/video0']
02/16/21 17:01:06.445535 zmc_dvideo0[3486].DB1-zm_monitor.cpp/2004 [Got 1 monitors]
02/16/21 17:01:06.445563 zmc_dvideo0[3486].DB2-zm_camera.cpp/56 [New camera id: 11 width: 752 line size: 3008 height: 582 colours: 4 subpixelorder: 8 capture: 1]
02/16/21 17:01:06.445586 zmc_dvideo0[3486].DB2-zm_local_camera.cpp/346 [V4L support enabled, using V4L2 api]
02/16/21 17:01:06.445602 zmc_dvideo0[3486].DB2-zm_local_camera.cpp/365 [little-endian processor detected]
02/16/21 17:01:06.445615 zmc_dvideo0[3486].DB2-zm_local_camera.cpp/377 [Using automatic format selection]
02/16/21 17:01:06.445648 zmc_dvideo0[3486].DB3-zm_local_camera.cpp/1219 [Got format: 16-bit A/XRGB 1-5-5-5 (0x4f424752) at index 0]
02/16/21 17:01:06.445665 zmc_dvideo0[3486].DB3-zm_local_camera.cpp/1219 [Got format: 16-bit A/XRGB 1-5-5-5 BE (0x51424752) at index 1]
02/16/21 17:01:06.445679 zmc_dvideo0[3486].DB3-zm_local_camera.cpp/1219 [Got format: 16-bit RGB 5-6-5 (0x50424752) at index 2]
02/16/21 17:01:06.445693 zmc_dvideo0[3486].DB3-zm_local_camera.cpp/1219 [Got format: 16-bit RGB 5-6-5 BE (0x52424752) at index 3]
02/16/21 17:01:06.445706 zmc_dvideo0[3486].DB3-zm_local_camera.cpp/1219 [Got format: 24-bit BGR 8-8-8 (0x33524742) at index 4]
02/16/21 17:01:06.445720 zmc_dvideo0[3486].DB3-zm_local_camera.cpp/1219 [Got format: 24-bit RGB 8-8-8 (0x33424752) at index 5]
02/16/21 17:01:06.445733 zmc_dvideo0[3486].DB3-zm_local_camera.cpp/1219 [Got format: 32-bit BGRA/X 8-8-8-8 (0x34524742) at index 6]
02/16/21 17:01:06.445746 zmc_dvideo0[3486].DB3-zm_local_camera.cpp/1219 [Got format: 32-bit A/XRGB 8-8-8-8 (0x34424752) at index 7]
02/16/21 17:01:06.445760 zmc_dvideo0[3486].DB3-zm_local_camera.cpp/1219 [Got format: YUYV 4:2:2 (0x56595559) at index 8]
02/16/21 17:01:06.445783 zmc_dvideo0[3486].DB3-zm_local_camera.cpp/1219 [Got format: UYVY 4:2:2 (0x59565955) at index 9]
02/16/21 17:01:06.445799 zmc_dvideo0[3486].DB6-zm_local_camera.cpp/1254 [No match for format: 16-bit A/XRGB 1-5-5-5 (0x5247424f) at index 0]
02/16/21 17:01:06.445812 zmc_dvideo0[3486].DB6-zm_local_camera.cpp/1254 [No match for format: 16-bit A/XRGB 1-5-5-5 BE (0x52474251) at index 1]
02/16/21 17:01:06.445825 zmc_dvideo0[3486].DB6-zm_local_camera.cpp/1254 [No match for format: 16-bit RGB 5-6-5 (0x52474250) at index 2]
02/16/21 17:01:06.445837 zmc_dvideo0[3486].DB6-zm_local_camera.cpp/1254 [No match for format: 16-bit RGB 5-6-5 BE (0x52474252) at index 3]
02/16/21 17:01:06.445849 zmc_dvideo0[3486].DB6-zm_local_camera.cpp/1254 [No match for format: 24-bit BGR 8-8-8 (0x42475233) at index 4]
02/16/21 17:01:06.445861 zmc_dvideo0[3486].DB6-zm_local_camera.cpp/1254 [No match for format: 24-bit RGB 8-8-8 (0x52474233) at index 5]
02/16/21 17:01:06.445874 zmc_dvideo0[3486].DB6-zm_local_camera.cpp/1248 [Choosing format: 32-bit BGRA/X 8-8-8-8 (0x42475234) at index 6]
02/16/21 17:01:06.445890 zmc_dvideo0[3486].INF-zm_local_camera.cpp/385 [Selected capture palette: 32-bit BGRA/X 8-8-8-8 (0x34524742)]
[... cut ...]
02/16/21 17:02:06.841168 zmc_dvideo0[3497].DB4-zm_image.cpp/364 [Delta: Using SSSE3 delta functions]
02/16/21 17:02:06.841182 zmc_dvideo0[3497].DB4-zm_image.cpp/465 [Deinterlace: Using standard functions]
02/16/21 17:02:06.841195 zmc_dvideo0[3497].DB4-zm_image.cpp/478 [Image buffer copy: Using standard memcpy]
02/16/21 17:02:06.841225 zmc_dvideo0[3497].DB1-zm_monitor.cpp/382 [encoder params # Lines beginning with # are a comment 
# For changing quality, use the crf option
# 1 is best, 51 is worst quality
#crf=23]

`

connortechnology, 2021-02-17:
We are likely not going to be able to fix this without access to this card to test against.  Would that be possible?

In the meantime, I have changed the code in master to make this a non-fatal error.  So please update and report.

TP234-CPE, 2021-02-17:
Thank you for taking time ! Yes that is doable, how do you want to proceed?

TP234-CPE, 2021-02-17:
I updated the software I can now access the video stream even if the image is a bit shaky and glitchy but the driver tw68 generates tons of system error messages (it was doing the same when I used LibVLC as source so probably a tw68 issue) :
```
[ 7067.100538] tw68 0000:03:05.0: PABORT interrupt
[ 7067.107737] tw68 0000:03:04.0: PABORT interrupt
[ 7067.108057] tw68 0000:03:05.0: PABORT interrupt
```
I still have the error but not fatal as it was planned :
```
2021-02-17 19:39:12	zmc_dvideo0		27797	ERR	Failed to set video standard 255: 16 Device or resource busy]	zm_local_camera.cpp	952
2021-02-17 19:39:12	zmdc		25572	WAR	Can't find process with command of 'zmc -d /dev/video0'	zmdc.pl
```

The real issue is that I still cannot use 2 cameras on the the same device but using different channel. When I try to do so, none of them are working, I have this new fatal error "[Unable to initialise memory mapping: Device or resource busy]" and I get this logs (level 9) :

```
02/17/21 22:03:21.730335 zmc_dvideo0[5560].DB9-zm_image.cpp/2526 [475 - 1: min_y: 0, max_y:479, min_x:639.00, 1/m:0.00]
02/17/21 22:03:21.730353 zmc_dvideo0[5560].DB9-zm_image.cpp/2526 [476 - 0: min_y: 0, max_y:479, min_x:0.00, 1/m:-0.00]
02/17/21 22:03:21.730366 zmc_dvideo0[5560].DB9-zm_image.cpp/2526 [476 - 1: min_y: 0, max_y:479, min_x:639.00, 1/m:0.00]
02/17/21 22:03:21.730383 zmc_dvideo0[5560].DB9-zm_image.cpp/2526 [477 - 0: min_y: 0, max_y:479, min_x:0.00, 1/m:-0.00]
02/17/21 22:03:21.730396 zmc_dvideo0[5560].DB9-zm_image.cpp/2526 [477 - 1: min_y: 0, max_y:479, min_x:639.00, 1/m:0.00]
02/17/21 22:03:21.730413 zmc_dvideo0[5560].DB9-zm_image.cpp/2526 [478 - 0: min_y: 0, max_y:479, min_x:0.00, 1/m:-0.00]
02/17/21 22:03:21.730426 zmc_dvideo0[5560].DB9-zm_image.cpp/2526 [478 - 1: min_y: 0, max_y:479, min_x:639.00, 1/m:0.00]
02/17/21 22:03:21.730442 zmc_dvideo0[5560].DB9-zm_image.cpp/2567 [Deleting active_edge]
02/17/21 22:03:21.730454 zmc_dvideo0[5560].DB9-zm_image.cpp/2567 [Deleting active_edge]
02/17/21 22:03:21.731118 zmc_dvideo0[5560].DB2-zm_camera.cpp/64 [New camera id: 14 width: 640 line size: 640 height: 480 colours: 1 subpixelorder: 2 capture: 1]
02/17/21 22:03:21.731141 zmc_dvideo0[5560].DB2-zm_local_camera.cpp/359 [little-endian processor detected]
02/17/21 22:03:21.731154 zmc_dvideo0[5560].DB2-zm_local_camera.cpp/371 [Using automatic format selection]
02/17/21 22:03:21.731180 zmc_dvideo0[5560].DB3-zm_local_camera.cpp/1214 [Got format: 16-bit A/XRGB 1-5-5-5 (0x4f424752) at index 0]
02/17/21 22:03:21.731197 zmc_dvideo0[5560].DB3-zm_local_camera.cpp/1214 [Got format: 16-bit A/XRGB 1-5-5-5 BE (0x51424752) at index 1]
02/17/21 22:03:21.731212 zmc_dvideo0[5560].DB3-zm_local_camera.cpp/1214 [Got format: 16-bit RGB 5-6-5 (0x50424752) at index 2]
02/17/21 22:03:21.731226 zmc_dvideo0[5560].DB3-zm_local_camera.cpp/1214 [Got format: 16-bit RGB 5-6-5 BE (0x52424752) at index 3]
02/17/21 22:03:21.731240 zmc_dvideo0[5560].DB3-zm_local_camera.cpp/1214 [Got format: 24-bit BGR 8-8-8 (0x33524742) at index 4]
02/17/21 22:03:21.731254 zmc_dvideo0[5560].DB3-zm_local_camera.cpp/1214 [Got format: 24-bit RGB 8-8-8 (0x33424752) at index 5]
02/17/21 22:03:21.731268 zmc_dvideo0[5560].DB3-zm_local_camera.cpp/1214 [Got format: 32-bit BGRA/X 8-8-8-8 (0x34524742) at index 6]
02/17/21 22:03:21.731282 zmc_dvideo0[5560].DB3-zm_local_camera.cpp/1214 [Got format: 32-bit A/XRGB 8-8-8-8 (0x34424752) at index 7]
02/17/21 22:03:21.731296 zmc_dvideo0[5560].DB3-zm_local_camera.cpp/1214 [Got format: YUYV 4:2:2 (0x56595559) at index 8]
02/17/21 22:03:21.731310 zmc_dvideo0[5560].DB3-zm_local_camera.cpp/1214 [Got format: UYVY 4:2:2 (0x59565955) at index 9]
02/17/21 22:03:21.731324 zmc_dvideo0[5560].DB6-zm_local_camera.cpp/1249 [No match for format: 16-bit A/XRGB 1-5-5-5 (0x5247424f) at index 0]
02/17/21 22:03:21.731337 zmc_dvideo0[5560].DB6-zm_local_camera.cpp/1249 [No match for format: 16-bit A/XRGB 1-5-5-5 BE (0x52474251) at index 1]
02/17/21 22:03:21.731350 zmc_dvideo0[5560].DB6-zm_local_camera.cpp/1249 [No match for format: 16-bit RGB 5-6-5 (0x52474250) at index 2]
02/17/21 22:03:21.731363 zmc_dvideo0[5560].DB6-zm_local_camera.cpp/1249 [No match for format: 16-bit RGB 5-6-5 BE (0x52474252) at index 3]
02/17/21 22:03:21.731376 zmc_dvideo0[5560].DB6-zm_local_camera.cpp/1249 [No match for format: 24-bit BGR 8-8-8 (0x42475233) at index 4]
02/17/21 22:03:21.731389 zmc_dvideo0[5560].DB6-zm_local_camera.cpp/1249 [No match for format: 24-bit RGB 8-8-8 (0x52474233) at index 5]
02/17/21 22:03:21.731401 zmc_dvideo0[5560].DB6-zm_local_camera.cpp/1249 [No match for format: 32-bit BGRA/X 8-8-8-8 (0x42475234) at index 6]
02/17/21 22:03:21.731413 zmc_dvideo0[5560].DB6-zm_local_camera.cpp/1249 [No match for format: 32-bit A/XRGB 8-8-8-8 (0x52474234) at index 7]
02/17/21 22:03:21.731426 zmc_dvideo0[5560].DB6-zm_local_camera.cpp/1249 [No match for format: YUYV 4:2:2 (0x59555956) at index 8]
02/17/21 22:03:21.731438 zmc_dvideo0[5560].DB6-zm_local_camera.cpp/1249 [No match for format: UYVY 4:2:2 (0x55595659) at index 9]
02/17/21 22:03:21.731451 zmc_dvideo0[5560].DB6-zm_local_camera.cpp/1249 [No match for format: 16-bit A/XRGB 1-5-5-5 (0x5247424f) at index 0]
02/17/21 22:03:21.731469 zmc_dvideo0[5560].DB6-zm_local_camera.cpp/1249 [No match for format: 16-bit A/XRGB 1-5-5-5 BE (0x52474251) at index 1]
02/17/21 22:03:21.731483 zmc_dvideo0[5560].DB6-zm_local_camera.cpp/1249 [No match for format: 16-bit RGB 5-6-5 (0x52474250) at index 2]
[... cut ...]
02/17/21 22:03:31.993331 zmc_dvideo0[5562].DB9-zm_image.cpp/2526 [0 - 1: min_y: 0, max_y:479, min_x:639.00, 1/m:0.00]
02/17/21 22:03:31.993348 zmc_dvideo0[5562].DB9-zm_image.cpp/2526 [1 - 0: min_y: 0, max_y:479, min_x:0.00, 1/m:-0.00]
02/17/21 22:03:31.993361 zmc_dvideo0[5562].DB9-zm_image.cpp/2526 [1 - 1: min_y: 0, max_y:479, min_x:639.00, 1/m:0.00]
02/17/21 22:03:31.993377 zmc_dvideo0[5562].DB9-zm_image.cpp/2526 [2 - 0: min_y: 0, max_y:479, min_x:0.00, 1/m:-0.00]
02/17/21 22:03:31.993390 zmc_dvideo0[5562].DB9-zm_image.cpp/2526 [2 - 1: min_y: 0, max_y:479, min_x:639.00, 1/m:0.00]

```

TP234-CPE, 2021-04-01:
@connortechnology Do you need anything else?