ZoneMinder/zoneminder #3106: WEB_EVENTS_PER_PAGE does not work using API

0rsa, 2021-01-05:
**Describe Your Environment**
Zoneminder 1.34.22
Debian Buster

**Describe the bug**
WEB_EVENTS_PER_PAGE seems to be not used using API /events/index

Steps to reproduce the behavior:
1. Call documentation example curl -XGET "http://server/zm/api/events/index/StartTime%20>=:2015-05-15%2018:43:56/EndTime%20<=:208:43:56.json"
2. API returns 100 results

**Expected behavior**
API returns WEB_EVENTS_PER_PAGE items (default is 25)

pliablepixels, 2021-01-14:
The API does not follow WEB_EVENTS_PER_PAGE - that is primarily for the console/UI.
When it comes to API, we would like to reduce overheads of API calls to get relevant events and that is why it is set to 100.

0rsa, 2021-01-14:
According to the documentation, API respects WEB_EVENTS_PER_PAGE 

Return a list of all events
http://server/zm/api/events.json
Note that events list can be quite large and this API (as with all other APIs in ZM) uses pagination. Each page returns a specific set of entries. **By default this is 25 and ties into WEB_EVENTS_PER_PAGE in the ZM options menu**.

pliablepixels, 2021-01-15:
The doc is dated then. 
https://github.com/ZoneMinder/zoneminder/blob/master/web/api/app/Controller/EventsController.php#L77

0rsa, 2021-01-15:
Ok thanks you.
Maybe can you think about URL parameter to define items number per page?
Currently I manage a double pagination.

connortechnology, 2021-01-15:
I actually don't like that we limit our results to 100.  When we do, it should be considered an ERROR.  Caller should specify a limit. We should deal with the issues around ram limits DOS etc...