ZoneMinder/zoneminder #3107: AMCREST PTZ errors

alabamatoy, 2021-01-11:
**Describe Your Environment**
- Version of ZoneMinder: 1.34.22
- How you installed ZoneMinder: PPA updated from 1.30
- Full name and version of OS: Ubuntu 16.04
- Browser name and version: N/A

**If the issue concerns a camera**
- Make and Model: Amcrest
- frame rate: 10FPS
- resolution: 1080P
- ZoneMinder Source Type: TCP

**Describe the bug**
Logs are filled with following"
`2021-01-11 08:25:15 | zmcontrol_20 |   | 29978 | ERR | Failed to get http://{login}:{password}@10.1.1.158/cgi-bin/ptz.cgi 400 Bad Request | zmcontrol.pl`

**To Reproduce**
Steps to reproduce the behavior:
View the monitor live.

**Expected behavior**
No errors.

**Debug Logs**
```
<insert debug logs here, please make sure they are within the ``` quotes so they are formatted properly>
```

I changed the following in Amcrest_HTTP.PM
    124  #Changed following line to get rid of the constant barrage of errors
    125  #Error("Failed to get $$self{base_url}cgi-bin/ptz.cgi ".$res->status_line());
    126  Info("Failed to get $$self{base_url}cgi-bin/ptz.cgi ".$res->status_line()." -- ignoring and continuing...");

No more errors, everything still works.

connortechnology, 2021-01-11:
I think what is happening here is that we are hitting ptz.cgi but not giving it a command.  So it is returning bad request.  We should probably be hitting a different url, or giving some command that effectively does nothing but gives us a successful result. Otherwise I'd be ok with removing this line entirely.

How about we send the command ?action=getCurrentProtocolCaps

alabamatoy, 2021-01-11:
I agree, but I understood that means changing the call which is setting the realm, and I was afraid to tinker with that.  There are some ptz.cgi status calls which could be used and the returned content ignored as you suggest.

alabamatoy, 2021-04-14:
In current version, this change works:

Following line is line #78
  #my $res = $self->{ua}->get($$self{base_url}.'cgi-bin/ptz.cgi');

changed above line to following
  my $res = $self->{ua}->get($$self{base_url}.'cgi-bin/magicBox.cgi?action=getDeviceType');

May want to change line 124 to give an accurate error message, since its no longer using cgi-bin/ptz.cgi

Latest (1.34.25) still has the old "cgi-bin/ptz.cgi" call that results in the errors..

connortechnology, 2021-04-20:
Merged it in 1.34.26

closed 2021-04-20 by connortechnology