ZoneMinder/zoneminder #3114: [1.34.23] Build fails during submodule init stage

Nocifer, 2021-01-24:
- ZoneMinder 1.34.23
- Building from source
- Arch Linux

Hello, I'm trying to build the just released 1.34.23 and getting this error during the git submodule init stage:

```
fatal: remote error: upload-pack: not our ref 2ab7e23fb4f5b31a2f043ba9b9f607a518007f75
Fetched in submodule path 'web/api/app/Plugin/Crud', but it did not contain 2ab7e23fb4f5b31a2f043ba9b9f607a518007f75. Direct fetching of that commit failed.
```

Building from the latest master produces no errors, it only happens with the 1.34.23 tag.

connortechnology, 2021-01-24:
release is no good.  Will re-release when I have it figured out.

Nocifer, 2021-01-25:
Great, with the latest re-release of .23 it now works properly. Thanks.

closed 2021-01-25 by Nocifer