ZoneMinder/zoneminder #3116: Zoneminder not saving any event

victordariovera, 2021-02-01:
**Describe Your Environment**
- Version of ZoneMinder: 1.34.23
- How you installed ZoneMinder : Official docker image (one docker for zoneminder, another for mysql)
- Full name and version of OS: Ubuntu 18
- Browser name and version: Issue is in all browsers and in zoneminder

**Describe the bug**
Zoneminder not saving any event. Record only a temporary event and then disappear. I have 8 cameras, some in mocord and others in modect. The only events that appears are the archived. I export and delete them, clean all the events from the web. Disable al cameras and enable only one for testing. Nothing happens, no saved events in record mode. 

Viewing cameras online is ok!
Data folder of mysql database is 30gb+ after delete all events. Thats not normal.

I try to clone one of the cameras and rename it. The same thing happens, no events saved.

I dont want to delete database, and initialize them from the ground because I need to understand that issue and resolve it. That installation are personal, but how I can resolve that if reinstalling or deleting database is not an option?

Logs are absolutly clean, no errors or warnings.
Mysql logs are absolutly clean, no error or warnings.
Storage have a lot of space. I can create files, move and rename it.
Storage permissions are 777 to avoid problems (I dont care security at this level).

I really lost, I dont know how to troubleshoot this. That happens from one day to another with no explanation. Sometimes I resolve this restarting dockers, but yesterday that workarround doesnt work anymore.

**To Reproduce**
Steps to reproduce the behavior:
1. Restart docker, or disable/enable camera. No events are saved. Live monitoring is ok.

**Expected behavior**
Zoneminder must save events in record, mocord o modect modes.

victordariovera, 2021-02-01:
Seems to resolve the issue by myself. I have the filter "Purge When Full" without specifying my dedicated storage area. When the default storage area are above the porcentage configured in the filter, zoneminder deletes all events in spite of dedicated storage area free space. I dont use default storage area at all.

knight-of-ni, 2021-02-01:
Hi - The core development team does not develop docker containers, nor do we support them in this github forum. We have dockerfiles [here](https://github.com/ZoneMinder/zmdockerfiles), but I'm not sure if that is what you used to build the container or not. You can open an issue in that repo if indeed that is what you are using. 

However, the quickest way to get help with zm in docker is to ask in our slack channel or post in our user forum, in the docker subforum.

closed 2021-02-01 by knight-of-ni