ZoneMinder/zoneminder #3118: Log reveals authentication details

alabamatoy, 2021-02-02:
**Describe Your Environment**

**If the issue concerns a camera**

**Describe the bug**

**To Reproduce**

**Expected behavior**

**Debug Logs**
```

<!--
N/A
-->

```

knight-of-ni, 2021-02-02:
You've posted a bug report with no content, which basically amounts to a feature request. This has been requested before. Try searching this forum to see our position on this matter. If you want this changed, posting empty bug reports is not the way to do it.

alabamatoy, 2021-02-02:
Well, that's not what I did.  I posted exactly what the format required, with the bug details, example log entry, versions etc etc.  I dont know why/how all that got deleted.

I disagree that this is a feature request, its a cybersecurity requirement.  Posting the creds to the logs (which often get emailed unencrypted like with logcheck) is really bad practice.

knight-of-ni, 2021-02-02:
The point of contention is you posted a bug report with no content. That by definition demands it be closed, regardless of the topic.

I edited the content of your message.  FYI, `` are HTML comments... anything you type in-between these gets interpreted as a comment and will not render.

anon8675309, 2021-09-29:
I am also affected by this security issue of passwords appearing in the logs.  I can confirm that this is still an issue as of today (tested in HEAD).

This ticket is marked as a "duplicate" but there's no indication of what ticket it duplicates and I was not able to find any when searching through open issues.  Is there another ticket that I should be commenting on?

This ticket should also have the "Security issue" label, as is seen on other tickets related to security, such as #2466.  I'd add it, but it appears that adding a label is an administrative action in GitHub (either that or I just don't know how to add labels in GitHub, which is entirely possible).

anon8675309, 2022-04-16:
Passwords are still being written out to the logs. If anyone would be willing to point me to where the code is that is writing this log message, I will take a stab at creating a patch and submitting a pull request.

This should still be tagged as a security issue as I mentioned in my previous comment. It is not a duplicate of any other open issue.

commit 939e1d83 (2022-04-18): add mMaskedPath and mMaskedSecondPath and use them when logging so that usernames and passwords are shown in logs. Fixes #3118
commit 57fdbda7 (2022-04-28): add mMaskedPath and mMaskedSecondPath and use them when logging so that usernames and passwords are shown in logs. Fixes #3118
closed 2022-04-28 by connortechnology