ZoneMinder/zoneminder #3135: Documentation issues

alabamatoy, 2021-02-06:
Instructions here [ https://github.com/ZoneMinder/ZoneMinder/blob/master/docs/installationguide/ubuntu.rst ] state "During installation it will ask you to set up a master/root password for the MySQL."  Just did this on 18.04, and it did not ask me to setup master password.

Same instructions state "Please note that these changes are NOT required for ZoneMinder 1.32.0 and some people have reported them causing problems in 1.32.0."  Should this not instead state "Please note that these changes are NOT required for ZoneMinder 1.32.0 **or newer versions**."

Same instructions state: "Step 5: Configure the ZoneMinder Database...This step should not be required on ZoneMinder 1.32.0."  Should it not instead state "This step should not be required on ZoneMinder **version 1.32.0 or newer**."

closed 2021-02-08 by connortechnology

caseyjhol, 2022-08-20:
This appears to still be an issue for step 5.

SteveGilvarry, 2022-08-20:
Fixed in latest version of documentation, readthedocs stable is as of release. So any changes you need to swap to latest, which is at the bottom on the menu.
https://zoneminder.readthedocs.io/en/latest/installationguide/ubuntu.html

commit 43761a2d (2021-02-08): Remove instructions for trusty. Trusty is no longer supported.  Apply recommendations for #3135.  Fixes #3135