ZoneMinder/zoneminder #3139: zms: segmentation fault

markcs, 2021-02-08:
**Describe Your Environment**
- Version of ZoneMinder: Ubuntu v1.35.18
- How you installed ZoneMinder : installed on ubuntu from http://ppa.launchpad.net/iconnor/zoneminder-master/ubuntu focal main
- Full name and version of OS: Ubuntu 20.04 LTS

**Describe the bug**
Since upgrading to 1.35.18, the camera streams are not showing and I'm seeing a segmentation fault when looking at zms

**Expected behavior**
View cameras

**Debug Logs**
```
02/08/21 12:18:29.619769 zms_m1[7746].ERR-zm_signal.cpp/49 [Got signal 11 (Segmentation fault), crashing]
02/08/21 12:18:29.660441 zms_m1[7746].ERR-zm_signal.cpp/79 [Signal address is (nil), from 0x55a73f087df5]
02/08/21 12:18:29.878313 zms_m1[7746].ERR-zm_signal.cpp/100 [Backtrace 0: /usr/lib/zoneminder/cgi-bin/nph-zms(_Z14zm_die_handleriP9siginfo_tPv+0x7a) [0x55a73f0f75ea]]
02/08/21 12:18:30.019540 zms_m1[7746].ERR-zm_signal.cpp/100 [Backtrace 1: /lib/x86_64-linux-gnu/libpthread.so.0(+0x153c0) [0x7fec375283c0]]
02/08/21 12:18:30.178094 zms_m1[7746].ERR-zm_signal.cpp/100 [Backtrace 2: /usr/lib/zoneminder/cgi-bin/nph-zms(_ZN5Image6AssignERKS_+0x15) [0x55a73f087df5]]
02/08/21 12:18:30.261550 zms_m1[7746].ERR-zm_signal.cpp/100 [Backtrace 3: /usr/lib/zoneminder/cgi-bin/nph-zms(_ZN13MonitorStream11SingleImageEi+0x11b) [0x55a73f0a13db]]
02/08/21 12:18:30.303611 zms_m1[7746].ERR-zm_signal.cpp/100 [Backtrace 4: /usr/lib/zoneminder/cgi-bin/nph-zms(_ZN13MonitorStream9runStreamEv+0x11a8) [0x55a73f0a7928]]
02/08/21 12:18:30.345424 zms_m1[7746].ERR-zm_signal.cpp/100 [Backtrace 5: /usr/lib/zoneminder/cgi-bin/nph-zms(main+0x12c5) [0x55a73f05ee05]]
02/08/21 12:18:30.387204 zms_m1[7746].ERR-zm_signal.cpp/100 [Backtrace 6: /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf3) [0x7fec34e750b3]]
02/08/21 12:18:30.428948 zms_m1[7746].ERR-zm_signal.cpp/100 [Backtrace 7: /usr/lib/zoneminder/cgi-bin/nph-zms(_start+0x2e) [0x55a73f06021e]]
02/08/21 12:18:30.470787 zms_m1[7746].INF-zm_signal.cpp/107 [Backtrace complete, please execute the following command for more information]
02/08/21 12:18:30.512550 zms_m1[7746].INF-zm_signal.cpp/108 [addr2line -e /usr/lib/zoneminder/cgi-bin/nph-zms 0x55a73f0f75ea 0x7fec375283c0 0x55a73f087df5 0x55a73f0a13db 0x55a73f0a7928 0x55a73f05ee05 0x7fec34e750b3 0x55a73f06021e]

```
```
/var/log/zm$ addr2line -e /usr/lib/zoneminder/cgi-bin/nph-zms 0x55a73f0f75ea 0x7fec375283c0 0x55a73f087df5 0x55a73f0a13db 0x55a73f0a7928 0x55a73f05ee05 0x7fec34e750b3 0x55a73f06021e
??:0
??:0
??:0
??:0
??:0
??:0
??:0
??:0
```

zhuykovkb, 2021-02-09:
maybe the same like https://github.com/ZoneMinder/zoneminder/issues/3132

Tony763, 2021-02-16:
Same goes for me, I also tried few previous commits. v1.35.17 and .16 has a same behavior. I run this on Jetson nano with L4T 4.4.  v1.35.14 works ok.

connortechnology, 2021-02-17:
59cdf971fa3083c2733c271390f966baadc3631f may have solved this.

Tony763, 2021-02-18:
I tried it , but still same error.

Tony763, 2021-02-18:
Small update:
If HW accelerated codec is not used: [Failed to find decoder (h264_mmal)], it works, but I get log full of:

2021-02-18 22:41:28 | zmc_m1 |   | 12766 | INF | deprecated pixel format used, make sure you did set range correctly] | zm_ffmpeg.cpp | 71
-- | -- | -- | -- | -- | -- | -- | --
2021-02-18 22:41:28 | zmc_m1 |   | 12766 | INF | No accelerated colorspace conversion found from yuv420p to rgba.] | zm_ffmpeg.cpp | 71

If I use HW accelerated codec, I get segmentation fault.

My FFmpeg is 4.2.2, self compiled.

zhuykovkb, 2021-02-19:
me to :( but i have `exit code = 7`

```

2021-02-19 20:19:48 | zmdc |   | 474 | ERR | 'zmc -m 1' exited abnormally, exit status 7 | zmdc.pl | -
-- | -- | -- | -- | -- | -- | -- | --
2021-02-19 20:19:48 | zmc_m1 |   | 26193 | INF | addr2line -e /usr/local/bin/zmc 0x4968d1 0x7fc76a6c6390 0x7fc767243bb8 0x43326c 0x4090b5 0x7fc7670f1840 0x40cdd9] | zm_signal.cpp | 108
2021-02-19 20:19:48 | zmc_m1 |   | 26193 | INF | Backtrace complete, please execute the following command for more information] | zm_signal.cpp | 107
2021-02-19 20:19:47 | zmc_m1 |   | 26193 | ERR | Backtrace 6: /usr/local/bin/zmc() [0x40cdd9]] | zm_signal.cpp | 100
2021-02-19 20:19:47 | zmc_m1 |   | 26193 | ERR | Backtrace 5: /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0) [0x7fc7670f1840]] | zm_signal.cpp | 100
2021-02-19 20:19:47 | zmc_m1 |   | 26193 | ERR | Backtrace 4: /usr/local/bin/zmc() [0x4090b5]] | zm_signal.cpp | 100
2021-02-19 20:19:47 | zmc_m1 |   | 26193 | ERR | Backtrace 3: /usr/local/bin/zmc() [0x43326c]] | zm_signal.cpp | 100
2021-02-19 20:19:47 | zmc_m1 |   | 26193 | ERR | Backtrace 2: /lib/x86_64-linux-gnu/libc.so.6(+0x172bb8) [0x7fc767243bb8]] | zm_signal.cpp | 100
2021-02-19 20:19:46 | zmc_m1 |   | 26193 | ERR | Backtrace 1: /lib/x86_64-linux-gnu/libpthread.so.0(+0x11390) [0x7fc76a6c6390]] | zm_signal.cpp | 100
2021-02-19 20:19:46 | zmc_m1 |   | 26193 | ERR | Backtrace 0: /usr/local/bin/zmc() [0x4968d1]] | zm_signal.cpp | 100
2021-02-19 20:19:46 | zmc_m1 |   | 26193 | ERR | Signal address is 0x7fc758780000, from 0x7fc767243bb8] | zm_signal.cpp | 79
2021-02-19 20:19:46 | zmc_m1 |   | 26193 | ERR | Got signal 7 (Bus error), crashing] | zm_signal.cpp | 49
2021-02-19 20:19:45 | zmc_m1 |   | 26193 | INF | Starting Capture version 1.35.18] | zmc.cpp | 220
2021-02-19 20:19:44 | zmdc |   | 474 | INF | 'zmc -m 1' starting at 21/02/19 20:19:44, pid = 26193 | zmdc.pl | -
2021-02-19 20:19:44 | zmdc |   | 26193 | INF | 'zmc -m 1' started at 21/02/19 20:19:44 | zmdc.pl | -
2021-02-19 20:19:44 | zmdc |   | 474 | INF | Command 'zmc -m 1' removed from pending list at 21/02/19 20:19:43 | zmdc.pl | -
2021-02-19 20:19:43 | zmwatch |   | 526 | INF | Restarting capture daemon for 726-1 cam1, shared data not valid | zmwatch.pl | -

```

> I tried it , but still same error.
> 
> 2021-02-18 18:02:50	zmdc	 	25558	ERR	'zmc -m 1' exited abnormally, exit status 11	zmdc.pl	-
> 2021-02-18 18:02:50	zmc_m1	 	30060	ERR	Got signal 11 (Segmentation fault), crashing]	zm_signal.cpp	49

Tony763, 2021-02-19:
I tried it today with latest commits. FFmpeg updated to 4.3.1. ZM returned Segmentation fault, after restart of ZM I get `'zmc -m 1' crashed, signal 138`.

With `sudo gdb --args zmc  "-m 1"`

```
Thread 1 "zmc" received signal SIGSEGV, Segmentation fault.
av_hwframe_transfer_data (dst=0x55587be1b0, src=0x55587be3d0, flags=flags@entry=0) at libavutil/hwcontext.c:449
449	        return transfer_data_alloc(dst, src, flags);

0x00000055555a5984 in ZMPacket::decode (this=this@entry=0x55588af870, ctx=0x55589106b0) at ./src/zm_packet.cpp:214

/* retrieve data from GPU to CPU */                                                                                                                                             
 zm_dump_video_frame(in_frame, "Before hwtransfer");                                                                                                                              
 ret = av_hwframe_transfer_data(new_frame, in_frame, 0);                                                                                                                          
 if ( ret < 0 ) {                               
```

connortechnology, 2021-02-20:
The zmc crashes are a completely separate issue, Stop hijacking threads.  That being said, I may have just fixed the crash in zmc.  The zms crash may or may not still be an issue. Who can tell?

Tony763, 2021-02-20:
@connortechnology: Sorry, I overlocked this issue was about zms and not zmc, I will test your latest commit and if not fixed, I will open new issue.

zhuykovkb, 2021-02-20:
i'm sorry for the hijacking @connortechnology

connortechnology, 2021-02-22:
@markcs Please give us an update on this and close if it is no longer a problem.

connortechnology, 2021-03-31:
This issue will be closed without an update.

closed 2021-05-10 by connortechnology

jofemodo, 2022-12-21:
Hi @connortechnology !

It seems that the issue is still not solved, or there is a new issue with the same symptoms ;-)

```

21 Dec 08:56:25 | web_js |   | 2099 | ERR | getCmdResponse stream error: socket_sendto( /run/zm/zms-278120s.sock ) failed: Connection refused - checkStreamForErrors() | ?view=event&eid=32&filter%5BQuery%5D%5Bterms%5D%5B0%5D%5Battr%5D=MonitorId&filter%5BQuery%5D%5Bterms%5D%5B0%5D%5Bop%5D=%3D&filter%5BQuery%5D%5Bterms%5D%5B0%5D%5Bval%5D=1&filter%5BQuery%5D%5Bsort_asc%5D=0&filter%5BQuery%5D%5Bsort_field%5D=StartDateTime&fil 
21 Dec 08:56:25 | web_php |   | 2099 | ERR | socket_sendto( /run/zm/zms-278120s.sock ) failed: Connection refused | /usr/share/zoneminder/www/includes/functions.php | 1882
21 Dec 08:56:25 | web_js |   | 2099 | ERR | getCmdResponse stream error: socket_sendto( /run/zm/zms-278120s.sock ) failed: Connection refused - checkStreamForErrors() | ?view=event&eid=32&filter%5BQuery%5D%5Bterms%5D%5B0%5D%5Battr%5D=MonitorId&filter%5BQuery%5D%5Bterms%5D%5B0%5D%5Bop%5D=%3D&filter%5BQuery%5D%5Bterms%5D%5B0%5D%5Bval%5D=1&filter%5BQuery%5D%5Bsort_asc%5D=0&filter%5BQuery%5D%5Bsort_field%5D=StartDateTime&fil 
21 Dec 08:56:24 | web_php |   | 2130 | ERR | socket_sendto( /run/zm/zms-278120s.sock ) failed: Connection refused | /usr/share/zoneminder/www/includes/functions.php | 1882
21 Dec 08:56:24 | zms_e32 |   | 2571 | INF | Backtrace  complete, please execute the following command for more information:  addr2line -e /usr/lib/zoneminder/cgi-bin/nph-zms 0x556913aac0c8  0x7efcdc834140 0x556913a58131 0x556913a42fe9 0x7efcda6bed0a  0x556913a44d8a | zm_signal.cpp | 108
21 Dec 08:56:24 | zms_e32 |   | 2571 | ERR | Backtrace 5: /usr/lib/zoneminder/cgi-bin/nph-zms(+0x1ad8a) [0x556913a44d8a] | zm_signal.cpp | 101
21 Dec 08:56:24 | zms_e32 |   | 2571 | ERR | Backtrace 4: /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xea) [0x7efcda6bed0a] | zm_signal.cpp | 101
21 Dec 08:56:24 | zms_e32 |   | 2571 | ERR | Backtrace 3: /usr/lib/zoneminder/cgi-bin/nph-zms(+0x18fe9) [0x556913a42fe9] | zm_signal.cpp | 101
21 Dec 08:56:24 | zms_e32 |   | 2571 | ERR | Backtrace 2: /usr/lib/zoneminder/cgi-bin/nph-zms(+0x2e131) [0x556913a58131] | zm_signal.cpp | 101
21 Dec 08:56:24 | zms_e32 |   | 2571 | ERR | Backtrace 1: /lib/x86_64-linux-gnu/libpthread.so.0(+0x13140) [0x7efcdc834140] | zm_signal.cpp | 101
21 Dec 08:56:24 | zms_e32 |   | 2571 | ERR | Backtrace 0: /usr/lib/zoneminder/cgi-bin/nph-zms(+0x820c8) [0x556913aac0c8] | zm_signal.cpp | 101
21 Dec 08:56:24 | zms_e32 |   | 2571 | ERR | Signal address is 0x7efcd2407f98, from 0x556913a58131 | zm_signal.cpp | 80
21 Dec 08:56:24 | zms_e32 |   | 2571 | ERR | Got signal 11 (Segmentation fault), crashing | zm_signal.cpp

```

+ System: LXC container, running Debian 11
+ Zoneminder v1.36.32, installed from zmrepo
+ Testing with a single camera, RTSP stream

Thanks!