ZoneMinder/zoneminder #3149: Camera not running, not displaying correctly

ken-crozier, 2021-02-09:
**Describe Your Environment**

Version 1.34.23
installed from source
Ubuntu 20.04.2 LTS (GNU/Linux 5.4.0-65-generic x86_64)
Firefox 85.0

**If the issue concerns a camera**
- 5 reolink cameras, 4 POE 1 Wifi RCL-410-5MP
- frame rate
- resolution 2560:1920
- ZoneMinder Source Type: rtmp

**Describe the bug**
Sometime yesterday one of the cameras disappeared from the console, although data for it is being updated in the GUI .... See screen shot .... When I added the camera to the system again, it shows as capturing, not running

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Debug Logs**
```

2021-02-09 12:22:54.680104 zma_m1[1379].INF-zm_monitor.cpp/1310 [Back: 12200 - Analysing at 6.25 fps]
2021-02-09 12:22:54.115840 zmdc[1205].ERR-zmdc.pl ['zma -m 6' exited abnormally, exit status 255]
2021-02-09 12:22:54.049396 zma_m6[5149].ERR-zm_monitor.cpp/501 [Shared data not initialised by capture daemon for monitor Patio]
2021-02-09 12:22:53.926890 zmdc[5149].INF-zmdc.pl ['zma -m 6' started at 21/02/09 12:22:53]
2021-02-09 12:22:53.924970 zmdc[1205].INF-zmdc.pl ['zma -m 6' starting at 21/02/09 12:22:53, pid = 5149]
2021-02-09 12:22:53.913110 zmdc[1205].INF-zmdc.pl [Starting pending process, zma -m 6]
2021-02-09 12:22:53.083823 zma_m2[1378].INF-zm_monitor.cpp/1310 [Garage: 12800 - Analysing at 6.67 fps]
2021-02-09 12:22:52.874400 zmdc[1205].ERR-zmdc.pl ['zma -m 6' exited abnormally, exit status 255]
2021-02-09 12:22:52.817605 zma_m6[5146].ERR-zm_monitor.cpp/501 [Shared data not initialised by capture daemon for monitor Patio]
2021-02-09 12:22:52.687680 zmdc[5146].INF-zmdc.pl ['zma -m 6' started at 21/02/09 12:22:52]
2021-02-09 12:22:52.686050 zmdc[1205].INF-zmdc.pl ['zma -m 6' starting at 21/02/09 12:22:52, pid = 5146]
2021-02-09 12:22:52.641560 zma_m3[1377].INF-zm_monitor.cpp/1310 [Front: 11900 - Analysing at 5.88 fps]
2021-02-09 12:22:52.524110 zmdc[1205].INF-zmdc.pl ['zmc -m 6' exited, signal 14]
2021-02-09 12:22:52.484400 zmdc[1205].INF-zmdc.pl ['zmc -m 6' sending stop to pid 5140 at 21/02/09 12:22:52]
2021-02-09 12:22:52.312240 zmdc[1205].ERR-zmdc.pl ['zma -m 6' exited abnormally, exit status 255]
2021-02-09 12:22:52.262550 zmdc[1205].INF-zmdc.pl ['zma -m 6' sending stop to pid 5139 at 21/02/09 12:22:52]
2021-02-09 12:22:52.253425 zmc_m6[5140].INF-zm_ffmpeg.cpp/84 [Not enabling ffmpeg logs, as LOG_FFMPEG and/or LOG_DEBUG is disabled in options, or this monitor is not part of your debug targets]
2021-02-09 12:22:52.244165 zma_m6[5139].ERR-zm_monitor.cpp/501 [Shared data not initialised by capture daemon for monitor Patio]
2021-02-09 12:22:52.144870 zmdc[5140].INF-zmdc.pl ['zmc -m 6' started at 21/02/09 12:22:52]
2021-02-09 12:22:52.142700 zmdc[1205].INF-zmdc.pl ['zmc -m 6' starting at 21/02/09 12:22:52, pid = 5140]
2021-02-09 12:22:52.126940 zmdc[1205].INF-zmdc.pl [Starting pending process, zmc -m 6]
2021-02-09 12:22:52.046080 zmdc[5139].INF-zmdc.pl ['zma -m 6' started at 21/02/09 12:22:52]
2021-02-09 12:22:52.046070 zmdc[1205].INF-zmdc.pl ['zma -m 6' starting at 21/02/09 12:22:52, pid = 5139]
2021-02-09 12:22:51.556601 zmc_m5[1266].INF-zm_monitor.cpp/2554 [Tait: images:11700 - Capturing at 6.25 fps, capturing bandwidth 23173bytes/sec]
2021-02-09 12:22:50.900926 zmc_m1[1235].INF-zm_monitor.cpp/2554 [Back: images:12800 - Capturing at 7.14 fps, capturing bandwidth 31842bytes/sec]
2021-02-09 12:22:50.433868 zmc_m2[1248].INF-zm_monitor.cpp/2554 [Garage: images:12800 - Capturing at 6.67 fps, capturing bandwidth 29164bytes/sec]
2021-02-09 12:22:49.804420 zmwatch[1293].INF-zmwatch.pl [Restarting capture daemon for Patio, shared data not valid]
2021-02-09 12:22:48.097290 zmdc[1205].INF-zmdc.pl [Starting pending process, zma -m 6]
2021-02-09 12:22:47.993729 zmc_m3[1258].INF-zm_monitor.cpp/2554 [Front: images:12800 - Capturing at 7.14 fps, capturing bandwidth 31573bytes/sec]
2021-02-09 12:22:47.435484 zma_m5[1376].INF-zm_monitor.cpp/1310 [Tait: 11200 - Analysing at 5.88 fps]
2021-02-09 12:22:46.025100 zmdc[1205].INF-zmdc.pl ['zmc -m 6' crashed, signal 134]
2021-02-09 12:22:43.723518 zmc_m6[5124].INF-zm_ffmpeg.cpp/84 [Not enabling ffmpeg logs, as LOG_FFMPEG and/or LOG_DEBUG is disabled in options, or this monitor is not part of your debug targets]
2021-02-09 12:22:43.645540 zmdc[1205].ERR-zmdc.pl ['zma -m 6' exited abnormally, exit status 255]
2021-02-09 12:22:43.582680 zmdc[1205].INF-zmdc.pl ['zmc -m 6' starting at 21/02/09 12:22:43, pid = 5124]
2021-02-09 12:22:43.582680 zmdc[5124].INF-zmdc.pl ['zmc -m 6' started at 21/02/09 12:22:43]
2021-02-09 12:22:43.484100 zmdc[1205].INF-zmdc.pl [Starting pending process, zmc -m 6]
2021-02-09 12:22:43.420717 zma_m6[5123].ERR-zm_monitor.cpp/501 [Shared data not initialised by capture daemon for monitor Patio]
2021-02-09 12:22:43.335290 zmdc[1205].INF-zmdc.pl ['zmc -m 6' crashed, signal 134]
2021-02-09 12:22:43.294940 zmdc[5123].INF-zmdc.pl ['zma -m 6' started at 21/02/09 12:22:43]
2021-02-09 12:22:43.293770 zmdc[1205].INF-zmdc.pl ['zma -m 6' starting at 21/02/09 12:22:43, pid = 5123]
2021-02-09 12:22:41.021650 zmdc[1205].INF-zmdc.pl [Starting pending process, zma -m 6]
[... 37 lines cut ...]
2021-02-09 12:22:25.901900 zmdc[5077].INF-zmdc.pl ['zmc -m 6' started at 21/02/09 12:22:25]
2021-02-09 12:22:25.900770 zmdc[1205].INF-zmdc.pl ['zmc -m 6' starting at 21/02/09 12:22:25, pid = 5077]
2021-02-09 12:22:25.826810 zmdc[1205].INF-zmdc.pl [Starting pending process, zmc -m 6]
2021-02-09 12:22:24.521577 zma_m1[1379].INF-zm_monitor.cpp/1310 [Back: 12000 - Analysing at 7.14 fps]
2021-02-09 12:22:23.608025 zma_m2[1378].INF-zm_monitor.cpp/1310 [Garage: 12600 - Analysing at 6.67 fps]
2021-02-09 12:22:22.762090 zmdc[1205].ERR-zmdc.pl ['zma -m 6' exited abnormally, exit status 255]
2021-02-09 12:22:22.681154 zma_m6[5070].ERR-zm_monitor.cpp/501 [Shared data not initialised by capture daemon for monitor Patio]
2021-02-09 12:22:22.545970 zmdc[5070].INF-zmdc.pl ['zma -m 6' started at 21/02/09 12:22:22]
2021-02-09 12:22:22.545960 zmdc[1205].INF-zmdc.pl ['zma -m 6' starting at 21/02/09 12:22:22, pid = 5070]
2021-02-09 12:22:22.486680 zmdc[1205].INF-zmdc.pl [Starting pending process, zma -m 6]
2021-02-09 12:22:21.424595 zmc_m1[1235].INF-zm_monitor.cpp/2554 [Back: images:12600 - Capturing at 6.67 fps, capturing bandwidth 24764bytes/sec]
2021-02-09 12:22:20.917099 zmc_m2[1248].INF-zm_monitor.cpp/2554 [Garage: images:12600 - Capturing at 7.14 fps, capturing bandwidth 26078bytes/sec]
2021-02-09 12:22:20.050460 zma_m3[1377].INF-zm_monitor.cpp/1310 [Front: 11700 - Analysing at 7.14 fps]
2021-02-09 12:22:19.186356 zmc_m5[1266].INF-zm_monitor.cpp/2554 [Tait: images:11500 - Capturing at 6.25 fps, capturing bandwidth 24118bytes/sec]
2021-02-09 12:22:18.479935 zmc_m3[1258].INF-zm_monitor.cpp/2554 [Front: images:12600 - Capturing at 6.67 fps, capturing bandwidth 24199bytes/sec]
2021-02-09 12:22:17.451510 zmdc[1205].ERR-zmdc.pl ['zma -m 6' exited abnormally, exit status 255]
2021-02-09 12:22:17.407222 zma_m6[5059].ERR-zm_monitor.cpp/501 [Shared data not initialised by capture daemon for monitor Patio]
2021-02-09 12:22:17.277200 zmdc[5059].INF-zmdc.pl ['zma -m 6' started at 21/02/09 12:22:17]
2021-02-09 12:22:17.277190 zmdc[1205].INF-zmdc.pl ['zma -m 6' starting at 21/02/09 12:22:17, pid = 5059]
2021-02-09 12:22:17.218780 zmdc[1205].INF-zmdc.pl [Starting pending process, zma -m 6]
2021-02-09 12:22:16.192040 zmdc[1205].ERR-zmdc.pl ['zma -m 6' exited abnormally, exit status 255]
2021-02-09 12:22:16.146996 zma_m6[5058].ERR-zm_monitor.cpp/501 [Shared data not initialised by capture daemon for monitor Patio]

```
[image]

connortechnology, 2021-02-09:
You see the big red line that says /dev/shm: 100% ?  Do you think it is red for a reason?

You are out of space in /dev/shm. Reduce buffers, reduce resolution, add ram.

closed 2021-02-09 by connortechnology

ken-crozier, 2021-02-09:
I lower to 1080p and /dev/shm is now about 42% .... but the extra line in the GUI ????
[image]

connortechnology, 2021-02-10:
What extra line?

ken-crozier, 2021-02-11:
Sorry, just figured it out that it's the totals ....