ZoneMinder/zoneminder #3150: ZoneMinder causing kernel panic onCentos 8.3 with build 1.120.20210209gitg250686377.el8

dodexahedron, 2021-02-09:
**Describe Your Environment**
- Version of ZoneMinder: 1.35.16 (release 1.120.20210209gitg250686377.el8)
- How you installed ZoneMinder: rpmfusion
- Full name and version of OS: Centos 8.3 Kernel 4.18.0-240.10.1.el8_3.x86_64

**Describe the bug**
Upon start of the zoneminder service with the latest current dev release, ( 1.120.20210209gitg250686377.el8), the system immediately crashes. Log files are created in /var/log/zoneminder, but they are all zero bytes.
If I disable the service and reboot, the server can boot normally.
This did not happen with the build from February 8, and only began after the upgrade to the February 9 build.

**To Reproduce**
Steps to reproduce the behavior:
1. On CentOS 8.3, attempt `systemctl start zoneminder`
2. ???
3. un-profit

**Expected behavior**
ZoneMinder should start and server should stay operating.

**Debug Logs**
None available, unless you'd like to see the kernel crash logs.

dodexahedron, 2021-02-09:
Also noting there were no configuration changes made.
What caused the same, repeatable, behavior on 3 servers was:

```
systemctl stop zoneminder
dnf upgrade
systemctl start zoneminder
```

dodexahedron, 2021-02-09:
Confirming it's the new build that broke it.

```
dnf downgrade zoneminder-common-1.35.16-1.120.20210208gitg250686377.el8.x86_64 zoneminder-nginx-1.35.16-1.120.20210208gitg250686377.el8.x86_64
systemctl start zoneminder
```

Service comes up and server keeps running.
Upon repeating the upgrade to 1.120.20210209gitg250686377, the server crashes immediately, upon start of the zoneminder service.

dodexahedron, 2021-02-10:
More info after messing with clones of the original VMs...

I upgraded to 1.127.20210210gitg393e8b582.el8 and ran zmupdate. Same result - server crashed nearly instantly.

As a lark, I enabled the kernel-ml repository from elrepo and installed kernel-ml-5.10.13 and rebooted into that kernel (which also required disabling secure boot, just to point out all differences).
Zoneminder starts and runs normally.
I then repeated the procedure on the clones of the second and third servers, and they worked properly, as well, running the newer kernel.

Have not yet reverted back to the older/standard kernel to see if it crashes again, but I should be able to do that testing in the next few days.

connortechnology, 2021-02-10:
This is pretty alarming.  ZM is a userspace app.  There is no way it should crash linux.  

Are you using v4l cameras?  Or network cameras?  Are you running out of ram?  What errors does the kernel output?

dodexahedron, 2021-02-10:
They're network cameras. Specifically, ubiquiti G3 cameras of various models.

I did allocate more memory and vCPUs to the VMs just in case it was running out of memory, but that didn't fix it, and there were no out of memory errors in the system logs.

Here's where the crash happens, from the last crash log on one of them, after a service start:

```
[71079.319465] BUG: unable to handle kernel NULL pointer dereference at 0000000000000060
[71079.319552] PGD 0 P4D 0 
[71079.319568] Oops: 0000 [#1] SMP PTI
[71079.319592] CPU: 9 PID: 15331 Comm: rm Kdump: loaded Not tainted 4.18.0-240.10.1.el8_3.x86_64 #1
[71079.319631] Hardware name: VMware, Inc. VMware7,1/440BX Desktop Reference Platform, BIOS VMW71.00V.16707776.B64.2008070230 08/07/2020
[71079.319686] RIP: 0010:d_path+0x34/0x140
[71079.319729] Code: 48 89 fb 48 83 ec 28 48 8b 7f 08 89 54 24 04 65 48 8b 04 25 28 00 00 00 48 89 44 24 20 31 c0 48 63 c2 48 01 f0 48 89 44 24 08 <48> 8b 47 60 48 85 c0 74 37 48 8b 40 48 48 85 c0 74 2e 48 3b 7f 18
[71079.319874] RSP: 0018:ffffa38a841afe18 EFLAGS: 00010282
[71079.319916] RAX: ffff91016dd8b00b RBX: ffff910124b9dab0 RCX: 0000000000002a56
[71079.319970] RDX: 000000000000100b RSI: ffff91016dd8a000 RDI: 0000000000000000
[71079.320024] RBP: ffffa38a841afe48 R08: ffff9102b9aef1c0 R09: ffff910087c06400
[71079.320078] R10: ffff9101349bbec0 R11: ffff9101349bb831 R12: ffff91016dd8a000
[71079.320132] R13: 0000000000000000 R14: ffff910124b9d800 R15: 0000000000000000
[71079.320187] FS:  00007f07ee8b9540(0000) GS:ffff9102b9ac0000(0000) knlGS:0000000000000000
[71079.320247] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[71079.320292] CR2: 0000000000000060 CR3: 00000001d202c005 CR4: 00000000003606e0
[71079.320408] Call Trace:
[71079.320501]  audit_log_d_path+0x6b/0xd0
[71079.320535]  audit_log_exit+0x66f/0xd40
[71079.320569]  ? audit_filter_inodes+0x2e/0x100
[71079.320606]  ? audit_filter_syscall.constprop.16+0x2c/0xd0
[71079.320650]  __audit_syscall_exit+0x238/0x2a0
[71079.320689]  syscall_slow_exit_work+0x101/0x130
[71079.320728]  do_syscall_64+0x170/0x1a0
[71079.320763]  entry_SYSCALL_64_after_hwframe+0x65/0xca
[71079.320805] RIP: 0033:0x7f07ee3ca3fb
[71079.320837] Code: 73 01 c3 48 8b 0d 8d da 2c 00 f7 d8 64 89 01 48 83 c8 ff c3 66 2e 0f 1f 84 00 00 00 00 00 90 f3 0f 1e fa b8 07 01 00 00 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d 5d da 2c 00 f7 d8 64 89 01 48
[71079.320983] RSP: 002b:00007ffd800eb578 EFLAGS: 00000246 ORIG_RAX: 0000000000000107
[71079.321041] RAX: fffffffffffffff3 RBX: 000055758463e850 RCX: 00007f07ee3ca3fb
[71079.321094] RDX: 0000000000000000 RSI: 000055758463e958 RDI: 0000000000000004
[71079.321148] RBP: 000055758463d440 R08: 0000000000000003 R09: 0000000000000000
[71079.321202] R10: 0000000000000000 R11: 0000000000000246 R12: 00007ffd800eb760
[71079.321266] R13: 0000000000000000 R14: 000055758463e850 R15: 0000000000000000
```

I mean, to me, that looks like a kernel issue that's being triggered by something (perhaps the threading changes?), rather than something wrong with zm itself.

I'm going to clone one of the working ones and roll its kernel back to the CentOS standard kernel package in a bit to see if the problem comes back.

Carbenium, 2021-02-10:
That looks like a selinux problem. You could try booting with `selinux=0` and see if it goes away. Not sure what in ZM could trigger this (if it's ZM at all).

dodexahedron, 2021-02-10:
It's set to Permissive right now, so that's surprising also.
I'll add that to my testing when I roll back, though.

dodexahedron, 2021-02-10:
> **Edit**: Just found this: https://forums.centos.org/viewtopic.php?t=76680
> Seems to happen on non-ZM systems as well.

Interesting for sure.
Will definitely test with selinux disabled entirely.

dodexahedron, 2021-02-10:
Digging into the crash dump, the thing that caused it was the `rm` of the old file.

```
crash> ps 15331
   PID    PPID  CPU       TASK        ST  %MEM     VSZ    RSS  COMM
> 15331  15321   9  ffff91028e1b4740  RU   0.0  217080   1924  rm
crash> ps 15331 -p
PID: 0      TASK: ffffffffb3612780  CPU: 0   COMMAND: "swapper/0"
 PID: 1      TASK: ffff910147075f00  CPU: 4   COMMAND: "systemd"
  PID: 15247  TASK: ffff91028e1b2f80  CPU: 6   COMMAND: "zmdc.pl"
   PID: 15321  TASK: ffff9101ba594740  CPU: 8   COMMAND: "zmfilter.pl"
    PID: 15331  TASK: ffff91028e1b4740  CPU: 9   COMMAND: "rm"
```

dodexahedron, 2021-02-11:
Here are my results. All other items not specifically noted were left the same as a working system for each test.
Each test was performed by starting the service and waiting for the filters to run.

### Test with previous kernel, selinux on/Permissive and secureboot off
Worked fine, initially.

### Test with previous kernel, selinux on/Permissive and secureboot on
Worked fine, initially

OK, so that seemed odd, since it was pretty consistently happening, before. Perhaps related to events from a specifc monitor?
So I waited a little while longer.
Next time the filter ran, in this configuration, it crashed on an rm again.

So I tried to see what could be different between executions of the filter.
The time it crashed, it was deleting a file from a folder that had been created for a camera, but the folder's ownership was root:root, vs the other folders, which are owned by nginx.

This is almost certainly an selinux/kernel bug that causes the actual crash.

As for the condition that led to the crash...
I'm curious why the folders were created with the wrong ownership.

I repeated the test again, after setting the folder ownership to nginx, and the crash no longer occurs.

I think this can be closed, since I've isolated the issue to be selinux/kernel-related, but I was curious about the folder ownership.
The users exist on the zm system as well as on the nfs server, and have the same uid/gid, so I'd expect anything zoneminder creates to be created with nginx as the owner.

Then it occured to me that I had manually run zmfilter, as root, on a system, to move a bunch of events. That's probably what made the root-owned folders and then, later on, when those events were subject to deletion, resulted in the crash.

Summary:

- Does not appear to be a zm bug causing the actual crash.
- Folder ownership for some monitors appear to have been bad because of manual run of zmfilter as root.
- Crash does not occur when folder ownership is correct.
- CentOS with kernel-ml-5.10.13 does not experience the bug, so it's clearly been handled already, there.

dodexahedron, 2021-02-12:
Closing this.

closed 2021-02-12 by dodexahedron