ZoneMinder/zoneminder #3160: Incorrect duration causing playback issues with libx264 encoding

dodexahedron, 2021-02-12:
**Describe Your Environment**
- Version of ZoneMinder 1.35.18-1.145.20210211gitg5663b0526.el8
- How you installed ZoneMinder rpmfusion
- Full name and version of OS CentOS 8.3 Mainline Kernel 5.10.15

**Describe the bug**
When a monitor is configured as shown:
[image]
[image]

Duration of the resulting video files is consistently incorrect. The duration always ends up being 1h:50m, when the actual duration is 10m. As a result, attempting to play the videos back gives you an incredibly slow playback.
The cameras themselves are configured to output 5FPS, and videos recorded using h.264 passthrough have correct durations and play back correctly.

Here's what mediainfo says about it:
```
# mediainfo 396042/396042-video.mp4 
General
Complete name                            : 396042/396042-video.mp4
Format                                   : MPEG-4
Format profile                           : Base Media
Codec ID                                 : isom (isom/iso2/avc1/iso6/mp41)
File size                                : 188 MiB
Duration                                 : 1 h 50 min
Overall bit rate                         : 237 kb/s
Movie name                               : Zoneminder Security Recording
Writing application                      : Lavf58.29.100

Video
ID                                       : 1
Format                                   : AVC
Format/Info                              : Advanced Video Codec
Format profile                           : High@L3.2
Format settings                          : CABAC / 3 Ref Frames
Format settings, CABAC                   : Yes
Format settings, Reference frames        : 3 frames
Format settings, GOP                     : M=2, N=12
Codec ID                                 : avc1
Codec ID/Info                            : Advanced Video Coding
Duration                                 : 1 h 50 min
Bit rate                                 : 237 kb/s
Width                                    : 1 920 pixels
Height                                   : 1 080 pixels
Display aspect ratio                     : 16:9
Frame rate mode                          : Variable
Frame rate                               : 0.360 FPS
Minimum frame rate                       : 0.308 FPS
Maximum frame rate                       : 0.444 FPS
Color space                              : YUV
Chroma subsampling                       : 4:2:0
Bit depth                                : 8 bits
Scan type                                : Progressive
Bits/(Pixel*Frame)                       : 0.318
Stream size                              : 188 MiB (100%)
Writing library                          : x264 core 157 r2980 34c06d1
Encoding settings                        : cabac=1 / ref=3 / deblock=1:0:0 / analyse=0x3:0x113 / me=hex / subme=7 / psy=1 / psy_rd=1.00:0.00 / mixed_ref=1 / me_range=16 / chroma_me=1 / trellis=1 / 8x8dct=1 / cqm=0 / deadzone=21,11 / fast_pskip=1 / chroma_qp_offset=-2 / threads=24 / lookahead_threads=4 / sliced_threads=0 / nr=0 / decimate=1 / interlaced=0 / bluray_compat=0 / constrained_intra=0 / bframes=1 / b_pyramid=0 / b_adapt=1 / b_bias=0 / direct=1 / weightb=1 / open_gop=0 / weightp=2 / keyint=12 / keyint_min=1 / scenecut=40 / intra_refresh=0 / rc_lookahead=12 / rc=crf / mbtree=1 / crf=27.0 / qcomp=0.60 / qpmin=0 / qpmax=69 / qpstep=4 / ip_ratio=1.40 / aq=1:1.00
Codec configuration box                  : avcC
```

According to the database, this video has 3000 frames, which is exactly what I'd expect, given 600 seconds and 5FPS.
Yet the framerate is shown as variable and very very low.

Here's what one looks like, in mediainfo, when configured to use passthrough:
```
# mediainfo 396073/396073-video.mp4 
General
Complete name                            : 396073/396073-video.mp4
Format                                   : MPEG-4
Format profile                           : Base Media
Codec ID                                 : isom (isom/iso2/avc1/iso6/mp41)
File size                                : 187 MiB
Duration                                 : 10 min 2 s
Overall bit rate                         : 2 602 kb/s
Movie name                               : Zoneminder Security Recording
Writing application                      : Lavf58.29.100

Video
ID                                       : 1
Format                                   : AVC
Format/Info                              : Advanced Video Codec
Format profile                           : Main@L4.1
Format settings                          : CABAC / 1 Ref Frames
Format settings, CABAC                   : Yes
Format settings, Reference frames        : 1 frame
Format settings, GOP                     : M=1, N=4
Codec ID                                 : avc1
Codec ID/Info                            : Advanced Video Coding
Duration                                 : 10 min 2 s
Bit rate                                 : 2 597 kb/s
Width                                    : 1 920 pixels
Height                                   : 1 080 pixels
Display aspect ratio                     : 16:9
Original display aspect ratio            : 16:9
Frame rate mode                          : Variable
Frame rate                               : 4.000 FPS
Minimum frame rate                       : 3.750 FPS
Maximum frame rate                       : 4.286 FPS
Color space                              : YUV
Chroma subsampling                       : 4:2:0
Bit depth                                : 8 bits
Scan type                                : Progressive
Bits/(Pixel*Frame)                       : 0.313
Stream size                              : 186 MiB (100%)
Color range                              : Full
Color primaries                          : BT.709
Transfer characteristics                 : BT.709
Matrix coefficients                      : BT.709
Codec configuration box                  : avcC
```

This file plays back correctly.

Side note: That particular camera was getting no space savings from the re-encoding, so it's been switched to passthrough, anyway.
The CPU load is never over 40% across all cores, while running 6 cameras at the settings shown above, and there's plenty of memory headroom.

**To Reproduce**
Steps to reproduce the behavior:
1. Configure a monitor as shown, using libx264.
2. Wait for video to be recorded

**Expected behavior**
Videos encoded with libx264 should have the correct framerate and play back at the correct rate.

connortechnology, 2021-02-19:
I may have fixed this in the latest commit.
2f917b4fac84ed384a871b3ce0fe31876d55c00c

dodexahedron, 2021-02-19:
Unfortunately, the build from this commit exhibits the same behavior, as far as playback of the file goes, via zoneminder or just downloading the file and playing it back in, for example, vlc.

```
# mediainfo 411593/411593-video.mp4 
General
Complete name                            : 411593/411593-video.mp4
Format                                   : MPEG-4
Format profile                           : Base Media
Codec ID                                 : isom (isom/iso2/avc1/iso6/mp41)
File size                                : 126 MiB
Duration                                 : 1 h 50 min
Overall bit rate                         : 158 kb/s
Movie name                               : Zoneminder Security Recording
Writing application                      : Lavf58.29.100

Video
ID                                       : 1
Format                                   : AVC
Format/Info                              : Advanced Video Codec
Format profile                           : High@L3.2
Format settings                          : CABAC / 2 Ref Frames
Format settings, CABAC                   : Yes
Format settings, Reference frames        : 2 frames
Format settings, GOP                     : M=2, N=12
Codec ID                                 : avc1
Codec ID/Info                            : Advanced Video Coding
Duration                                 : 1 h 50 min
Bit rate                                 : 158 kb/s
Width                                    : 1 920 pixels
Height                                   : 1 080 pixels
Display aspect ratio                     : 16:9
Frame rate mode                          : Variable
Frame rate                               : 0.362 FPS
Minimum frame rate                       : 0.104 FPS
Maximum frame rate                       : 8.000 FPS
Color space                              : YUV
Chroma subsampling                       : 4:2:0
Bit depth                                : 8 bits
Scan type                                : Progressive
Bits/(Pixel*Frame)                       : 0.211
Stream size                              : 126 MiB (100%)
Writing library                          : x264 core 157 r2980 34c06d1
Encoding settings                        : cabac=1 / ref=2 / deblock=1:0:0 / analyse=0x3:0x113 / me=hex / subme=6 / psy=1 / psy_rd=1.00:0.00 / mixed_ref=1 / me_range=16 / chroma_me=1 / trellis=1 / 8x8dct=1 / cqm=0 / deadzone=21,11 / fast_pskip=1 / chroma_qp_offset=-2 / threads=24 / lookahead_threads=4 / sliced_threads=0 / nr=0 / decimate=1 / interlaced=0 / bluray_compat=0 / constrained_intra=0 / bframes=1 / b_pyramid=0 / b_adapt=1 / b_bias=0 / direct=1 / weightb=1 / open_gop=0 / weightp=1 / keyint=12 / keyint_min=1 / scenecut=40 / intra_refresh=0 / rc_lookahead=12 / rc=crf / mbtree=1 / crf=29.0 / qcomp=0.60 / qpmin=0 / qpmax=69 / qpstep=4 / ip_ratio=1.40 / aq=1:1.00
Codec configuration box                  : avcC
```

```
# dnf info zoneminder-common
Last metadata expiration check: 1:39:32 ago on Fri 19 Feb 2021 02:31:28 PM MST.
Installed Packages
Name         : zoneminder-common
Version      : 1.35.18
Release      : 1.226.20210219gitg2f917b4fa.el8
Architecture : x86_64
Size         : 34 M
Source       : zoneminder-1.35.18-1.226.20210219gitg2f917b4fa.el8.src.rpm
```

That file should also be just 10 minutes and should be a constant 4FPS.

commit 88f3d732 (2021-02-22): Always use AV_TIME_BASE_Q in video encoding mode as we use packet->timestamp instead of pts. Fixes duration problems with encoded video. Fixes #3160
closed 2021-02-22 by connortechnology