ZoneMinder/zoneminder #3162: Next button no longer goes to next event.

botzen67, 2021-02-12:
**Describe Your Environment**
- v1.35.18
- apt-get update
- Ubuntu 20.04
- Chrome (latest version)

**If the issue concerns a camera**
- Not a camera issue.
- 
**Describe the bug**
Clicking next while replaying events no longer goes to next event.
Gives "No More Events" message even though there are more events in the list

**To Reproduce**
Steps to reproduce the behavior:
1. Click on Console
2. Open a list of events by clicking on any of the numbers which bring up a list of events
3. Click on first event number or name to start playback
4. While event is replaying click on "Next" button
5. "No More Events" is displayed at the top of the playback window even though there are more events in the list.

**Expected behavior**
Clicking next should start playing the next event.

**Debug Logs**
``` None taken

<insert debug logs here, please make sure they are within the ``` quotes so they are formatted properly>

```

botzen67, 2021-02-12:
Also, trying to use the previous button results in an "Event was not found."  error message.
I'm guessing both of these behaviors are related.

cotman, 2021-02-14:
Seeing this too with `1.35.18~20210212115535-focal` and Chrome on Ubuntu 20.04.

When clicking "Next" I see the following in the console logs:

    js_logger-classic-1613148935.js:93 getCmdResponse stream error: Socket /run/zm/zms-278209s.sock does not exist.  This file is created by zms, and since it does not exist, either zms did not run, or zms exited early.  Please check your zms logs and ensure that CGI is enabled in apache and check that the PATH_ZMS is set correctly.  Make sure that ZM is actually recording.  If you are trying to view a live stream and the capture process (zmc) is not running then zms will exit. Please go to http://zoneminder.readthedocs.io/en/latest/faq.html#why-can-t-i-see-streamed-images-when-i-can-see-stills-in-the-zone-window-etc for more information.

Additionally, the "Scale" drop-down doesn't work in the viewer either. I get the above error, but also this:

    skins_classic_views_js_event-classic-1613148935.js:843 Uncaught ReferenceError: vidWidth is not defined
        at onStatsResize (skins_classic_views_js_event-classic-1613148935.js:843)
        at HTMLSelectElement.changeScale (skins_classic_views_js_event-classic-1613148935.js:204)

Otherwise, single event viewing works fine.

closed 2021-02-14 by connortechnology

botzen67, 2021-02-15:
Confirmed the specified code chance fixes the problem.

darfij, 2021-03-01:
same problem with next, previous event
additionally deleting the event does not bring up the next event any more
v1.35.18
on Ubuntu 20.04.2 LTS

aaronsta1, 2022-01-26:
this doesnt seem to have been fixed? im getting this exact issue.
i have gapless events enabled but when it goes to the next video it says no more events. even clicking back doesnt work.

connortechnology, 2022-01-31:
There can be many reasons.  This issue was addressed.  You have a different issue, with similar symptoms.

commit 71183a60 (2021-02-14): fix case of vidWidth.  Fixes #3162