ZoneMinder/zoneminder #3163: Error accessing the API with 1.34.23 due to CakePHP's incompatibility with PHP 8.0

Nocifer, 2021-02-17:
**Describe Your Environment**
- Version of ZoneMinder: 1.34.23
- How you installed ZoneMinder: built from source
- OS: Arch Linux

**Describe the bug**
In the last few days, some users reported on [Zoneminder's AUR page](https://aur.archlinux.org/packages/zoneminder/) that after upgrading to PHP 8.0, it was impossible to access the API (e.g. for using with zmNinja).

It was found that the problem was with CakePHP, and that applying 2fa0fbe448058f2826dadc44572a822af2a177d5 fixes it, so if there is ever an additional minor 1.34.x release before 1.35 and CakePHP 3.x, please consider including that commit as well.

Thank you.

mike-lloyd03, 2021-02-27:
I'm also unable to access the API running ZM on Arch (installed from the AUR). I downloaded the latest release and copied from the git repo `web/api/lib/Cake` to `/usr/share/webapps/zoneminder/www/api/lib` and renamed `bootstrap.php.in` to `bootstrap.php` (per [a suggestion on the AUR](https://aur.archlinux.org/packages/zoneminder/#comment-790004)). However, this doesn't work.

How does one apply [2fa0fbe](https://github.com/ZoneMinder/zoneminder/commit/2fa0fbe448058f2826dadc44572a822af2a177d5) as suggested above?

Nocifer, 2021-02-27:
@mike-lloyd03 You apply it by using git's [cherry-pick command](https://git-scm.com/docs/git-cherry-pick), but I've already included this fix in the latest 1.34.23-2 AUR release so you shouldn't have any issues with the API any more (I can't guarantee it first hand, but there have been no new complaints so far, so I suppose it does work). Keep in mind though that applying the fix mentioned in that comment you linked will probably break things all over again.

mike-lloyd03, 2021-02-28:
I thought I had updated to 1.34.23-2 but it turns out that the build was failing because I didn't have git setup properly and I didn't notice. After fixing that and rebuilding, everything works now! Thanks so much! People like you are what makes Arch so great.

connortechnology, 2021-04-20:
1.34.24 updated cakephp to support php8

closed 2021-04-20 by connortechnology