ZoneMinder/zoneminder #3166: ZMStats.pl Keeps Crashing

manjotsc, 2021-02-20:
ZMStats.pl is keep crashing continuously.

Ubuntu 18.04
ZoneMinder v1.34.23

```
2021-02-20 03:46:47.209950 zmstats[454].INF-zmstats.pl [Stats Daemon starting in 30 seconds]
2021-02-20 03:46:47.108610 zmdc[454].INF-zmdc.pl ['zmstats.pl' started at 21/02/20 03:46:47]
2021-02-20 03:46:47.105420 zmdc[371].INF-zmdc.pl ['zmstats.pl' starting at 21/02/20 03:46:47, pid = 454]
2021-02-20 03:46:46.876660 zmdc[449].INF-zmdc.pl ['zmupdate.pl -c' started at 21/02/20 03:46:46]
2021-02-20 03:46:46.874540 zmdc[371].INF-zmdc.pl ['zmupdate.pl -c' starting at 21/02/20 03:46:46, pid = 449]
2021-02-20 03:46:46.861540 zmwatch[443].INF-zmwatch.pl [Watchdog starting, pausing for 30 seconds]
2021-02-20 03:46:46.790810 zmfilter_2[436].INF-zmfilter.pl [Scanning for events using filter id '2']
2021-02-20 03:46:46.629840 zmdc[443].INF-zmdc.pl ['zmwatch.pl' started at 21/02/20 03:46:46]
2021-02-20 03:46:46.627200 zmdc[371].INF-zmdc.pl ['zmwatch.pl' starting at 21/02/20 03:46:46, pid = 443]
2021-02-20 03:46:46.532300 zmfilter_1[430].INF-zmfilter.pl [Scanning for events using filter id '1']
2021-02-20 03:46:46.336560 zmdc[436].INF-zmdc.pl ['zmfilter.pl --filter_id=2 --daemon' started at 21/02/20 03:46:46]
2021-02-20 03:46:46.323650 zmdc[371].INF-zmdc.pl ['zmfilter.pl --filter_id=2 --daemon' starting at 21/02/20 03:46:46, pid = 436]
2021-02-20 03:46:46.123910 zmdc[430].INF-zmdc.pl ['zmfilter.pl --filter_id=1 --daemon' started at 21/02/20 03:46:46]
2021-02-20 03:46:46.121590 zmdc[371].INF-zmdc.pl ['zmfilter.pl --filter_id=1 --daemon' starting at 21/02/20 03:46:46, pid = 430]
2021-02-20 03:46:46.028300 zmpkg[315].INF-zmpkg.pl [Single server configuration detected. Starting up services.]
2021-02-20 03:46:45.874910 zmdc[371].INF-zmdc.pl [Socket should be open at /run/zm/zmdc.sock]
2021-02-20 03:46:44.828440 zmdc[371].INF-zmdc.pl [Server starting at 21/02/20 03:46:44]
2021-02-20 03:46:44.592270 zmpkg[315].INF-zmpkg.pl [Command: start]
2021-02-20 03:46:44.586160 zmpkg[315].INF-zmpkg.pl [Sanity checking States table...]
2021-02-20 03:46:39.605110 zmdc[372].INF-zmdc.pl [Server shutdown at 21/02/20 03:46:39]
2021-02-20 03:46:39.598320 zmdc[372].INF-zmdc.pl ['zmwatch.pl' exited, signal 14]
2021-02-20 03:46:38.595940 zmdc[372].INF-zmdc.pl ['zmfilter.pl --filter_id=2 --daemon' exited normally]
2021-02-20 03:46:38.591800 zmdc[372].INF-zmdc.pl ['zmfilter.pl --filter_id=1 --daemon' exited normally]
2021-02-20 03:46:38.585270 zmdc[372].INF-zmdc.pl ['zmstats.pl' exited, signal 14]
2021-02-20 03:46:38.579490 zmdc[372].INF-zmdc.pl ['zmupdate.pl -c' exited, signal 14]
2021-02-20 03:46:38.567020 zmfilter_2[437].INF-zmfilter.pl [Received TERM, exiting]
2021-02-20 03:46:38.566900 zmdc[372].INF-zmdc.pl ['zmwatch.pl' sending stop to pid 444 at 21/02/20 03:46:38]
2021-02-20 03:46:38.561180 zmdc[372].INF-zmdc.pl ['zmfilter.pl --filter_id=2 --daemon' sending stop to pid 437 at 21/02/20 03:46:38]
2021-02-20 03:46:38.548430 zmfilter_1[431].INF-zmfilter.pl [Received TERM, exiting]
2021-02-20 03:46:38.548250 zmdc[372].INF-zmdc.pl ['zmstats.pl' sending stop to pid 455 at 21/02/20 03:46:38]
2021-02-20 03:46:38.542210 zmdc[372].INF-zmdc.pl ['zmfilter.pl --filter_id=1 --daemon' sending stop to pid 431 at 21/02/20 03:46:38]
2021-02-20 03:46:38.538190 zmdc[372].INF-zmdc.pl ['zmupdate.pl -c' sending stop to pid 450 at 21/02/20 03:46:38]
2021-02-20 03:46:38.533610 zmdc[372].INF-zmdc.pl [Server exiting at 21/02/20 03:46:38]
2021-02-20 03:46:38.320870 zmpkg[509].INF-zmpkg.pl [Command: stop]
2021-02-20 03:46:38.315290 zmpkg[509].INF-zmpkg.pl [Sanity checking States table...]
2021-02-20 03:45:55.162088 web_php[223].ERR-/usr/share/zoneminder/www/includes/functions.php/2423 [ZoneMinder is not configured properly: php's date.timezone -0500 does not match the system timezone +0000! Please check Options->System->Timezone.]
2021-02-20 03:45:52.421040 zmstats[455].INF-zmstats.pl [Stats Daemon starting in 30 seconds]
2021-02-20 03:45:52.323800 zmdc[455].INF-zmdc.pl ['zmstats.pl' started at 21/02/20 08:45:52]
2021-02-20 03:45:52.310240 zmdc[372].INF-zmdc.pl ['zmstats.pl' starting at 21/02/20 08:45:52, pid = 455]
2021-02-20 03:45:52.121320 zmdc[450].INF-zmdc.pl ['zmupdate.pl -c' started at 21/02/20 08:45:52]
2021-02-20 03:45:52.119450 zmdc[372].INF-zmdc.pl ['zmupdate.pl -c' starting at 21/02/20 08:45:52, pid = 450]
2021-02-20 03:45:52.105180 zmwatch[444].INF-zmwatch.pl [Watchdog starting, pausing for 30 seconds]
2021-02-20 03:45:52.021630 zmfilter_2[437].INF-zmfilter.pl [Scanning for events using filter id '2']
2021-02-20 03:45:51.910030 zmdc[444].INF-zmdc.pl ['zmwatch.pl' started at 21/02/20 08:45:51]
2021-02-20 03:45:51.884250 zmdc[372].INF-zmdc.pl ['zmwatch.pl' starting at 21/02/20 08:45:51, pid = 444]
2021-02-20 03:45:51.810450 zmfilter_1[431].INF-zmfilter.pl [Scanning for events using filter id '1']
2021-02-20 03:45:51.599460 zmdc[437].INF-zmdc.pl ['zmfilter.pl --filter_id=2 --daemon' started at 21/02/20 08:45:51]
2021-02-20 03:45:51.597240 zmdc[372].INF-zmdc.pl ['zmfilter.pl --filter_id=2 --daemon' starting at 21/02/20 08:45:51, pid = 437]
2021-02-20 03:45:51.402310 zmdc[431].INF-zmdc.pl ['zmfilter.pl --filter_id=1 --daemon' started at 21/02/20 08:45:51]
2021-02-20 03:45:51.399920 zmdc[372].INF-zmdc.pl ['zmfilter.pl --filter_id=1 --daemon' starting at 21/02/20 08:45:51, pid = 431]
2021-02-20 03:45:51.303320 zmpkg[316].INF-zmpkg.pl [Single server configuration detected. Starting up services.]
2021-02-20 03:45:51.160340 zmdc[372].INF-zmdc.pl [Socket should be open at /run/zm/zmdc.sock]
2021-02-20 03:45:50.102590 zmdc[372].INF-zmdc.pl [Server starting at 21/02/20 08:45:50]
2021-02-20 03:45:49.868260 zmpkg[316].INF-zmpkg.pl [Command: start]
2021-02-20 03:45:49.861400 zmpkg[316].INF-zmpkg.pl [Sanity checking States table...]
2021-02-20 03:45:44.734030 zmdc[366].INF-zmdc.pl [Server shutdown at 21/02/20 08:45:44]
2021-02-20 03:45:44.727420 zmdc[366].INF-zmdc.pl ['zmstats.pl' exited, signal 14]
2021-02-20 03:45:43.723510 zmdc[366].INF-zmdc.pl ['zmupdate.pl -c' exited, signal 14]
2021-02-20 03:45:43.719690 zmdc[366].INF-zmdc.pl ['zmfilter.pl --filter_id=2 --daemon' exited normally]
2021-02-20 03:45:43.715860 zmdc[366].INF-zmdc.pl ['zmfilter.pl --filter_id=1 --daemon' exited normally]
2021-02-20 03:45:43.709730 zmdc[366].INF-zmdc.pl ['zmwatch.pl' exited, signal 14]
2021-02-20 03:45:43.697360 zmfilter_2[437].INF-zmfilter.pl [Received TERM, exiting]
2021-02-20 03:45:43.697230 zmdc[366].INF-zmdc.pl ['zmstats.pl' sending stop to pid 455 at 21/02/20 08:45:43]
2021-02-20 03:45:43.691940 zmdc[366].INF-zmdc.pl ['zmfilter.pl --filter_id=2 --daemon' sending stop to pid 437 at 21/02/20 08:45:43]
[... 13 lines cut ...]
2021-02-20 03:42:40.969490 zmfilter_2[437].INF-zmfilter.pl [Scanning for events using filter id '2']
2021-02-20 03:42:40.830390 zmdc[444].INF-zmdc.pl ['zmwatch.pl' started at 21/02/20 08:42:40]
2021-02-20 03:42:40.807830 zmdc[366].INF-zmdc.pl ['zmwatch.pl' starting at 21/02/20 08:42:40, pid = 444]
2021-02-20 03:42:40.727480 zmfilter_1[431].INF-zmfilter.pl [Scanning for events using filter id '1']
2021-02-20 03:42:40.536330 zmdc[437].INF-zmdc.pl ['zmfilter.pl --filter_id=2 --daemon' started at 21/02/20 08:42:40]
2021-02-20 03:42:40.532180 zmdc[366].INF-zmdc.pl ['zmfilter.pl --filter_id=2 --daemon' starting at 21/02/20 08:42:40, pid = 437]
2021-02-20 03:42:40.335780 zmdc[431].INF-zmdc.pl ['zmfilter.pl --filter_id=1 --daemon' started at 21/02/20 08:42:40]
2021-02-20 03:42:40.333790 zmdc[366].INF-zmdc.pl ['zmfilter.pl --filter_id=1 --daemon' starting at 21/02/20 08:42:40, pid = 431]
2021-02-20 03:42:40.239270 zmpkg[314].INF-zmpkg.pl [Single server configuration detected. Starting up services.]
2021-02-20 03:42:40.080630 zmdc[366].INF-zmdc.pl [Socket should be open at /run/zm/zmdc.sock]
2021-02-20 03:42:39.039720 zmdc[366].INF-zmdc.pl [Server starting at 21/02/20 08:42:39]
2021-02-20 03:42:38.813790 zmpkg[314].INF-zmpkg.pl [Command: start]
2021-02-20 03:42:38.808190 zmpkg[314].INF-zmpkg.pl [Sanity checking States table...]
2021-02-20 03:42:33.270860 zmdc[383].INF-zmdc.pl [Server shutdown at 21/02/20 08:42:33]
2021-02-20 03:42:33.264900 zmdc[383].INF-zmdc.pl ['zmstats.pl' exited, signal 14]
2021-02-20 03:42:32.260990 zmdc[383].INF-zmdc.pl ['zmwatch.pl' exited, signal 14]
2021-02-20 03:42:32.258860 zmdc[383].INF-zmdc.pl ['zmfilter.pl --filter_id=2 --daemon' exited normally]
2021-02-20 03:42:32.255000 zmdc[383].INF-zmdc.pl ['zmfilter.pl --filter_id=1 --daemon' exited normally]
2021-02-20 03:42:32.248840 zmdc[383].INF-zmdc.pl ['zmupdate.pl -c' exited, signal 14]
2021-02-20 03:42:32.237290 zmfilter_2[437].INF-zmfilter.pl [Received TERM, exiting]
2021-02-20 03:42:32.237160 zmdc[383].INF-zmdc.pl ['zmstats.pl' sending stop to pid 455 at 21/02/20 08:42:32]
2021-02-20 03:42:32.231820 zmdc[383].INF-zmdc.pl ['zmfilter.pl --filter_id=2 --daemon' sending stop to pid 437 at 21/02/20 08:42:32]
2021-02-20 03:42:32.217600 zmfilter_1[431].INF-zmfilter.pl [Received TERM, exiting]

```

connortechnology, 2021-02-22:
This log doesn't show zmstats crashing. It shows them being killed as part of a ZM shutdown.  We would likely need a debug level log to even guess what is going on.

knight-of-ni, 2021-02-23:
Before you do anything, set your timezone like the message says

```
2021-02-20 03:45:55.162088 web_php[223].ERR-/usr/share/zoneminder/www/includes/functions.php/2423 [ZoneMinder is not configured properly: php's date.timezone -0500 does not match the system timezone +0000! Please check Options->System->Timezone.]
``

manjotsc, 2021-02-25:
@connortechnology  @knight-of-ni  I set timezone correclty now, it still doing same thing 

`2021-02-24 19:17:51.793771 web_php[23550].DBG-ajax/log.php/225 [Exporting to /tmp/zm/zm-log-4cc9b196.txt]
2021-02-24 19:17:51.788501 web_php[23550].DBG-index.php/213 [View: request Request: log Action:  User: admin]
2021-02-24 19:17:51.782405 web_php[23550].DBG-includes/session.php/20 [Setting cookie parameters to lifetime(3600) path(/) domain () secure() httpOnly(1) name:ZMSESSID]
2021-02-24 19:17:49.773825 web_php[23550].DBG-index.php/213 [View: request Request: log Action:  User: admin]
2021-02-24 19:17:49.769002 web_php[23550].DBG-includes/session.php/20 [Setting cookie parameters to lifetime(3600) path(/) domain () secure() httpOnly(1) name:ZMSESSID]
2021-02-24 19:17:47.709027 web_php[23550].DBG-index.php/213 [View: request Request: log Action:  User: admin]
2021-02-24 19:17:47.704373 web_php[23550].DBG-includes/session.php/20 [Setting cookie parameters to lifetime(3600) path(/) domain () secure() httpOnly(1) name:ZMSESSID]
2021-02-24 19:17:45.652670 web_php[23550].DBG-index.php/213 [View: request Request: log Action:  User: admin]
2021-02-24 19:17:45.647693 web_php[23550].DBG-includes/session.php/20 [Setting cookie parameters to lifetime(3600) path(/) domain () secure() httpOnly(1) name:ZMSESSID]
2021-02-24 19:17:43.597014 web_php[23550].DBG-index.php/213 [View: request Request: log Action:  User: admin]
2021-02-24 19:17:43.590957 web_php[23550].DBG-includes/session.php/20 [Setting cookie parameters to lifetime(3600) path(/) domain () secure() httpOnly(1) name:ZMSESSID]
2021-02-24 19:17:41.536749 web_php[23550].DBG-index.php/213 [View: request Request: log Action:  User: admin]
2021-02-24 19:17:41.531301 web_php[23550].DBG-includes/session.php/20 [Setting cookie parameters to lifetime(3600) path(/) domain () secure() httpOnly(1) name:ZMSESSID]
2021-02-24 19:17:39.477683 web_php[23550].DBG-index.php/213 [View: request Request: log Action:  User: admin]
2021-02-24 19:17:39.470442 web_php[23550].DBG-includes/session.php/20 [Setting cookie parameters to lifetime(3600) path(/) domain () secure() httpOnly(1) name:ZMSESSID]
2021-02-24 19:17:37.426374 web_php[23550].DBG-index.php/213 [View: request Request: log Action:  User: admin]
2021-02-24 19:17:37.421729 web_php[23550].DBG-includes/session.php/20 [Setting cookie parameters to lifetime(3600) path(/) domain () secure() httpOnly(1) name:ZMSESSID]
2021-02-24 19:17:35.379073 web_php[23550].DBG-index.php/213 [View: request Request: log Action:  User: admin]
2021-02-24 19:17:35.373144 web_php[23550].DBG-includes/session.php/20 [Setting cookie parameters to lifetime(3600) path(/) domain () secure() httpOnly(1) name:ZMSESSID]
2021-02-24 19:17:33.325175 web_php[23550].DBG-index.php/213 [View: request Request: log Action:  User: admin]
2021-02-24 19:17:33.319697 web_php[23550].DBG-includes/session.php/20 [Setting cookie parameters to lifetime(3600) path(/) domain () secure() httpOnly(1) name:ZMSESSID]
2021-02-24 19:17:31.269044 web_php[23550].DBG-index.php/213 [View: request Request: log Action:  User: admin]
2021-02-24 19:17:31.264317 web_php[23550].DBG-includes/session.php/20 [Setting cookie parameters to lifetime(3600) path(/) domain () secure() httpOnly(1) name:ZMSESSID]
2021-02-24 19:17:29.219846 web_php[23550].DBG-index.php/213 [View: request Request: log Action:  User: admin]
2021-02-24 19:17:29.215037 web_php[23550].DBG-includes/session.php/20 [Setting cookie parameters to lifetime(3600) path(/) domain () secure() httpOnly(1) name:ZMSESSID]
2021-02-24 19:17:27.169982 web_php[23550].DBG-index.php/213 [View: request Request: log Action:  User: admin]
2021-02-24 19:17:27.165193 web_php[23550].DBG-includes/session.php/20 [Setting cookie parameters to lifetime(3600) path(/) domain () secure() httpOnly(1) name:ZMSESSID]
2021-02-24 19:17:24.733027 web_php[23550].DBG-index.php/213 [View: request Request: log Action:  User: admin]
2021-02-24 19:17:24.727033 web_php[23550].DBG-includes/session.php/20 [Setting cookie parameters to lifetime(3600) path(/) domain () secure() httpOnly(1) name:ZMSESSID]
2021-02-24 19:17:22.609222 web_php[23550].DBG-index.php/213 [View: request Request: log Action:  User: admin]
2021-02-24 19:17:22.603446 web_php[23550].DBG-includes/session.php/20 [Setting cookie parameters to lifetime(3600) path(/) domain () secure() httpOnly(1) name:ZMSESSID]
2021-02-24 19:17:22.509826 web_php[23550].DBG-index.php/213 [View: request Request: log Action:  User: admin]
2021-02-24 19:17:22.505015 web_php[23550].DBG-includes/session.php/20 [Setting cookie parameters to lifetime(3600) path(/) domain () secure() httpOnly(1) name:ZMSESSID]
2021-02-24 19:17:22.432791 web_php[23550].DBG-index.php/213 [View: request Request: log Action:  User: admin]
2021-02-24 19:17:22.427201 web_php[23550].DBG-includes/session.php/20 [Setting cookie parameters to lifetime(3600) path(/) domain () secure() httpOnly(1) name:ZMSESSID]
2021-02-24 19:17:22.353268 web_php[23550].DBG-index.php/213 [View: request Request: log Action:  User: admin]
2021-02-24 19:17:22.348956 web_php[23550].DBG-includes/session.php/20 [Setting cookie parameters to lifetime(3600) path(/) domain () secure() httpOnly(1) name:ZMSESSID]
2021-02-24 19:17:22.279660 web_php[23550].DBG-index.php/213 [View: request Request: log Action:  User: admin]
2021-02-24 19:17:22.273868 web_php[23550].DBG-includes/session.php/20 [Setting cookie parameters to lifetime(3600) path(/) domain () secure() httpOnly(1) name:ZMSESSID]
[... cut ...]
2021-02-24 19:17:20.010813 web_php[23569].DBG-includes/session.php/20 [Setting cookie parameters to lifetime(3600) path(/) domain () secure() httpOnly(1) name:ZMSESSID]
2021-02-24 19:17:19.951173 web_php[23569].DBG-index.php/213 [View: request Request: log Action:  User: admin]
2021-02-24 19:17:19.946191 web_php[23569].DBG-includes/session.php/20 [Setting cookie parameters to lifetime(3600) path(/) domain () secure() httpOnly(1) name:ZMSESSID]
2021-02-24 19:17:19.887210 web_php[23569].DBG-index.php/213 [View: request Request: log Action:  User: admin]
`

connortechnology, 2021-03-31:
Check the LOG_DEBUG checkbox and restart.

connortechnology, 2021-04-20:
There is nothing here to support the use of the word crash in relation to zmstats.pl.  I will be closing this unless you provide further information.

closed 2021-05-10 by connortechnology