ZoneMinder/zoneminder #3167: zmc -m 1' crashed, signal 138 when HW accelerated

Tony763, 2021-02-20:
**Describe Your Environment**
- Version of ZoneMinder [commit e562888ed0f1dce5f636674fe58c91355faae82a]
- How you installed ZoneMinder [from-source]
- Full name and version of OS L4T 4.4 - Ubuntu 18.04 -Jetson Nano
- FFmpeg 4.3.1

**Describe the bug**
When hw accelerated codec is used, zmc is crashing.

**Expected behavior**
HW accell as in 1.35.14 was working

**Debug Logs**
```
'zmc -m 1' crashed, signal 138

libavutil/hwcontext.c
448:      if (!dst->buf[0])                                                                                                                                                               
449:          return transfer_data_alloc(dst, src, flags);      <= This crash

./src/zm_packet.cpp
212           /* retrieve data from GPU to CPU */                                                                                                                                            213           zm_dump_video_frame(in_frame, "Before hwtransfer");                                               
214           ret = av_hwframe_transfer_data(new_frame, in_frame, 0);   <= Called by this
```

If more information is needed or if I can help with debugging, let me know.

connortechnology, 2021-02-22:
I need a debug level 4 log.

Tony763, 2021-02-22:
zmc m1 level 4 [zmc_m1_4.txt](https://github.com/ZoneMinder/zoneminder/files/6024937/zmc_m1_4.txt)

connortechnology, 2021-02-22:
Try latest.

Tony763, 2021-02-22:
[all_l4.txt](https://github.com/ZoneMinder/zoneminder/files/6025165/all_l4.txt)
[zmc_m1_l4.txt](https://github.com/ZoneMinder/zoneminder/files/6025167/zmc_m1_l4.txt)

Better, after each restart of zmc I can see first image in monitor, then it hang up.

connortechnology, 2021-02-22:
Seems to die reliably after 48 or 49 frames.

more useful than gdb would be valgrind -v output

Tony763, 2021-02-22:
`valgrind -v zmc -m 1   2>&1 | tee  valgrind.txt`

[valgrind.txt](https://github.com/ZoneMinder/zoneminder/files/6025491/valgrind.txt)

Tony763, 2021-02-22:
Maybe stupid question but `avcodec_register_all();` is no longer needed?

connortechnology, 2021-02-22:
No it was deprecated a while ago.  It stills gets called for older avformat libs in zm_ffmpeg.cpp

Tony763, 2021-02-23:
I tried to change codec to second hw accelerated and it seems to running without crashing zmc, but video is still chocking and I have log full with: 
`Monitor-1: 2900 - Analysing at 0.00 fps from 0 - 0=0 / 1614076054.397177 - 1614075901.833043 = 152.564134]  `
`RTP: PT=69: bad cseq 41ca expected=29fa]`
`CSeq 12 expected, 0 received.]`

And sometime:

`Failed to capture image from monitor 1 Monitor-1 (1/1)] `
`Unable to read packet from stream 0: error -541478725 "End of file".]`

I can't tell if its related or if its new error.

connortechnology, 2021-02-23:
The End Of file thing usually happens for me because zmwatch.pl has told us to restart because we havn't captured an image in a while.
The other errors we sometimes see when we don't keep up with the processing or there is a wifi dropout or something like that.

Tony763, 2021-02-23:
Cameras are on ethernet directly connected to jetson, overal bitrate is about 1.00MB/s on 100MB/s. Even a HW coddec seems to be used, cpu load on zmc is still high, collide between 80 to 100%  then fall down to 10% for few seconds.
Buffers are set to 20 images and there is no overflow message as if I run non hw accelerated codec.

CPU peak seems to happen each time zmc crash and start with `zmc -m 3' crashed, signal 8`

I tried GDB zmc but I only received `Program terminated with signal SIGKILL, Killed.` with no error.

Can I provide some debug that would be usefull to solve this?

connortechnology, 2021-03-31:
I would be nice to have a debug level4 log from latest master code.  

What you are seeing is zmwatch is killing zmc because it isn't getting images fast enough.  Maybe hung up somewhere.

connortechnology, 2021-05-11:
As I don't have a jetson nano, if you don't respond soon I will close this.

Tony763, 2021-05-12:
Sorry, I do not have it in hands now.

closed 2021-05-12 by Tony763