ZoneMinder/zoneminder #3173: ONVIF control module is NOT at all an ONVIF protocol implementation and doesn't work for ONVIF enabled cameras.

akeyx, 2021-02-26:
ZM version: v1.34.23
ZM installation mode: docker container
OS: docker image

Camera details:
- Any camera supporting ONVIF 
- ZoneMinder Source Type: any/ffmpeg

Bug description:
The onvif.pm control module is not at all an ONVIF protocol implementation but rather some camera specific CGI commands. This is very misleading to the end users as one may think that the issue is with the configuration options wheres really this module is not even trying to implement ONVIF in any way.
Netcat ONVIF module is the closest thing that should send ONVIF commands to supported cameras but it's very limited in supported hardware specific onvif API endpoint address. It's authentication nonce generation is also broken.

Proper ONVIF implementation should just require specifying target camera's IP and potentially port + authentication. The reset like actual URI to connect to for sending commands should be discovered automatically. 
Current implementation hard codes the only supported control URI as: /onvif/PTZ

The following pull request does not implement automatic discovery of the control URI but allows specifying it in the user-accessible "Control Address" field of the source's Control tab which along with the "Control Device" field allows configuring the ONVIF basics. 

As the bare minimum the module should allow specifying all necessary connection parameters in this or similar form:
```
Control Address: <scheme>://[<user>:<password>@]<ip_or_host_of_onvif_enabled_camera>[:port][control_uri]
```

akeyx, 2021-02-26:
In regards to the broken nonce in authentication header. This piece of code:
```
my $nonce = chr(int(rand(254))) for (0 .. 20);
```
...doesn't work. Since it simply generates a random string in the following implementation it's replaced with this:
```
my @set = ('0' ..'9', 'A' .. 'Z', 'a' .. 'z');
my $nonce = join '' => map $set[rand @set], 1 .. 20;
```

akeyx, 2021-02-26:
Pull request: https://github.com/ZoneMinder/zoneminder/pull/3174

connortechnology, 2021-03-31:
I merged the PR

closed 2021-03-31 by connortechnology