ZoneMinder/zoneminder #3183: FFMpeg breaks with with Reolink Cameras using RTSP

jeremysherriff, 2021-03-01:
**Describe Your Environment**
- ZM v1.34.23 on Ubuntu 20.04.2
- Following Ubuntu install instructions in Wiki
- Ubuntu Server 20.04.2 in LXC on Proxmox 6.3-2
(I have also tried docker containers on different hosts, with the same outcome)

**If the issue concerns a camera**
- Reolink E1 Pro (note that E1 Pro does not have RTMP stream, only RTSP)
- Anything, tried 6fps - 25fps
- 4MP and 1MP
- ffmpeg with rtsp

**Describe the bug**
FFMpeg decoding appears to glitch badly after some time, with image smears/colour bands/pixilation. Image never recovers.
Bug can be introduced immediately by making a small edit in the monitor definition, such as changing the buffers from '20' to '21'.  Bug happens when ffmpeg stream is stopped and started.
Debug logs for ffmpeg exposed 'bad cseq' errors that lead me to known ffmpeg bugs specifically with Reolink cameras.
I suspect this is why the Reolink instructions for ZM indicate a preference for RTMP.

Issue does not appear to happen with the low-quality 'sub' stream when configured for UDP.
UDP is not supported for the main stream.

**To Reproduce**
Steps to reproduce the behavior:
1. Configure a Reolink camera with ffmpef and source of RTSP stream
2. Wait for natural cycling of ffmpeg process, or make a small edit in the monitor definition such as changing the buffers from '20' to '21'.  Bug happens when ffmpeg stream is stopped and started quickly.
3. View monitor - note screen smearing etc
4. Notice in Reolink app and VLC that the image is fine, but ZM has problems with viewing the camera console.  Also any recorded H264-passthru video is corrupt.

**Expected behavior**
I expect to the camera image to reflect the same as the Reolink app and as per VLC's RTSP stream.

Monitoring the same camera stream using VLC also does not show any issue, confirming that the issue is in ZM not the camera or network.  The Reolink app shows a perfect image the entire time.

Screenshot:
[image]

Example of image smearing in ZM while same camera is still running fine in VLC:
[image]

The image never recovers.

Investigation of the ffmpeg error took me to these links, where similar experiences and a potential fix can be found:
- https://superuser.com/questions/1289019/reolink-rtsp-ffmpeg-rtp-pt-xx-bad-cseq-if-stopped-restarted-quickly
- https://github.com/agsh/rtsp-ffmpeg/issues/35

**Debug Logs**
```
03/01/21 14:05:39.014290 zmc_m3[2652].WAR-zm_ffmpeg.cpp/70 [RTP: PT=61: bad cseq c893 expected=cee4]
03/01/21 14:05:39.023781 zmc_m3[2652].WAR-zm_ffmpeg.cpp/70 [RTP: PT=61: bad cseq 6be0 expected=cee5]
03/01/21 14:05:39.057052 zmc_m3[2652].WAR-zm_ffmpeg.cpp/70 [RTP: PT=60: bad cseq d88c expected=4fd1]
03/01/21 14:05:39.063449 zmc_m3[2652].WAR-zm_ffmpeg.cpp/70 [RTP: PT=60: bad cseq f807 expected=d88e]
03/01/21 14:05:39.082638 zmc_m3[2652].WAR-zm_ffmpeg.cpp/70 [RTP: PT=60: bad cseq 4fd1 expected=f809]
03/01/21 14:05:39.083027 zmc_m3[2652].WAR-zm_ffmpeg.cpp/70 [RTP: PT=61: bad cseq c894 expected=cee5]
03/01/21 14:05:39.094609 zmc_m3[2652].WAR-zm_ffmpeg.cpp/70 [RTP: PT=61: bad cseq 6be1 expected=cee6]
03/01/21 14:05:39.100879 zmc_m3[2652].WAR-zm_ffmpeg.cpp/70 [RTP: PT=61: bad cseq c895 expected=cee6]
03/01/21 14:05:39.112849 zmc_m3[2652].WAR-zm_ffmpeg.cpp/70 [RTP: PT=61: bad cseq 6be2 expected=cee7]
03/01/21 14:05:39.168550 zmc_m3[2652].WAR-zm_ffmpeg.cpp/70 [RTP: PT=60: bad cseq f809 expected=4fd3]
03/01/21 14:05:39.180451 zmc_m3[2652].DB1-zm_ffmpeg_camera.cpp/976 [raw frame from decoder: format 0 yuv420p 1920x1072 linesize:1920x960 pts: 2009881828]
03/01/21 14:05:39.180615 zmc_m3[2652].DB1-zm_ffmpeg_camera.cpp/1086 [ffmpegcamera: width 1920 height 1072 linesize 7680 colours 4 imagesize 8232960]
03/01/21 14:05:39.186193 zmc_m3[2652].WAR-zm_ffmpeg.cpp/70 [RTP: PT=60: bad cseq 4fd3 expected=f80e]
03/01/21 14:05:39.194339 zmc_m3[2652].WAR-zm_ffmpeg.cpp/70 [error while decoding MB 39 4, bytestream -42]
03/01/21 14:05:39.194790 zmc_m3[2652].DB1-zm_ffmpeg.cpp/70 [concealing 7570 DC, 7570 AC, 7570 MV errors in I frame]
03/01/21 14:05:39.313347 zmc_m3[2652].DB1-zm_ffmpeg_camera.cpp/976 [raw frame from decoder: format 0 yuv420p 1920x1072 linesize:1920x960 pts: 2775984748]
03/01/21 14:05:39.313525 zmc_m3[2652].DB1-zm_ffmpeg_camera.cpp/1086 [ffmpegcamera: width 1920 height 1072 linesize 7680 colours 4 imagesize 8232960]
03/01/21 14:05:39.319112 zmc_m3[2652].WAR-zm_ffmpeg.cpp/70 [RTP: PT=60: bad cseq f80f expected=4fd5]
03/01/21 14:05:39.319305 zmc_m3[2652].WAR-zm_ffmpeg.cpp/70 [RTP: PT=60: bad cseq 4fd6 expected=f811]
03/01/21 14:05:39.319457 zmc_m3[2652].WAR-zm_ffmpeg.cpp/70 [RTP: PT=60: bad cseq f812 expected=4fd8]
03/01/21 14:05:39.319595 zmc_m3[2652].WAR-zm_ffmpeg.cpp/70 [RTP: PT=60: bad cseq 4fd9 expected=f814]
03/01/21 14:05:39.319738 zmc_m3[2652].WAR-zm_ffmpeg.cpp/70 [RTP: PT=60: bad cseq f815 expected=4fdb]
03/01/21 14:05:39.319879 zmc_m3[2652].WAR-zm_ffmpeg.cpp/70 [RTP: PT=61: bad cseq c896 expected=cee7]
03/01/21 14:05:39.320015 zmc_m3[2652].WAR-zm_ffmpeg.cpp/70 [RTP: PT=61: bad cseq 6be3 expected=cee8]
03/01/21 14:05:39.320142 zmc_m3[2652].WAR-zm_ffmpeg.cpp/70 [RTP: PT=60: bad cseq 4fdc expected=f817]
03/01/21 14:05:39.320291 zmc_m3[2652].WAR-zm_ffmpeg.cpp/70 [RTP: PT=60: bad cseq f818 expected=4fde]
03/01/21 14:05:39.320370 zmc_m3[2652].WAR-zm_ffmpeg.cpp/70 [RTP: PT=61: bad cseq c897 expected=cee8]
03/01/21 14:05:39.320486 zmc_m3[2652].WAR-zm_ffmpeg.cpp/70 [RTP: PT=61: bad cseq 6be4 expected=cee9]
03/01/21 14:05:39.320632 zmc_m3[2652].WAR-zm_ffmpeg.cpp/70 [RTP: PT=60: bad cseq 4fdf expected=f81a]
03/01/21 14:05:39.320710 zmc_m3[2652].WAR-zm_ffmpeg.cpp/70 [RTP: PT=60: bad cseq d88e expected=f81a]
03/01/21 14:05:39.320855 zmc_m3[2652].WAR-zm_ffmpeg.cpp/70 [RTP: PT=60: bad cseq 4fe0 expected=f81b]
03/01/21 14:05:39.320939 zmc_m3[2652].WAR-zm_ffmpeg.cpp/70 [RTP: PT=60: bad cseq d88f expected=f81b]
03/01/21 14:05:39.326430 zmc_m3[2652].WAR-zm_ffmpeg.cpp/70 [RTP: PT=60: bad cseq 4fe1 expected=f81c]
03/01/21 14:05:39.326547 zmc_m3[2652].WAR-zm_ffmpeg.cpp/70 [RTP: PT=60: bad cseq d890 expected=f81c]
03/01/21 14:05:39.326701 zmc_m3[2652].WAR-zm_ffmpeg.cpp/70 [RTP: PT=60: bad cseq 4fe2 expected=f81d]
03/01/21 14:05:39.326786 zmc_m3[2652].WAR-zm_ffmpeg.cpp/70 [RTP: PT=60: bad cseq d891 expected=f81d]
03/01/21 14:05:39.326895 zmc_m3[2652].WAR-zm_ffmpeg.cpp/70 [RTP: PT=61: bad cseq c898 expected=cee9]
```

I am unsure whether the suggested fix will help for ZM, but these two cameras of mine are unusable with ZM currently.
Happy to test anything or provide more logging if someone can provide specifics of what they want.

jeremysherriff, 2021-03-01:
I've had a look around src/zm_ffmpeg* files in the hope I could see something about ending the stream, but alas no.

In the meantime, I've set the cameras as remote/rtsp and they're working OK.  I'd prefer ffmpeg so that I can use h264 passthrough recording and audio recording, but this is OK for now.

connortechnology, 2021-03-01:
See the reorder_queue_size=nnn option.  It sets buffer for UDP mode.  Goes in Monitor->Source->Options. Not sure what a good value is.  I have had success with 100. Have gone higher as well.

closed 2021-03-01 by connortechnology

jeremysherriff, 2021-03-01:
Thanks. Previously I was unable to get the main streams to work with UDP but this is working fine so far.
I've caused a few disconnect/reconnect events (just switching from modetect to monitor and back) and the connection seems to be re-established without issue.