ZoneMinder/zoneminder #3194: ffplay works, zoneminder doesn't - DCS-8526LH

chrisrd, 2021-03-11:
**Describe Your Environment**
- Version of ZoneMinder: 1.35.20~20210309163038-focal
- How you installed ZoneMinder: PPA
- Full name and version of OS: Ubuntu 20.04 focal
- Browser name and version: n/a

**If the issue concerns a camera**
- Make and Model: D-Link DCS-8526LH
- frame rate: 15 fps
- resolution: 1920x1080
- ZoneMinder Source Type: remote

**Describe the bug**
The camera plays using `ffplay`, e.g.:
```
$ ffplay -rtsp_transport tcp rtsp://admin:pin@ip:554/live/profile.0
Input #0, rtsp, from 'rtsp://admin:pin@ip:554/live/profile.0':
  Metadata:
    title           : Session streamed by D-Link
    comment         : live/profile.0
  Duration: N/A, start: 0.000000, bitrate: N/A
    Stream #0:0: Video: h264 (High), yuvj420p(pc, bt709, progressive), 1920x1080, 15 fps, 14.99 tbr, 90k tbn, 30 tbc
    Stream #0:1: Audio: aac (LC), 16000 Hz, mono, fltp
    Stream #0:2: Audio: opus, 48000 Hz, mono, fltp
[swscaler @ 0x7fe8b0676940] deprecated pixel format used, make sure you did set range correctly
[rtsp @ 0x7fe8d0000bc0] CSeq 7 expected, 0 received.
  36.82 A-V: -0.019 fd=   3 aq=    0KB vq=    0KB sq=    0B f=1/1
```

With the ZM monitor configured as:
```
General
  Source Type: Remote
  Function:  Monitor
Source
  Protocol: RTSP
  Method:  RTP/RTSP
  Host Name:  admin:pin@ip
  Port:  554
  Path:  /live/profile.0
  Capture Resolution (pixels):  1920x1080 1080p
```

The monitor doesn't play, and ZM logs:
```
03/11/21 09:19:29.561324 zmc_m10[2655335].DB2-zm_rtsp.cpp/43 [Sending RTSP message: PLAY rtsp://ispy:554/live/profile.0 RTSP/1.0
Session: AB7F5196
Range: npt=0.000-
Authorization: Digest username="admin", realm="LIVE555 Streaming Media", nonce="61edcd2e69a2a5bc6e945cca58016197", uri="rtsp://ispy:554/live/profile.0
", response="8d2134a46c3e4f6e0b1d4c6a0f04ab73", algorithm="MD5"
User-Agent: ZoneMinder/1.35.20
CSeq: 5

]
03/11/21 09:19:29.563157 zmc_m10[2655335].DB2-zm_rtsp.cpp/63 [Received RTSP response: $^A (2 bytes)]
03/11/21 09:19:29.563213 zmc_m10[2655335].ERR-zm_rtsp.cpp/71 [Response parse failure, 2 bytes follow]
03/11/21 09:19:29.563298 zmc_m10[2655335].ERR-zm_rtsp.cpp/73 [2: 24 01]

```

Tcpdump of the monitor session, from the "PLAY" command:
```
09:19:29.561383 IP 192.168.88.252.52924 > 192.168.88.220.554: Flags [P.], seq 843:1185, ack 1504, win 501, options [nop,nop,TS val 2225334302 ecr 1237
45], length 342: RTSP: PLAY rtsp://ispy:554/live/profile.0 RTSP/1.0
        0x0000:  4500 018a d752 4000 4006 2ef2 c0a8 58fc  E....R@.@.....X.
        0x0010:  c0a8 58dc cebc 022a d4ca 6989 2acd b5f1  ..X....*..i.*...
        0x0020:  8018 01f5 34a6 0000 0101 080a 84a3 e81e  ....4........... 
        0x0030:  0001 e361 504c 4159 2072 7473 703a 2f2f  ...aPLAY.rtsp://
        0x0040:  6973 7079 3a35 3534 2f6c 6976 652f 7072  ispy:554/live/pr
        0x0050:  6f66 696c 652e 3020 5254 5350 2f31 2e30  ofile.0.RTSP/1.0
[... 124 lines cut ...]
        0x00d0:  0d0a 4461 7465 3a20 5765 642c 204d 6172  ..Date:.Wed,.Mar
        0x00e0:  2031 3020 3230 3231 2032 323a 3134 3a34  .10.2021.22:14:4
        0x00f0:  3320 474d 540d 0a52 616e 6765 3a20 6e70  3.GMT..Range:.np
        0x0100:  743d 302e 3030 302d 0d0a 5365 7373 696f  t=0.000-..Sessio
        0x0110:  6e3a 2036 3231 3335 4638 380d 0a52 5450  n:.62135F88..RTP
        0x0120:  2d49 6e66 6f3a 2075 726c 3d72 7473 703a  -Info:.url=rtsp:
        0x0130:  2f2f 3a3a 6666 6666 3a31 3932 2e31 3638  //::ffff:192.168
        0x0140:  2e38 382e 3232 302f 6c69 7665 2f70 726f  .88.220/live/pro
        0x0150:  6669 6c65 2e30 2f76 6964 656f 3b73 6571  file.0/video;seq
        0x0160:  3d32 3439 3533 3b72 7470 7469 6d65 3d32  =24953;rtptime=2
        0x0170:  3734 3932 3434 3338 362c 7572 6c3d 7274  749244386,url=rt
        0x0180:  7370 3a2f 2f3a 3a66 6666 663a 3139 322e  sp://::ffff:192.
        0x0190:  3136 382e 3838 2e32 3230 2f6c 6976 652f  168.88.220/live/
        0x01a0:  7072 6f66 696c 652e 302f 6175 6469 6f3b  profile.0/audio;
        0x01b0:  7365 713d 3335 3638 343b 7274 7074 696d  seq=35684;rtptim
        0x01c0:  653d 3232 3938 3637 3632 332c 7572 6c3d  e=229867623,url=
        0x01d0:  7274 7370 3a2f 2f3a 3a66 6666 663a 3139  rtsp://::ffff:19
        0x01e0:  322e 3136 382e 3838 2e32 3230 2f6c 6976  2.168.88.220/liv
        0x01f0:  652f 7072 6f66 696c 652e 302f 6175 6469  e/profile.0/audi
        0x0200:  6f62 6163 6b3b 7365 713d 303b 7274 7074  oback;seq=0;rtpt
        0x0210:  696d 653d 300d 0a0d 0a                   ime=0....
```

**Expected behavior**
The monitor plays

connortechnology, 2021-03-11:
Have you thought about trying the ffmpeg monitor type?  We actually don't plan to continue support for the remote type.

chrisrd, 2021-03-11:
Huh. Didn't realise that was an option, was thinking that was only for local cameras or something (complete newbie here).

...yep, that works, using:

```
General
   Source Type:  Ffmpeg
   Function:  Monitor
Source
   Source Path:  rtsp://admin:pin@ip:554/live/profile.0
```

Tks!

chrisrd, 2021-03-11:
Closing - given this is an issue with the remote monitor type which is going to go away

closed 2021-03-11 by chrisrd