ZoneMinder/zoneminder #3201: RPMFusion repo for master possibly not being updated?

dodexahedron, 2021-03-23:
Not sure if this is intentional or not, but, since the 20210310 build, no builds have been pushed to rpmfusion for the master branch.

Thought I'd mention in case something broke in the toolchain.

knight-of-ni, 2021-03-23:
RPMFusion? Sorry we only push releases to that repo.

Perhaps you zmrepo

dodexahedron, 2021-03-23:
Oops yes. zmrepo.

knight-of-ni, 2021-03-27:
So this breakage was caused by the addition of a new submodule on march 10. The rpm specfile needs to be updated and tested to reflect the change. I've got other priorities at the moment, but it's on my task list.

knight-of-ni, 2021-04-07:
This first step to fixing this has been made in  commit 3be21d0

knight-of-ni, 2021-04-07:
Rpms are building now and beginning to show in zmrepo. Dnf should report new packages in ~24 hours or less.

dodexahedron, 2021-04-07:
Confirmed they're showing up on a CentOS 8 Stream box now.
Will run an upgrade tonight and report results.

dodexahedron, 2021-04-07:
Upgrade went smoothly and zoneminder seems to be functioning properly.
Thanks!

closed 2021-04-07 by dodexahedron