ZoneMinder/zoneminder #3203: "disable alarms" in live view has no effect

alabamatoy, 2021-03-29:
**Describe Your Environment**
- Version of ZoneMinder [release version, development version, or commit]
    1.34.23
- How you installed ZoneMinder [e.g. PPA, RPMFusion, from-source, etc]
    PPA
- Full name and version of OS
  Ubuntu 18.04.5
- Browser name and version (if this is an issue with the web interface)
   Verified with both Firefox and Chrome on multiple client machines, and on 2 different ZM installs (second ZM install is 1.34.23 and Ubuntu 16.04)

**If the issue concerns a camera**
Does not seem to be tied to specific camera.

**Describe the bug**
In the monitor live view window, clicking on "disable alarms" should result in turning off the motion detection, and the "disable alarms" button changing to "enable alarms" (this is based upon my interpretation of the source code, but may be wrong).

**To Reproduce**
Steps to reproduce the behavior:
1. Go to console, click on monitor name to display live view for a camera in either MODECT or NODECT modes.
2. Once live monitor comes up, click on "disable alarms"  Nothing happens, motion detection continues unabated.

**Expected behavior**
Motion detection should be suspended for the amount of time specified in MAX_SUSPEND_TIME and "disable alarms" should change to "enable alarms"

**Debug Logs**
I am unable to find any errors in any logs pertaining to this.  But I may not know where to look.

<insert debug logs here, please make sure they are within the ``` quotes so they are formatted properly>

```

knight-of-ni, 2021-03-29:
The disable alarm button does not disable motion detection. It silences an active alarm. If there is no active alarm, then the button does nothing. What you see currently in the UI, is exactly the same functionality that was there previously, only there are buttons now instead of clickable text.

I may consolidate the two buttons down to one, which would change appearance, based on the alarm state of the monitor.

connortechnology, 2021-03-29:
I'm not sure that was the original intent.  There is code that disabled motion detection.  This needs review/cleanup to make it more intuitive. I personally don't know why anyone would WANT to turn off motion detection for a period of time.  Also between 1.34 and master the disable would also affect triggers and linked monitors.  I have changed it to only work for motion detection.

knight-of-ni, 2021-03-29:
This is one of the features I looked at when I revised the UI and the underlying JavaScript... the original code simply sent a stop command via ajax to stop the (active) alarm.

Doing something permanent would be problematic, because then the state of the monitor would not reflect the monitor config.

alabamatoy, 2021-03-29:
"Doing something permanent would be problematic, because then the state of the monitor would not reflect the monitor config."  Turn it back on after expiration of MAX_SUSPEND_TIME.....

knight-of-ni, 2021-03-29:
> "I personally don't know why anyone would WANT to turn off motion detection for a period of time. " PTZ creates gobs of alarms. Turn off motion detection, slew camera, let it come back on after timeout.

I agree. It's just one of those features that has been there since the beginning of time. I am hesitant to change legacy features like this, because it is hard to tell who might actually be using it... despite us agreeing in this thread that it is not useful to us.

alabamatoy, 2021-03-29:
How do yall suggest that PTZ be used with motion detection?  Go into the monitor and change the state to nodect and then slew, then turn it back on again?  That works, but its a very poor implementation IMHO.  I respectfully suggest that the button does not "disable alarms", but perhaps it does "suspend active alarm".  From my perspective, disabling alarms for a period of time would be far more useful.

"Your pull request is welcome"  Its just plain beyond my skill set.  Yall have made the design effective, but so complicated that simpletons like me can no longer reverse engineer it to make such enhancements.

alabamatoy, 2021-03-30:
The Amcrest PTZ script has an "open" and a "close" sub.  Would suspending motion detection not be as simple as adding to the "open" sub the following:
- Determine monitor ID
- clear flags
- Check monitor ID state
-  - If state is MODECT change it to NODECT, set MODECT flag
-  - If state is MOCORD change it to RECORD, set MOCORD flag

In the "close" sub, add the following
- Determine monitor ID
- Check monitor ID state
- If NODECT and MODECT flag, set state to MODECT
- if RECORD and MOCORD flag, set state to MOCORD
- clear flags

I can do this using the API, but I suspect that is not how yall would want it done since it would probably be very slow and would require the API credentials to be in the code.

connortechnology, 2021-03-30:
No because open and close happen at startup and shutdown of the script. Not at each command.  

Theoretically not all commands would need to stop motion detection (not many but maybe).  

So if there is a generic sendCmd function it could go there.  

I am concerned about the latency of all this.  Should be ok though.  Also... if analysis is behind capture, we could theoretically pause modetect, move cam, start modect and then still be modecting the frames captured while moving camera but after we turn modect back on.  

First step is to add a config element to monitors to enable this functionality.

connortechnology, 2021-04-20:
This was fixed in 1.34.24

closed 2021-04-20 by connortechnology