ZoneMinder/zoneminder #3219: The Stats table is no longer being updated

neillbell, 2021-04-21:
**Describe Your Environment**
- Version of ZoneMinder: 1.35.26 (1.35.26~20210421084449-bionic)
- How you installed ZoneMinder: PPA zoneminder-master
- Full name and version of OS: Ubuntu 18.04.5

**Describe the bug**
The ZoneMinder Stats table is no longer being updated.

**Expected behavior**
The ZoneMinder Stats table should be updated with the motion detection stats for each analyzed frame.

When I examine the Stats table the last event for which stats were recorded was at: 2021-04-19 20:16:09. This corresponds to the time that I installed zoneminder 1.35.26~20210419170703-bionic.

connortechnology, 2021-04-21:
So... I've done a partial fix on this. Needs more testing/work.  We are now writing some stats, even better we are queuing them and batch writing them, which should reduce db load.

neillbell, 2021-04-22:
I'm not seeing any stats being written at all. Is there something I need to do to get them working?

connortechnology, 2021-04-26:
A number of fixes have been made.  Please check latest.

connortechnology, 2021-05-10:
I believe this is fixed. CLosing.

closed 2021-05-10 by connortechnology