ZoneMinder/zoneminder #3225: Frame send time * msec too slow, throttling maxfps

Jsalas424, 2021-04-29:
**Describe Your Environment**

**If the issue concerns a camera**

**Describe the bug**

**To Reproduce**

**Expected behavior**

**Debug Logs**
```
<img width="1439" alt="image" src="https://user-images.githubusercontent.com/51519895/116555355-4934f680-a8ca-11eb-939f-97b1cf016564.png">

<!--
<insert debug logs here, please make sure they are within the ``` quotes so they are formatted properly>
-->

```

Jsalas424, 2021-04-29:
Updated post with level 9 debug logs for zms_m1: https://pastebin.com/uP63ByXe

connortechnology, 2021-05-10:
What is interesting here is that each frame gets progressively slower.  WHY?

A better approach might be to scale the image or otherwise reduce the size/complexity of the image.  This would save cpu as well as bandwidth.

connortechnology, 2021-05-10:
@alabamatoy for the session.

This issue would have to be either cpu or bandwidth/latency.  We take the time, then encode the jpeg in ram, then print it out to the socket, take the time again.  If the amount of time is greater than the desired fps, we decrease the fps so that we wait less between sending frames.

Your graph shows pretty low cpu use and encoding jpeg's is not a time consuming operation. So I'm assuming network problems.

Jsalas424, 2021-05-10:
I live in an apartment complex, 2.4GHz interference is definitely a reality and I'm assuming the wireless NIC in that Amcrest isn't the best, but there is only 30ft or so between the camera and the AP. I guess the only way to confirm is to hardwire it. I'll try to do that this weekend and report back.

connortechnology, 2021-05-10:
as already stated, the issue is not with the camera. It is between your browser and ZM.  wifi is inherently unstable.  So wiring the camera in will improve your available wifi bandwidth but I suspect the issues will persist.

Jsalas424, 2021-05-10:
> as already stated, the issue is not with the camera. It is between your browser and ZM. wifi is inherently unstable. So wiring the camera in will improve your available wifi bandwidth but I suspect the issues will persist.

I just reread that and saw the issue in my logic. So would a speed test between my end device (laptop on LAN wifi) and my ZM VM be provide more valuable information?

Jsalas424, 2021-05-10:
Or am I being dense here and are you specifically referring to just the browser and not the connection between the end devices and the ZM VM? In that case I should have stated that when I generated these error logs, I was viewing my stream via ZMNinja not through my Firefox browser, although I can recreate this through Firefox if that is of value.

Jsalas424, 2021-05-10:
Here I'm connecting from client to Zoneminder VM and getting 73 Mbits/s throughput tested with iperf3

[image]

Jsalas424, 2021-06-07:
This is solved and was due a 2.4GHz band wifi interference issue

The Amcrest IP2M-841W only uses 2.4GHz making it prone to signal interference is crowded environment, for example my apartment building in a major city. 

I noticed that the network adapter for my ZM VM had increased usage right when I went on vacation:
[image]

and then tanked when I got back:
[image]

My AP (Unifi Dream Machine) broadcasts 3 wifi channels as seperate VLANs. My main and guest channel broadcasts both on the 2.4GHz and 5GHz channels, but only at 2.4GHz for the camera channel. I turned off 2.4GHz on my main wifi channel since I know my main devices all support 5GHz anyways, and it fixed it!

[image]

closed 2021-06-07 by Jsalas424