ZoneMinder/zoneminder #3228: Frame skipping from first alert frame.

Jaska8, 2021-05-09:
**Describe Your Environment**
- Version of ZoneMinder v1.34.26
- How you installed ZoneMinder Unraid Docker https://hub.docker.com/r/dlandon/zoneminder.unraid/
- Full name and version of OS Unraid 6.9.2

**If the issue concerns a camera**
- Make and Model Some chinese copy of general camera
- frame rate 25fps
- resolution 1280x720
- ZoneMinder Source Type: ffmpeg

**Describe the bug**
When using Modect every recording from point of first alarm frame I get 5-10~ frame skip. This does not happen when using h264 passthrough mode.

When changing "Alarm Frame Count" it will delay the skip that many frames. For example using 5 Alarm Frame Count will make the video skip after 5 alarm frames.
[image]
[image]

connortechnology, 2021-05-10:
It is not entirely clear what you mean.

Are you comparing passthrough to recording jpegs?  Alarm frame count controls how many alarmed frames are required to trigger recording.  It should have no effect once recording has begun.

Jaska8, 2021-05-10:
> It is not entirely clear what you mean.
> 
> Are you comparing passthrough to recording jpegs? Alarm frame count controls how many alarmed frames are required to trigger recording. It should have no effect once recording has begun.

Yes I'm comparing recording jpeg/x264 encode to passthrough recording.

Let's just ignore the alarm frame count for now.

I have 50 Pre Event Images and 50 Post Event Images , so there is 50 frames before the first alarm frame, after the first alarm frame is shown the saved jpeg/x264 encode will skip bunch of frames but will record normally after the first skip.

Live streaming does not have this problem nor the passthrough mode. I just want to save the date and time to the events because the cameras own clock gets out of sync quite easily.

connortechnology, 2021-05-10:
I need a debug level 3 log from zma.

Jaska8, 2021-05-11:
> I need a debug level 3 log from zma.

Sorry, for some reason I only can get INF level logs from zma. What settings do I need to change and do I output the debug file direct to disk or export from web log viewer?

Jaska8, 2021-05-11:
Ok got it working, had to "restart" the source. Here is the dbg3 from zma.
[dedug.log](https://github.com/ZoneMinder/zoneminder/files/6460079/dedug.log)

connortechnology, 2021-05-11:
I think I sorta  know what's going on.  Problem is this code is so nasty it is going to be difficult for me to fix.  

Good news is I already rewrote it all in current master which is to be released as 1.36 in the next few days.  

I might suggest you try

Jaska8, 2021-05-26:
This has been resolved with 1.36. Thanks!

closed 2021-05-26 by Jaska8