ZoneMinder/zoneminder #3230: zmc crashing on startup

dodexahedron, 2021-05-13:
**Describe Your Environment**
- Version of ZoneMinder 1.35.28-1.45.20210513gitgd52b53e69
- How you installed ZoneMinder zmrepo
- Full name and version of OS CentOS Stream 8.3 Kernel version 5.12.3

**Describe the bug**
On start of the service, a stack trace is generated by zmc and no zmc processes are able to stay running.
Stack trace below.

**To Reproduce**
Steps to reproduce the behavior:
1. systemctl start zoneminder

**Debug Logs**
```

05/13/21 10:39:29.772259 zmc_m11[40779].INF-zmc.cpp/220 [Starting Capture version 1.35.28]
05/13/21 10:39:29.772436 zmc_m11[40779].ERR-zm_signal.cpp/49 [Got signal 6 (Aborted), crashing]
05/13/21 10:39:29.772473 zmc_m11[40779].ERR-zm_signal.cpp/79 [Signal address is 0x3000009f4b, from 0x7f7d51a197ff]
05/13/21 10:39:29.772654 zmc_m11[40779].ERR-zm_signal.cpp/100 [Backtrace 0: /usr/bin/zmc(+0xcf43a) [0x55f9c3acf43a]]
05/13/21 10:39:29.772671 zmc_m11[40779].ERR-zm_signal.cpp/100 [Backtrace 1: /lib64/libpthread.so.0(+0x12b20) [0x7f7d55170b20]]
05/13/21 10:39:29.772681 zmc_m11[40779].ERR-zm_signal.cpp/100 [Backtrace 2: /lib64/libc.so.6(gsignal+0x10f) [0x7f7d51a197ff]]
05/13/21 10:39:29.772691 zmc_m11[40779].ERR-zm_signal.cpp/100 [Backtrace 3: /lib64/libc.so.6(abort+0x127) [0x7f7d51a03c35]]
05/13/21 10:39:29.772701 zmc_m11[40779].ERR-zm_signal.cpp/100 [Backtrace 4: /usr/bin/zmc(+0x543e8) [0x55f9c3a543e8]]
05/13/21 10:39:29.772710 zmc_m11[40779].ERR-zm_signal.cpp/100 [Backtrace 5: /usr/bin/zmc(+0x6f23f) [0x55f9c3a6f23f]]
05/13/21 10:39:29.772719 zmc_m11[40779].ERR-zm_signal.cpp/100 [Backtrace 6: /usr/bin/zmc(main+0x57b) [0x55f9c3a42deb]]
05/13/21 10:39:29.772728 zmc_m11[40779].ERR-zm_signal.cpp/100 [Backtrace 7: /lib64/libc.so.6(__libc_start_main+0xf3) [0x7f7d51a057b3]]
05/13/21 10:39:29.772738 zmc_m11[40779].ERR-zm_signal.cpp/100 [Backtrace 8: /usr/bin/zmc(+0x4588e) [0x55f9c3a4588e]]
05/13/21 10:39:29.772749 zmc_m11[40779].INF-zm_signal.cpp/107 [Backtrace complete, please execute the following command for more information: addr2line -e /usr/bin/zmc 0x55f9c3acf43a 0x7f7d55170b20 0x7f7d51a197ff 0x7f7d51a03c35 0x55f9c3a543e8 0x55f9c3a6f23f 0x55f9c3a42deb 0x7f7d51a057b3 0x55f9c3a4588e]

# addr2line -e /usr/bin/zmc 0x55f9c3acf43a 0x7f7d55170b20 0x7f7d51a197ff 0x7f7d51a03c35 0x55f9c3a543e8 0x55f9c3a6f23f 0x55f9c3a42deb 0x7f7d51a057b3 0x55f9c3a4588e
??:0
??:0
??:0
??:0
??:0
??:0
??:0
??:0
??:0

```

dodexahedron, 2021-05-13:
Happens on 5.11 kernels, as well, and did not happen with zoneminder 1.35.23-1.54.20210407gitg2d7479cb5.

dodexahedron, 2021-05-13:
I walked back previous versions in zmrepo to find when the problem started.
Looks like the first version this starts happening in is 1.35.28-1.24.20210509gitg01834d4dd
All versions before that are able to start and stay running.

connortechnology, 2021-05-14:
I think we fixed this. Please confirm.

dodexahedron, 2021-05-14:
Looks like it works, with the 1.36.0-1.1.20210514gitg2afbbb17d build.
Thanks!
Closing.

closed 2021-05-14 by dodexahedron