ZoneMinder/zoneminder #3231: RtspServer folder is empty into the zip file

CyberGuerro, 2021-05-14:
**Describe Your Environment**
- Version of ZoneMinder 1.36.0
- Full name and version of OS: Gentoo

**/dep/RtspServer** folder is empty and build process fails.
In the repository that folder is a sim-link and I think that doesn't like when you create a zip file.

```

connortechnology, 2021-05-14:
I think you havn't done 
git submodule init
git submodule update --init --recursive

however... it should not be a symlink.

CyberGuerro, 2021-05-14:
I'm sorry, but my installation package get the sources from this url:
https://github.com/ZoneMinder/zoneminder/archive/1.36.0.tar.gz   (try it from your browser)
and the file downloaded has folder empty.
I think that operations you suggest can't be done by Package Manager (if yes how?) but only a real user. Package Manager wait for that archive file is autoconsistent.

SteveGilvarry, 2021-05-14:
GitHub releases does not support sub modules so that file has always been missing and sub modules. Somewhat annoying, @connortechnology in the past we added a file for completeness. https://github.community/t/create-release-that-contains-submodule/1329/36

SteveGilvarry, 2021-05-14:
https://stackoverflow.com/questions/52907659/build-a-project-containing-git-submodules-from-source-package-of-a-github-releas

connortechnology, 2021-05-18:
I think the real issue here is that we should still build without it.

@Carbenium you are the resident expert on cmake I think.  Can you take a look and just not build the rtspserver parts if rtspserver dir is unpopulated?

Carbenium, 2021-05-18:
This certainly can be done.
One thing to consider: IMHO if something is missing to build an activated option (`ZM_NO_RTSPSERVER` CMake option exists to disable this feature), the build should fail.
So instead of just silently (or even with a CMake warning) deactivate this feature when the folder is empty, we should consider to turn this into a CMake error and document there the appropriate steps (either set `ZM_NO_RTSPSERVER` or populate the folder).

What's your take on this @connortechnology ?