ZoneMinder/zoneminder #3232: monitor api get error: missing columns in Monitor table

CyberGuerro, 2021-05-15:
**Describe Your Environment**
- Version of ZoneMinder 1.36.0
- How you installed ZoneMinder: from source
- Full name and version of OS: Gentoo
- Browser name and version: firefox 78.10 (last)

**Describe the bug**
It seems that Monitor API is not coerent with Database Schema in particular on Monitor table.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'http://zoneminder_server/zm/api/monitors.json'
2. returned this error:

data |  
-- | --
name | "An Internal Error Has Occurred."
message | "An Internal Error Has Occurred."
url | "/zm/api/monitors.json"
exception |  
class | "PDOException"
code | "42S22"
message | "SQLSTATE[42S22]: Column not found: 1054 unknown Column 'Monitor.HourEvents' in 'field list'"

`{"success":false,"data":{"name":"An Internal Error Has Occurred.","message":"An Internal Error Has Occurred.","url":"\/zm\/api\/monitors.json","exception":{"class":"PDOException","code":"42S22","message":"SQLSTATE[42S22]: Column not found: 1054 unknown Column 'Monitor.HourEvents' in 'field list'"}}}`

During the fixing with a workaround (adding column to Monitor table) I found that these are the columns "apparently" required and missing:

- HourEvents
- DayEvents
- WeekEvents
- MonthEvents
- ArchivedEvents
- HourEventDiskSpace
- DayEventDiskSpace
- WeekEventDiskSpace
- MonthEventDiskSpace
- ArchivedEventDiskSpace

SteveGilvarry, 2021-05-15:
Those fields are now in event_summaries table, and api/monitors.json is working fine for me. I suspect you have an issue with version of code you have installed.

CyberGuerro, 2021-05-15:
I downloaded from here: https://github.com/ZoneMinder/zoneminder/archive/1.36.0.tar.gz
that is the LAST stable release.....   there is another url to download from?

SteveGilvarry, 2021-05-15:
As I mentioned on your other issue GitHub Releases does not work for sub modules. Repos cannot change or delete the file it auto creates. So have to tell people.
I am not familiar with gentoo, but maybe the existing package would help.

https://gpo.zugaina.org/www-misc/zoneminder
Hopefully that is not your package, but I think not as it seems to point to using live git for packaging source which is what other packages also do.
https://devmanual.gentoo.org/eclass-reference/git-r3.eclass/index.html

CyberGuerro, 2021-05-15:
I've correct that issue with workaround:
- downloaded release zip 
- inserting into it the content of foder RtspServer

AND NEW PACKAGE IS INSTALLED CORRECTLY.

The issue is inside the API's code and NOT in the "missing content of the folder", it is therefore deduced that the bugged source code is inside the tar.gz file.

So my question is:
IS THE 1.36.0 stable release THE LAST AND THE CORRECT VERSION? I'm not so sure of it.

Please can you post your tar.gz source file?
Thank you

connortechnology, 2021-05-18:
Stop yelling.

1.36.0 is the latest stable release, but it is brand new.  Issues are to be expected. 
Our files are posted.  As already mentioned they do not contain outside sources.

Using that api call works for us.  There is something specific wrong with your install.  The api code does not reference HourEvents or any of the other event counts.  You have misrepresented your test case.  I believe you are specifying HourEvents as a filter in your api call.  

The real issue here as far as I'm concerned is that we aren't including the event summaries in the monitor listing.  So I will fix that.

PR #3242 (2021-05-18): Add Event_Summary model so that it is included in monitor listing
Fixes #3232 by including Event_Summary in the api monitor listing.