ZoneMinder/zoneminder #3237: Plaintext Camera Password Shown in Console View

scottyjustknows, 2021-05-17:
**Describe Your Environment**
- ZoneMinder, v1.34.26.
- PPA Installation
- System load:  2.2 Usage of /:   15.6% of 365.16GB   Users logged in:      0
- Accessed on W10 Firefox (88.0 x64) and Chrome 90.0.4430.212 (Official Build) (64-bit)
- $ free -h
                    total        used          free      shared  buff/cache   available
Mem:            62G        3.1G         19G         24G         40G         34G
- $ uname -a 
   Linux sphinx 4.15.0-142-generic #146-Ubuntu SMP Tue Apr 13 01:11:19 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

**If the issue concerns a camera**
- ZoneMinder Source Type: Ffmpeg
- Multiple Hikvision cameras and at least one Lorex

**Describe the bug**
When password for camera contains a pound symbol, the full source URL is shown on the console (including, obviously, the plaintext password).  When a password is used without a pound symbol, only the Source IP is shown.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to source settings
2. Click on Source tab
3. In Source Path, use either a correct or incorrect password with a pound symbol and the entire source path will appear in the Console including plaintext password. 
      3a. `rtsp://<username>:<pass#word>@<IP>//Streaming/Channels/1`
4. When the pound symbol is removed, the Source returns to the correct string, in my case, just the IP address. 
      4a. `rtsp://<username>:<password>@<IP>//Streaming/Channels/1`

**Expected behavior**
Plaintext password not displayed in Console

**Debug Logs**
None

Thank you.

connortechnology, 2021-05-17:
You should find that if you edit the monitor that when you go to save it, it complains about unescaped characters in the url.  This is because it is NOT a valid url.  You need to url-encode those characters in the password.

alabamatoy, 2021-05-18:
I guess I dont see this as an error or bug, if you have access to the details of the monitor, you have access to the details of the monitor which include the credentials for the camera URL.

??

connortechnology, 2021-05-18:
I do think it is a big problem to display the password on the console.  The main issue though is that the url is invalid, and ZM SHOULD have complained and not allowed the saving the of the monitor until those characters were % escaped.  

At least that is the resolution I went with when addressing this very problem.  It gets worse because the ffmpeg libraries can't handle these invalid urls either and will not authenticate to the camera.  If there is a @ in the password it will get confused about what the hostname is in the url, etc.

PR #3240 (2021-05-18): fix warnings not being displayed
Fixes #3237 

While it doesn't fix the actual complaint, the fact is that the url is invalid and php's parse_url can't parse it.  

So instead we will warn about the invalid url.
closed 2021-05-18 by connortechnology