ZoneMinder/zoneminder #3239: The Zoneminder API does not work correctly for alarm operations when using username/password authentication

neillbell, 2021-05-18:
**Describe Your Environment**
- Version of ZoneMinder: 1.36.0 (1.36.0-bionic1)
- How you installed ZoneMinder: PPA
- Full name and version of OS: Ubuntu 18.04.5 LTS

**Describe the bug**
The Zoneminder API does not work correctly for alarm operations when using username/password authentication. Operations fail silently.

**To Reproduce**
Make an API request for the status of a monitor using username/password authentication.

```
curl 'https://example.org/zm/api/monitors/alarm/id:4/command:status.json?username=username&password=********'
{"status":""}
```

**Expected behavior**
The API call should return the monitor status rather than a blank. This works correctly when using token-based authentication.

```
https://example.org/zm/api/monitors/alarm/id:4/command:status.json?token=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJab25lTWluZGVyIiwiaWF0IjoxNjIxMjk1OTU1LCJleHAiOjE2MjEzMDMxNTUsInVzZXIiOiJuZWlsbCIsInR5cGUiOiJhY2Nlc3MifQ.8f3FodI9lDYk1Vhu2OZxlmL5cd_Hurfl2YB9kC3MJqY
{"status":"0"}
```

**Additional information**
The API operations to set a monitor alarm state on or off also fail silently when using username/password authentication.

**Debug Logs**
No errors are generated in the debug logs.

neillbell, 2021-05-25:
Username/password works correctly now, but I think that there is something incorrect about this fix. Querying the API for status used to return the following:
```
  0 => "IDLE",
  1 => "PREALARM",
  2 => "ALARM",
  3 => "ALERT",
  4 => "TAPE"
```
Now the API returns a status of "Ok" when idle and '0' when alarmed or alerted.

neillbell, 2021-05-25:
Also using the API to set the alarm state on or off for a monitor doesn't appear to have any effect.

connortechnology, 2021-05-26:
Ugh.  So yeah I agree we shouldn't be changing the api at this point, but also the api was never actually designed.  So...
How about on error the response will like:
{ "status":"", "output": "stuff", return_code: 1}
and on not error the response  will be
{ "status":"0", status_text:"alarmed", output:"something if verbose was turned on" }

neillbell, 2021-05-26:
I agree that the old way was a little opaque, so I have no problem with this sort of change. Will the non-error status also indicate the distinction between the ALARM and ALERT state like we see displayed on the web UI?

connortechnology, 2021-05-28:
I reverted the behaviour for status.  Also added output. 

Turning alarms on/off works fine here.

PR #3241 (2021-05-18): Fix 3239
Fixes #3239
closed 2021-05-28 by connortechnology