ZoneMinder/zoneminder #3243: Incompatibility with GCC11

Nocifer, 2021-05-18:
**Describe Your Environment**
- Version of ZoneMinder: 1.36.0
- How you installed ZoneMinder: built from source
- Full name and version of OS: Arch Linux

**Describe the bug**
Building on Arch with the recently updated GCC11 results in the following error and fails:

```
In file included from /tmp/makepkg/zoneminder/src/zoneminder-git/src/zm_image.cpp:24:
/tmp/makepkg/zoneminder/src/zoneminder-git/src/zm_utils.h: In instantiation of ‘constexpr const T& ZM::clamp(const T&, const T&, const T&) [with T = unsigned int]’:
/tmp/makepkg/zoneminder/src/zoneminder-git/src/zm_image.cpp:2034:24:   required from here
/tmp/makepkg/zoneminder/src/zoneminder-git/src/zm_utils.h:99:15: error: call of overloaded ‘clamp(const unsigned int&, const unsigned int&, const unsigned int&, std::less<unsigned int>)’ is ambiguous
   99 |   return clamp(v, lo, hi, std::less<T>{});
      |          ~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~
/tmp/makepkg/zoneminder/src/zoneminder-git/src/zm_utils.h:93:20: note: candidate: ‘constexpr const T& ZM::clamp(const T&, const T&, const T&, Compare) [with T = unsigned int; Compare = std::less<unsigned int>]’
   93 | constexpr const T &clamp(const T &v, const T &lo, const T &hi, Compare comp) {
      |                    ^~~~~
In file included from /usr/include/c++/11.1.0/algorithm:62,
                 from /tmp/makepkg/zoneminder/src/zoneminder-git/dep/span-lite/include/span.hpp:225,
                 from /tmp/makepkg/zoneminder/src/zoneminder-git/src/zm_font.h:24,
                 from /tmp/makepkg/zoneminder/src/zoneminder-git/src/zm_image.cpp:22:
/usr/include/c++/11.1.0/bits/stl_algo.h:3674:5: note: candidate: ‘constexpr const _Tp& std::clamp(const _Tp&, const _Tp&, const _Tp&, _Compare) [with _Tp = unsigned int; _Compare = std::less<unsigned int>]’
 3674 |     clamp(const _Tp& __val, const _Tp& __lo, const _Tp& __hi, _Compare __comp)
      |     ^~~~~
make[2]: *** [src/CMakeFiles/zm.dir/build.make:398: src/CMakeFiles/zm.dir/zm_image.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:648: src/CMakeFiles/zm.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```

Building from the latest master produces no issues, so I ran a bisect and found the relevant commit that restores compatibility to be a335e740f3b8abcd3085afaa06820074e4f20e95, which makes sense since the root of the issue is probably that GCC11 is more strict when it comes to C++17 stuff. Could we maybe backport this to the 1.36.x series?

Carbenium, 2021-05-19:
Just for completeness: With GCC11 C++17 got the new default (which is used to build ZM since we only specify a minimum required C++ standard). The C++17 STL introduced `std::clamp` which then gave a name collision with our implementation.

I opened https://github.com/ZoneMinder/zoneminder/pull/3246 with the cherry-picked commit for inclusion in 1.36.

Carbenium, 2021-05-19:
31adca440f1f51673c2725a3759cf80d89e9bf81

closed 2021-05-19 by Carbenium