ZoneMinder/zoneminder #3252: Loading of playback events slow on Chromium and incorrect length on Firefox

ASoTNetworks, 2021-05-21:
**Describe Your Environment**
- Version of ZoneMinder: `v1.36.1`
- How you installed: Deb package via do_debian_package.sh
- Full name and version of OS: Debian Buster
- Browser name and version (if this is an issue with the web interface): Firefox 88.0.1 and Chromium 90.0.4430.212

**Describe the bug**
On Firefox the playback of the video loads fast over WAN but the timestamp of the length of the video is wrong until you watch the whole video

On Chromium the video takes very long to load over a WAN connection but once loaded the video length is correct without playing the whole video first.

**To Reproduce**
Steps to reproduce the behavior:
1. Open a event to playback
2. You will see the issue described above.

**Expected behavior**
Video should load fast and show the right length when viewing events.

**Debug Logs**
none

HaoLVb, 2021-07-20:
I also have the same problem.Did you fix it yet?

ASoTNetworks, 2021-07-20:
This still happens on the latest version.

HaoLVb, 2021-07-20:
go to Monitor->Storage->Encoder Parameters and add movflags=frag_custom+dash+delay_moov
This worked for me!
https://github.com/ZoneMinder/zoneminder/issues/2984

ASoTNetworks, 2021-07-22:
Thanks, I had to use `movflags=faststart` to have it work like it did before.

ronnyadsetts, 2022-09-23:
This is still an issue on 1.36.24 (Debian Bullseye backports). It makes playback and especially scanning alerts pretty painful. Will try the workarounds mentioned above.

ronnyadsetts, 2022-09-23:
I can confirm that adding movflags=faststart to the encoder parameters made playback usable once more. This has been bugging me since I re-installed our camera system as it always worked with the older version of Zoneminder I was running with our previous cameras. Glad I finally found the time and energy to dig in and track down the problem.