ZoneMinder/zoneminder #3253: Repeated error and warning concerning zm_packetqueue.cpp

andornaut, 2021-05-23:
**Describe Your Environment**

- ZoneMinder 1.36.1 
- Running in Docker on a [customized image](https://github.com/andornaut/docker-zoneminder) based on [zoneminderhq/zoneminder:latest-ubuntu18.04](https://hub.docker.com/r/zoneminderhq/zoneminder), but upgraded to v1.36.1 using ppa:iconnor/zoneminder-1.36.
- Ubuntu bionic
- Linux 5.4.0-73-generic #82-Ubuntu SMP Wed Apr 14 17:39:42 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
- Firefox v88.0.1
e)

**If the issue concerns a camera**

I have 3 cameras, which all appear to be working as expected.

**Describe the bug**

AFAICT, everything is working correctly (I can stream, I see thumbnails in the Console, etc), but the logs (vied via the Web UI) are filling up with repeated entries like the following:

```
2021-05-22 20:32:14 | zmc_m3 | &nbsp; | 514 | ERR | &nbsp; | zm_packetqueue.cpp | 135
-- | -- | -- | -- | -- | -- | -- | --
2021-05-22 20:32:14 | zmc_m3 | &nbsp; | 514 | WAR | &nbsp; | zm_packetqueue.cpp | 95
```
[image]

CyberGuerro, 2021-05-23:
Same issue here.

MoHf7f4, 2021-05-23:
I'm seeing the same issue and my syslog is filling with these:
May 23 16:26:06 GlassEyes zmc_m16[8063]: ERR [zmc_m16] [Unable to free up older packets.  Not queueing this video packet.]
May 23 16:26:06 GlassEyes zmc_m22[885]: ERR [zmc_m22] [Unable to free up older packets.  Not queueing this video packet.]
May 23 16:26:06 GlassEyes zmc_m18[858]: ERR [zmc_m18] [Unable to free up older packets.  Not queueing this video packet.]
May 23 16:26:06 GlassEyes zmc_m22[885]: ERR [zmc_m22] [Unable to free up older packets.  Not queueing this video packet.]
May 23 16:26:06 GlassEyes zmc_m22[885]: ERR [zmc_m22] [Unable to free up older packets.  Not queueing this video packet.]
May 23 16:26:06 GlassEyes zmc_m18[858]: ERR [zmc_m18] [Unable to free up older packets.  Not queueing this video packet.]
May 23 16:26:07 GlassEyes zmc_m22[885]: ERR [zmc_m22] [Unable to free up older packets.  Not queueing this video packet.]
May 23 16:26:07 GlassEyes zmc_m18[858]: ERR [zmc_m18] [Unable to free up older packets.  Not queueing this video packet.]
May 23 16:26:07 GlassEyes zmc_m22[885]: ERR [zmc_m22] [Unable to free up older packets.  Not queueing this video packet.]
May 23 16:26:07 GlassEyes zmc_m16[8063]: ERR [zmc_m16] [Unable to free up older packets.  Not queueing this video packet.]

I have plenty of RAM

*@glasseyes:~# free -h
              total        used        free      shared  buff/cache   available
Mem:            23G         12G        5.2G        369M        5.5G        9.9G
Swap:           15G          0B         15G

version: 1.36.1-bionic1
Linux glasseyes 4.15.0-143-generic #147-Ubuntu SMP Wed Apr 14 16:10:11 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 18.04.5 LTS
Release:        18.04

happy to dig further or test.

connortechnology, 2021-05-23:
Increase MaxImageBufferCount.

andornaut, 2021-05-23:
> Increase MaxImageBufferCount.

That worked, thanks!

Suggested improvements:

* Include the error message in the web logs (please see the screenshot in the issue description)
* [UX] Add a link to the FAQ, or a description of the workaround/fix (ie. "Increase MaxImageBufferCount") to the error message.

MoHf7f4, 2021-05-24:
I bumped to the sum of all the other frame settings and errors seem to have stopped.

Big thanks

CyberGuerro, 2021-05-24:
@MoHf7f4 can you explain what do you mean with "I bumped to the sum of all the other frame settings"
which is the correct value for the **MaxImageBufferCount**?

MoHf7f4, 2021-05-24:
> @MoHf7f4 can you explain what do you mean with "I bumped to the sum of all the other frame settings"
> @connortechnology, which is the correct value for the **MaxImageBufferCount**?
@CyberGuerro I was making a best guess, here's what I did.
Sumed:
Image Buffer Size (frames)
Pre Event Image Count
Post Event Image Count
Stream Replay Image Buffer

and put that in: Maximum Image Buffer Size (frames)

CyberGuerro, 2021-05-24:
@MoHf7f4 thx for your explanation, I'll try that :-)

gerazo, 2021-06-09:
@ASoTNetworks 
Increasing MaxImageBufferCount was the key for. However, the behavior of 1.36 changed compared to 1.34 in several ways:
 - The shared memory usage was always MaxImageBufferCount in 1.34, now it is ImageBufferCount only
 - MaxImageBufferCount had to be bigger than max GOP size, now this is not enough and you probably need 2x times the GOP size.
 - In 1.34, you got a warning message explicitly telling what GOP size the decoder met, so you need bigger, now you only get a general message that X value is too small.
 - When using H264 passthrough, it is quite impossible to do alarm mode adaptive video slicing without having a full GOP all the time in the buffer. So I think that ImageBufferCount should also be GOP size, but if you set it lower, you do not get any error messages. But I am not sure whether this would improve anything.

Anyway, you should take a look into your IP cam settings. Many modern cams have dynamic GOP, so although their default setting is 32, they can go up the 256. It does make sense to lower it if you use dynamic GOP or dynamic framerate.

gerazo, 2021-06-09:
Well, I was wrong. After several hours, having MaxImageBufferCount 2x of the max GOP size also produced the error. Something fishy is going on here.

gerazo, 2021-06-10:
I had more look into it: I am using Settings/EVENT_CLOSE_MODE/alarm to have videos separated when alarm kicks in or when it is over, so I can test how accurately the cut is done (again, by using h264 passthrough). When setting ImageBufferCount to a larger value, this cut is way off, or sometimes not even happening (usually at the end of the event). If ImageBufferCount is 3 or 5, the cut is mostly done correctly.
Now I did not have the above error for 24hours by having ImageBufferCount at 3 and MaxImageBufferCount 2x max GOP size.  (and miraculously the memory consumption is only 3 frames large)

CyberGuerro, 2021-06-10:
Excuse me all, can someone explain to me what does GOP mean?
So I can set correctly MaxImageBufferCount.

Thanx

connortechnology, 2021-06-10:
GOP is Group Of Pictures.  It is basically your keyframe interval.  Set MaxImageBufferCount as high as your ram can sustain.  Likely at least 2*GOP size but setting higher should be ok, it just sets the maximum in case slow disks, slow db or something else causes capturing to go faster than analysis.

connortechnology, 2021-09-13:
Please update to 1.36.7 which massively reduces cpu use, which means zm can keep up with freeing the memory. It may resolve this issue.

MoHf7f4, 2021-09-23:
I'm having no issues now.  On 1.36.7.

PhilippeRaven, 2021-11-13:
Hi there,
I recently updated from 1.34 to 1.36.10 on Ubuntu 20.04 through the PPA, and I got this issue too.

hspaay, 2022-03-15:
Same issue here after upgrading to 1.36.12.
@gerazo had an interesting comment:
> When using H264 passthrough, it is quite impossible to do alarm mode adaptive video slicing without having a full GOP all the time in the buffer. So I think that ImageBufferCount should also be GOP [...]

While I don't understand exactly what it means my guess is that with imagepassthrough enabled, there is a higher demand for the buffer. 

After disabling image passthrough and using h264 encoding, the warning went away. Problem solved for me. Hope this helps someone.

hspaay, 2022-03-17:
@gerazo Thank you for the clarification and it makes sense.
So if I understand correctly, the ImageBuffer must be able to hold at least a keyframe + intermediate frames. Therefore if a camera's H264 stream IFrame interval is 5. Does that mean the buffer must be 6 or maybe 12 (two GOP's)?

In my case, the ImageBufferCount is set to 30 and the camera IFrame interval is set to 5. The frame rate is 4 FPS. 
Based on the above it sounds like the buffer is large enough so would that mean that the analysis process can't empty it fast enough?

gerazo, 2022-03-17:
@hspaay You are right. Having a buffer of 5 in case of a 5 GOP size is adequate even in the worst case scenario: a new record is to be done starting with the current frame and even if this is the last p-frame, we still have the corresponding i-frame in the buffer. So the new recording will not miss the current frame (yes, it has a couple of extra frames). So this is a reason for a minimal buffer size.
But no matter how large your buffer is, if your CPU cannot keep up with the filling rate, you can still have problems. Large buffers can survive fluctuations but not constant high load which your hardware cannot keep up with.

hspaay, 2022-03-17:
@gerazo, yes, I understand the CPU constraint, so it could be a factor at play here too. I'm quite happy with zoneminder's performance actually. Zoneminder 1.36.12 is running in a proxmox container with 2 XEON E3-1220 @3.10GHZ CPU's assigned and 10GB of RAM. Configuration is 8 cameras which are set to 4FPS. Proxmox shows a system CPU load of 30-40%, not exceeding 50%, and 4-6GB memory consumed. 

A couple of additional observations from experimentation:

1. The errors depend on the camera. Two cheap Chinese cameras always generate the error, regardless of their IFrame settings but a lower frame rate does make a difference. Makes me wonder if they use the I-Frame setting.
The Hikvisions work as expected.

2. Comments in the ffmpeg encoding output configuration can cause problems. In my case the logs show an error that ffmpeg does not recognize the command '# comment'. The result was that the alarm video started fuzzy and only lasted for the duration of the motion. Removing the comments altogether solved this.