ZoneMinder/zoneminder #3258: Ubuntu PPA for 1.36 is missing tmpfiles.d config

andrewheberle, 2021-05-25:
**Environment**
- Version of ZoneMinder: 1.36.1
- How you installed ZoneMinder: PPA
- Full name and version of OS: Ubuntu 20.04
- Browser name and version: N/A

**Describe the bug**
The Ubuntu 20.04 PPA does not include a systemd tmpfiles config so required directories in /run  and /tmp are not created on boot or at all (if this is an install from scratch)

**To Reproduce**
Steps to reproduce the behavior:
1. Install the Ubunut 20.04 PPA fresh
2. ls -l /usr/lib/tmpfiles.d/zoneminder.conf
3. ls -l /run/zm
4. ls -l /tmp/zm

**Expected behavior**
The file /usr/lib/tmpfiles.d/zoneminder.conf should be present in order to create the required locations and ensure permissions are correct.

connortechnology, 2021-05-26:
Not able to recreate.

My package puts a tmpfiles config at /usr/lib/tmpfiles.d/zoneminder.conf

apt-file on our packages shows it.  

perhaps it didn't install due to being an upgrade or some other conf file already existing.

andrewheberle, 2021-06-13:
Just so anyone else come across this in the future, the `tmpfiles.d` omission was confirmed and resolved in 1.36.3 release as per this forum [post](https://forums.zoneminder.com/viewtopic.php?f=43&t=30821#p121812).

Closing to avoid clogging up the list of issues.

closed 2021-06-13 by andrewheberle