ZoneMinder/zoneminder #3260: Fail to upgrade from 1.26.1 to 1.36.1

marillat, 2021-05-26:
```
$ sudo zmupdate.pl

Initiating database upgrade to version 1.36.1 from version 1.26.0

Do you wish to take a backup of your database prior to upgrading?
This may result in a large file in /tmp/zm if you have a lot of events.
Press 'y' for a backup or 'n' to continue : n

Upgrading database to version 1.36.1
Use of uninitialized value $Config{"ZM_DIR_IMAGES"} in string ne at /usr/bin/zmupdate.pl line 1078, <STDIN> line 1.

Migrating custom config values from Options -> Paths
to /etc/zm/conf.d/zmcustom.conf.

Please verify these values before starting ZoneMinder.

Use of uninitialized value $value in concatenation (.) or string at /usr/bin/zmupdate.pl line 1113, <STDIN> line 1.
Loading config from DB 221 entries
Saving config to DB 221 entries
Upgrading DB to 1.26.1 from 1.26.0

Database successfully upgraded to version 1.26.1.
Upgrading DB to 1.26.2 from 1.26.0

Database successfully upgraded to version 1.26.2.
Upgrading DB to 1.26.3 from 1.26.0

Database successfully upgraded to version 1.26.3.
Upgrading DB to 1.26.5 from 1.26.0

Database successfully upgraded to version 1.26.5.
Upgrading DB to 1.27.0 from 1.26.0

Database successfully upgraded to version 1.27.0.
Upgrading DB to 1.28.0 from 1.26.0

Database successfully upgraded to version 1.28.0.
Upgrading DB to 1.28.1 from 1.26.0

Database successfully upgraded to version 1.28.1.
Upgrading DB to 1.28.99 from 1.26.0
ERROR 1136 (21S01) at line 10: Column count doesn't match value count at row 1
Output: 
Command 'mysql -uzmuser -p'zmpass' -hlocalhost zm < /usr/share/zoneminder/db/zm_update-1.28.99.sql' exited with status: 1
```

closed 2021-05-28 by connortechnology

connortechnology, 2021-05-28:
Hopefully there aren't any other issues. That's a huge upgrade

marillat, 2021-05-28:
I don't use zoneminder often, only for tests.
Maybe I'm the only one who upgrade to a so old DB.
Now I can't try again because I deleted the DB.

commit 829708f8 (2021-05-28): Change SQL to add Controls for ONVIF camera, Foscam 9831W and 8918W to specify columns so that if other columns are present the sql still executes.  Fixes #3260
commit 16cc69ca (2021-05-28): Change SQL to add Controls for ONVIF camera, Foscam 9831W and 8918W to specify columns so that if other columns are present the sql still executes.  Fixes #3260